cjcliffe/CubicSDR #483: Compilation errors on OS X with the new version

mehdideveloper, 2017-01-03:
Hi
After checking out the new version, I've got multiple compilation errors (It was OK the last time I checked, about 1 month ago)

`[  1%] Building CXX object CMakeFiles/CubicSDR.dir/src/CubicSDR.cpp.o

[  2%] Building CXX object CMakeFiles/CubicSDR.dir/src/AppFrame.cpp.o

/Users/mehdi/Source/CubicSDR/src/AppFrame.cpp:535:5: error:
      use of undeclared identifier 'hideBookmarksItem'
    hideBookmarksItem = displayMenu->AppendCheckItem(wxID_DISPLAY_BOOKMA...
    ^
/Users/mehdi/Source/CubicSDR/src/AppFrame.cpp:536:5: error:
      use of undeclared identifier 'hideBookmarksItem'
    hideBookmarksItem->Check(!wxGetApp().getConfig()->getBookmarksVisible());
    ^
/Users/mehdi/Source/CubicSDR/src/AppFrame.cpp:692:5: error:
      use of undeclared identifier 'saveDisabled'
    saveDisabled = false;
    ^
/Users/mehdi/Source/CubicSDR/src/AppFrame.cpp:1028:9: error:
      use of undeclared identifier 'hideBookmarksItem'
        hideBookmarksItem->Check(false);
        ^
/Users/mehdi/Source/CubicSDR/src/AppFrame.cpp:1062:13: error:
      use of undeclared identifier 'hideBookmarksItem'
        if (hideBookmarksItem->IsChecked()) {
            ^
/Users/mehdi/Source/CubicSDR/src/AppFrame.cpp:1319:9: error:
      use of undeclared identifier 'saveDisabled'; did you mean
      'wxColourBase::MakeDisabled'?
    if (saveDisabled) {
        ^~~~~~~~~~~~
        wxColourBase::MakeDisabled
/usr/local/include/wx-3.0/wx/colour.h:161:26: note:
      'wxColourBase::MakeDisabled' declared here
    static void          MakeDisabled(unsigned char* r, unsigned char* g...
                         ^
/Users/mehdi/Source/CubicSDR/src/AppFrame.cpp:1319:9: warning:
      address of function 'MakeDisabled' will always evaluate to 'true'
      [-Wpointer-bool-conversion]
    if (saveDisabled) {
    ~~  ^~~~~~~~~~~~
/Users/mehdi/Source/CubicSDR/src/AppFrame.cpp:1319:9: note:
      prefix with the address-of operator to silence this warning
    if (saveDisabled) {
        ^
        &
/Users/mehdi/Source/CubicSDR/src/AppFrame.cpp:1336:10: error:
      use of undeclared identifier 'hideBookmarksItem'
    if (!hideBookmarksItem->IsChecked()) wxGetApp().getConfig()->setBook...
         ^
/Users/mehdi/Source/CubicSDR/src/AppFrame.cpp:1337:50: error:
      use of undeclared identifier 'hideBookmarksItem'
    wxGetApp().getConfig()->setBookmarksVisible(!hideBookmarksItem->IsCh...
                                                 ^
/Users/mehdi/Source/CubicSDR/src/AppFrame.cpp:1959:5: error:
      use of undeclared identifier 'saveDisabled'
    saveDisabled = state;
    ^
1 warning and 9 errors generated.
make[2]: *** [CMakeFiles/CubicSDR.dir/src/AppFrame.cpp.o] Error 1
make[1]: *** [CMakeFiles/CubicSDR.dir/all] Error 2
make: *** [all] Error 2`

starvald, 2017-01-03:
no good for me either on ubuntu 16.04 x64.

same errors.

vsonnier, 2017-01-03:
Confirmed also on Win64. Problem should be fixed by #484.

cjcliffe, 2017-01-03:
Thanks all for getting this reported and fixed quickly! 👍

closed 2017-01-03 by cjcliffe