cjcliffe/CubicSDR #485: Automatic device selection

Toontje, 2017-01-05:
Can we have an option where CubicSDR automatically selects the device i want to use? I can imagine scenarios where CubicSDR selects the last used device or where i select a device and make it the default device using a checkbox. This prevents the device selection screen to come up and enables for scripted automation.

cjcliffe, 2017-01-06:
@Toontje Didn't make 0.2.1 but I was going to add saving the last selected device in the SDR Devices dialog then select it on startup and focus the run button (so you can just hit enter) -- is that enough for you to script the startup?

x-f, 2017-01-06:
Could you please save the device options as well?
I'm using CubicSDR with FCD++ and I have to change the "Channel Setup" from "Mono Left" to "Complex L/R = I/Q" each time.

Toontje, 2017-01-06:
@cjcliffe I would prefer that CubicSDR just started. Without any interaction from the user side. Not even pressing the Enter or Start button. Once a device is selected it will be used in any future session unless the user selects a different device from the Devices menu.

regice202, 2021-12-30:
@cjcliffe Any chance we can look back into this so I don't have to click through the same menu every time I start Cubic? I'm only using this one device. If I want to change its configuration there's a menu for it inside Cubic already. Thanks for your time!

cjcliffe, 2026-11-14:
Version 0.3.0 dialog will automatically re-select the last used device; only have to click the 'start' button on startup to continue with previous settings

closed 2026-11-14 by cjcliffe