cjcliffe/CubicSDR #486: comments on the bookmark funktion

chpalm, 2017-01-06:
Hi!
When I first saw CubicSDR it was the 0.2.0 version. I was impressed by the cleanliness of the user and of the function. I have with delight and anticipation followed your work with bookmarks that goes with impressive speed. I have downloaded the alpha version 0.2.1, and tested on my system. Thus I have one major comment. I ask you to consider the layout as it makes me unhappy. Please consider:
- the location of the bookmark feature I think the clean layout deserves another placement in the window, probably horizontal in the bottom of the window. A "show / hide" switch is also needed from my point of view.
- automatic saving of configuration in all the glory good but a manual function feels safer. The best is, of course, both. The possibility of a crash exists.

Another minor issues in CubicSDR 0.2.0 as well as 0.2.1. 
Cut and Paste does not work between “Label edit” and “Save session as” 

Regards 
/Christer

cjcliffe, 2017-01-09:
@chpalm thank you for the feedback!

- I hope to someday to make the bookmark placement configurable and able to be moved into a separate window on demand.
- For now there is not a show/hide but you can use the 'Display' -> 'Hide Bookmarks' when needed.
- Adding a way to save / backup bookmarks without having to quit is a good idea; I'll add that to the list too.

Thanks!
-CJ

chleggett, 2017-01-12:
+1 on adding an arrow to the side of the bookmarks panel to collapse it.

P.S. Thanks for the bookmarks @cjcliffe! I came here to request the feature and realized that you had already added it to the 0.2.1-alpha

dunaybd, 2017-08-02:
I'm using CubicSDR under Linux (Linux Mint, derived from Ubuntu). The Bookmarks are not saved in config.xml, nor anywhere visible in my home directory. Each time I start CubicSDR, the bookmarks are gone.

The only error I see in the console window is:
"DemodulatorThread::run() cannot push ati into audioOutputQueue, is full !"

vsonnier, 2017-08-03:
I @dunaybd , I assume you are the Dunay from the SDRPlay Facebook group :). Did you finally solved your Bookmark problems, or are they still present ?
BTW, the message ``DemodulatorThread::run() cannot push ati into audioOutputQueue, is full !`` you see has nothing to do with bookmarks. It says that the Audio part has trouble processing samples because of lack of CPU power and also shows up at (big) Session loads, or while closing the application.   
On the other hand, if you see it popping up continuously it means that something is wrong.

vsonnier, 2017-08-03:
@chpalm Hello ! You described the following bug 'Copy / Paste not working between label editing and session save dialog' affecting Bookmarks, but I don't manage to reproduce it with the latest 0.22. 
Could you detail the steps to reproduce this particular problem ?

dunaybd, 2017-08-03:
Thanks for the help! I'm Dunay from the SDRPlay Facebook group. Yes, it is a permissions problem. I'll get it fixed now.

chpalm, 2017-08-06:
vsonnier. I will try to detail the steps as soon as I can i.e. next week. I is possible that I have found similar issues with other software and thus a completely different story.

dunaybd, 2017-08-06:
I had installed the app as root. This caused my user configuration folder (~/.CubicSDR) to be owned by root. Running as user did not allow the creation of "bookmarks.xml" in ~/.CubicSDR because ~/.CubicSDR was owned by root.

I uninstalled the system version of CubicSDR and deleted the ~/.CubicSDR folder. I ran the application from the user folder, ~/CubicSDR/build/x64/CubicSDR. This recreated the ~/.CubicSDR folder with proper use permissions. 

All was well.

vsonnier, 2017-08-07:
@dunaybd Thanks for the info.

chpalm, 2017-08-07:
@dunaybd The steps (numbered) that i have problems with, when saving a bookmark and name it after the label:
1. copy (Ctrl-C) text in (eg. label field in the bookmarks menu or from a file) 
2. paste (Ctrl-P) don't work in the File/Save As field to create a named xml file 
My workaround is to 
- Write the file name in the Save As field to name the xml file. 
- Save the Untitled.xml file, open it in the finder and paste the new name.

Im running v0.22-alpha to be able to use SDRplay I have tried the same steps in 0.22 under Mac OSX Sierra with the same result. I guess it is a Mac only issue.

vsonnier, 2017-08-07:
@chpalm I don't think you responed to right person. Whatever. Why are you using Ctrl-P instead of Ctrl-V to paste ?

chpalm, 2017-08-07:
I miss spelled Ctrl-V :-)

”Worry is like a rocking chair. It gives you something to do, but it
doesn’t get you anywhere.”

vsonnier, 2017-08-12:
@cjcliffe Good day ! I've some ideas on how to handle "bookmark" files together with the existing Session files that would be both practical and powerful I think. So I self-assigned to take care of this in the future.
Don't know when I work on this though, real life issues and such.

cjcliffe, 2017-09-19:
@vsonnier sounds good to me; it's functional for now so no rush needed 👍

vsonnier, 2018-01-06:
I implemented the saving/loading of bookmark files into #596.

closed 2026-11-14 by cjcliffe