cjcliffe/CubicSDR #488: Please make demod snapping optional

Toontje, 2017-01-07:
Can you make the demodulator snapping in the waterfall optional?

cjcliffe, 2017-01-09:
@Toontje not sure what you mean exactly; you can disable snapping by right-clicking the highlighted digit on the frequency tuning bar?

Toontje, 2017-01-16:
I tried again and i understand what's happening. When i have a modem active and i want to create a second active modem i cannot create an active modem of which the center frequency falls within the bandwidth of the first one. The modem selector than "snaps" to the center frequency of the first created modem.
All clear. Please close the issue as a self-answered question. ;-)

cjcliffe, 2017-01-16:
@Toontje ahh; I mis-interpreted it as the frequency snap option.

There is actually a solution already built-in for for moving the active modem within it's own bandwidth since the first release; just double-click instead of single ;)

Will close this one up.

closed 2017-01-16 by cjcliffe

Toontje, 2017-01-17:
Thanks for the tip!