cjcliffe/CubicSDR #489: Confused about what Center does.

Toontje, 2017-01-08:
See the following video: https://www.dropbox.com/s/svjrmpy8nb19mv9/40m%20CW.mov?dl=0

I start CubicSDR. I select my SDR device, in my case the SDRPlay.
Then you see that it shows the 40m band. There is no demodulator selected, i just see the full 40m band and more.
Then i select the CW part of the band. All signals disappear when i'm sure that there are stations active in this band.
Then i press the C(enter) key and all signals appear.

So my question is: "What does the Center key actually do and why don't i see any signals on the band before i press the Center key?"

cjcliffe, 2017-01-09:
@Toontje center sets the center frequency; center frequency is the actual frequency that the SDR device is tuned to; all other frequencies are shifted to baseband before being demodulated.

When moving the center around while zoomed in you might introduce/remove aliases into the current view decimation depending on where the center frequency lands relative to the current view -- that's likely what you're seeing appear/disappear.   The center frequency can also affect the strength of the signals since the further away from baseband you are the signal will weaken a bit.

Generally if I'm using LSB dominant band I set the center frequency to the right edge; and for USB bands I set center to the left edge of the band; but sometimes I'll just tap 'C' around in the zoomed view to move aliases out of the way or to determine if that's what they are.

It's been suggested on certain hardware you can improve some aliasing just by setting the center frequency to exact MHz values like 3Mhz, 6Mhz, 9Mhz (no decimals)

Toontje, 2017-01-16:
> he center frequency can also affect the strength of the signals since the further away from baseband you are the signal will weaken a bit.
This is most probably what's happening, but not a bit. When i select a different band to view i hardly see any signals at all in my waterfall (see video) until i press the Center key.
I can see that sample rate can also have a huge effect on the reported signal strength in the waterfall. Having the same range in my waterfall (a couple of 100kHz) and switching Sample Rates using the menu gives me something between an very active and almost overloaded waterfall, when using a low sample rate, and almost no identifiable signals when using a high sample rate.
And BTW, sometimes switching sample rates i don't get any waterfall. Maybe this can help you troubleshoot https://github.com/cjcliffe/CubicSDR/issues/467

cjcliffe, 2017-01-16:
@Toontje interesting; might be related to the new SDRPlay module; will have to do some more experiments to see if I can reproduce it in a way I'm not expecting.

closed 2017-02-06 by cjcliffe