cjcliffe/CubicSDR #492: Strange UI artifacts when utilizing AppImage on Ubuntu

corq, 2017-01-09:
Image here, looks almost like development webform outlines? Hard to describe. My compiled version appears normal, this appears to only exist while opening 
 the appimage. Everything here appears functional, however when navigating menus, the net result is that submenus become difficult to read.

[image]

Thanks!

cjcliffe, 2017-01-09:
@corq that menu does look really strange; what's your Ubuntu release and what Window Decorator / Manager do you use?

nmaster2042, 2017-03-27:
Hi,

I just installed the CubicSDR-0.2.1-x86_64.AppImage on my xubuntu 16.10 and I have the same issue.

[image]

corq, 2017-03-28:
Same, Xubuntu 16.10/xfce. I've since rebuilt from GitHub the current
cubicsdr standard version with no issue, but I could try the appimage again
to verify if needed.

stuzog, 2017-03-28:
I don't get that on Linux Mint 18.1 with v.0.2.1-bookmarks1 installed by AppImage.

corq, 2017-03-29:
​​
Maybe only XFCE related then, my version is XFCE 4.12 : - this is from the
AppImage only, the regular build looks fine.

[image: Inline image 1]

nmaster2042, 2017-03-29:
Hi 

I tried CubicSDR AppImage on a other computer running kubuntu 16.10 and I get the same issue.

When I start the app, I get these warnings:

`
nicolas@xpsone:~/Téléchargements$ ./CubicSDR-0.2.1-x86_64.AppImage 
Loaded 216 rig models via hamlib.

(CubicSDR-0.2.1-x86_64.AppImage:25790): Gtk-WARNING **: Theme parsing error: gtk.css:104:2: Missing name of pseudo-class

(CubicSDR-0.2.1-x86_64.AppImage:25790): Gtk-WARNING **: Theme parsing error: gtk.css:116:21: Missing name of pseudo-class

(CubicSDR-0.2.1-x86_64.AppImage:25790): Gtk-WARNING **: Theme parsing error: gtk.css:142:20: Not a valid image

(CubicSDR-0.2.1-x86_64.AppImage:25790): Gtk-WARNING **: Theme parsing error: gtk.css:184:6: Missing name of pseudo-class

(CubicSDR-0.2.1-x86_64.AppImage:25790): Gtk-WARNING **: Theme parsing error: gtk.css:186:8: Missing name of pseudo-class

(CubicSDR-0.2.1-x86_64.AppImage:25790): Gtk-WARNING **: Theme parsing error: gtk.css:231:67: Missing name of pseudo-class

(CubicSDR-0.2.1-x86_64.AppImage:25790): Gtk-WARNING **: Theme parsing error: gtk.css:239:81: Missing name of pseudo-class

(CubicSDR-0.2.1-x86_64.AppImage:25790): Gtk-WARNING **: Theme parsing error: gtk.css:246:112: Missing name of pseudo-class

(CubicSDR-0.2.1-x86_64.AppImage:25790): Gtk-WARNING **: Theme parsing error: gtk.css:257:67: Missing name of pseudo-class

(CubicSDR-0.2.1-x86_64.AppImage:25790): Gtk-WARNING **: Theme parsing error: gtk.css:268:69: Missing name of pseudo-class

(CubicSDR-0.2.1-x86_64.AppImage:25790): Gtk-WARNING **: Theme parsing error: gtk.css:272:18: Missing name of pseudo-class

(CubicSDR-0.2.1-x86_64.AppImage:25790): Gtk-WARNING **: Theme parsing error: gtk.css:281:114: Missing name of pseudo-class

(CubicSDR-0.2.1-x86_64.AppImage:25790): Gtk-WARNING **: Theme parsing error: gtk.css:284:26: Missing name of pseudo-class

(CubicSDR-0.2.1-x86_64.AppImage:25790): Gtk-WARNING **: Theme parsing error: gtk.css:330:37: Missing name of pseudo-class                                                                                                        
                                                                                                                                                                                                                                 
(CubicSDR-0.2.1-x86_64.AppImage:25790): Gtk-WARNING **: Theme parsing error: gtk.css:339:95: Missing name of pseudo-class                                                                                                        
                                                                                                                                                                                                                                 
(CubicSDR-0.2.1-x86_64.AppImage:25790): Gtk-WARNING **: Theme parsing error: gtk.css:346:58: Missing name of pseudo-class                                                                                                        
                                                                                                                                                                                                                                 
(CubicSDR-0.2.1-x86_64.AppImage:25790): Gtk-WARNING **: Theme parsing error: gtk.css:358:70: Missing name of pseudo-class                                                                                                        
                                                                                                                                                                                                                                 
(CubicSDR-0.2.1-x86_64.AppImage:25790): Gtk-WARNING **: Theme parsing error: gtk.css:363:46: Missing name of pseudo-class                                                                                                        
                                                                                                                                                                                                                                 
(CubicSDR-0.2.1-x86_64.AppImage:25790): Gtk-WARNING **: Theme parsing error: gtk.css:368:48: Missing name of pseudo-class                                                                                                        
                                                                                                                                                                                                                                 
(CubicSDR-0.2.1-x86_64.AppImage:25790): Gtk-WARNING **: Theme parsing error: gtk.css:371:46: Missing name of pseudo-class                                                                                                        
                                                                                                                                                                                                                                 
(CubicSDR-0.2.1-x86_64.AppImage:25790): Gtk-WARNING **: Theme parsing error: gtk.css:377:48: Missing name of pseudo-class                                                                                                        
                                                                                                                                                                                                                                 
[... cut ...]
  default_input = True
  default_output = True
  device_id = 0
  driver = audio
  label = PulseAudio
Make device 0
  device_id=0
  origin=https://github.com/pothosware/SoapyAudio

Reporting enumeration complete.
SDR enumerator done.
Loaded font 'Bitstream Vera Sans Mono' from '/tmp/.mount_h3wGaF/usr/bin/fonts/vera_sans_mono24_0.png', parsed 255 characters.
Loaded font 'Bitstream Vera Sans Mono' from '/tmp/.mount_h3wGaF/usr/bin/fonts/vera_sans_mono27_0.png', parsed 255 characters.
`
Hope this will help.

cjcliffe, 2017-03-31:
Hmm; I'm thinking this is related to building on Ubuntu 14 and the GTK libs included from there; will try building on a newer version alongside the next release and see if it helps.

vsonnier, 2018-02-02:
@corq Good day. I know this issue is old, so sorry to wake you up:) 
With the recently released CubicSDR 0.23 @cjcliffe made a new Applimage: https://github.com/cjcliffe/CubicSDR/releases/download/0.2.3/CubicSDR-0.2.3-x86_64.AppImage
Could you please try it so we can close the issue, or not.

stuzog, 2018-02-02:
CubicSDR-0.2.3-Darwin crashes immediately on launch in OS X 10.11.6 iMac 7,1. Icon is in the Dock but only the CubicSDR menu item appears but is unresponsive. Had to quit using Activity Monitor.

nmaster2042, 2018-02-04:
Hi, tried the last appImage on a kubuntu 17.10, I get:

./CubicSDR-0.2.3-x86_64.AppImage: symbol lookup error: /usr/lib/x86_64-linux-gnu/libpangoft2-1.0.so.0: undefined symbol: hb_font_funcs_set_variation_glyph_func

I don't have kubuntu 16.10 anymore to try.

corq, 2018-02-05:
> 2.3/CubicSDR-0.2.3-x86_64.AppImage

Actually I need to update my local copy of cubicsdr on this machine, I'll
give it a spin this evening!

corq, 2018-02-05:
Hi All -

Actually get same result as nmaster2042:

$./CubicSDR-0.2.3-x86_64.AppImage

./CubicSDR-0.2.3-x86_64.AppImage: symbol lookup error:
/usr/lib/x86_64-linux-gnu/libpangoft2-1.0.so.0: undefined symbol:
hb_font_funcs_set_variation_glyph_func

XFCE version 4.12 - about to switch over to Gnome on my other box and see
if it persists there.

jhunolt, 2018-02-27:
Hi,
I have two Ubuntu systems, one (Ubuntu 17.10 4.14.0 Plasma) where I get the same error as corq, and the other (Skywave 16.04 4.13.4XFCE) where I do not.  The Skywave one is where my dongle is connected and I access it remotely from the Ubuntu box.  I even tried using the remote soapy connection on the Skywave box to try and replicate what I am doing in Ubuntu and it still works fine.

The other thing I noticed is that on the Ubuntu box libpangoft2 is version 1.40.12-1 and on Skywave it is 1.38.1-1.

If there is any other info I can provide that might help let me know.

Jon

vsonnier, 2018-03-23:
Hello, @jhunolt @corq @nmaster2042 @stuzog and all. I'm closing this issue as a (older) duplicate of the current #611, which tries to gather all the current ApplImage issues. Please continue the discussion there, thanks !

closed 2018-03-23 by vsonnier