cjcliffe/CubicSDR #498: Problem with CubicSDR

sangar233, 2017-01-18:
I have a problem with CubicSDR, I installed the program with one USB RTL-SDR Receiver NooElec NESDR Smart and I would say that works very well.
I have only one problem in the visualization of text fonts.
It appears across a series of graphic's fonts in place of the frequencies and other data as you can see in the attached file. The installation under Windows 7, he has had only a failure to install the C ** because he says that I have a latest version. the rest of the installation is ok.
I ask help about
Thank you in advance
Ferdinando
[image]

cjcliffe, 2017-01-19:
Hi @sangar233 -- can you open a command prompt and try running CubicSDR like this:

```
> cd "\Program Files\CubicSDR\"
> CubicSDR.exe > output_log.txt
```

Then exit the program and drag+drop the resulting output_log.txt file in the C:\Program Files\CubicSDR into your comment response on GitHub here to attach it.

It seems strange that some fonts have loaded but others have not and I'm interested to see the startup log for CubicSDR -- which isn't exposed unless you pipe it into a file from the command prompt.

Note:
- Use WindowsKey+R to open the Run dialog and type 'cmd' to launch the command prompt.
- You may need to use '\Program Files (x86)\CubicSDR' if you installed the 32-bit version on a 64-bit system.

sangar233, 2017-01-19:
Hi Cjcliffe,

Thanks for your prompt reply, I send you the file log.

My system is a portable, model: Samsung Q45

Intel Core Duo CPU T7250 2 GHz

2 GB Ram

System 32 bit and O.S.: Window7 professional Service Pack 1

SDR installed in Interface 0 and 1 as Generic RTL 2832U OEM:: 00000001 (see the attached)

Yesterday I uninstalled Your software and my latest version of C ++ and I installed again  your software (CubicSDR 0.2.0 - 32 Bit) with C ++ until the end, however, the problem remains.
Looking at the computer, I realized that there are different versions of C ++ (see the attached)
I think it is best to uninstall the old versions and just leave the last.
With this last installation I note that the program does not close normally, It remains waiting, I have to force closure of it.

Then today also update the Windows 7 Operating System.

Many thanks in advance

Ferdinando

I3BJP

A: cjcliffe/CubicSDR
Cc: sangar233; Mention

 

Questa e-mail è stata controllata per individuare virus con Avast antivirus.
https://www.avast.com/antivirus

Loading:: configuration file 'C:\Users\titportatile\AppData\Roaming\cubicsdr\config.xml'
Loaded 199 rig models via hamlib.

Audio Device #0 Default Device
	Default Output? Yes
	Default Input? Yes
	Input channels: 2
	Output channels: 2
	Duplex channels: 2
	Native formats:
		8-bit signed integer.
		16-bit signed integer.
	Supported sample rates:
		4000hz
		5512hz
		8000hz
		9600hz
		11025hz
		16000hz
		22050hz
		32000hz
		44100hz
		48000hz
		88200hz
		96000hz
		176400hz
		192000hz

Audio Device #1 Altoparlanti (Dispositivo High Definition Audio)
	Default Output? No
	Default Input? No
	Input channels: 0
	Output channels: 2
	Duplex channels: 0
	Native formats:
[... 176 lines cut ...]
Warning: ReBuffer 'SDRPostThreadVisualDataBuffers' count '103' exceeds threshold of '100'
Warning: ReBuffer 'SDRPostThreadVisualDataBuffers' count '104' exceeds threshold of '100'
Warning: ReBuffer 'SDRPostThreadVisualDataBuffers' count '105' exceeds threshold of '100'
Warning: ReBuffer 'SDRPostThreadVisualDataBuffers' count '106' exceeds threshold of '100'
Warning: ReBuffer 'SDRPostThreadVisualDataBuffers' count '107' exceeds threshold of '100'
Warning: ReBuffer 'SDRPostThreadVisualDataBuffers' count '108' exceeds threshold of '100'
Warning: ReBuffer 'SDRPostThreadVisualDataBuffers' count '109' exceeds threshold of '100'
Warning: ReBuffer 'SDRPostThreadVisualDataBuffers' count '110' exceeds threshold of '100'
Warning: ReBuffer 'SDRPostThreadVisualDataBuffers' count '111' exceeds threshold of '100'
Warning: ReBuffer 'SDRPostThreadVisualDataBuffers' count '112' exceeds threshold of '100'
Warning: ReBuffer 'SDRPostThreadVisualDataBuffers' count '113' exceeds threshold of '100'
Warning: ReBuffer 'SDRPostThreadVisualDataBuffers' count '114' exceeds threshold of '100'
Warning: ReBuffer 'SDRPostThreadVisualDataBuffers' count '115' exceeds threshold of '100'
Warning: ReBuffer 'SDRPostThreadVisualDataBuffers' count '116' exceeds threshold of '100'
Warning: ReBuffer 'SDRPostThreadVisualDataBuffers' count '117' exceeds threshold of '100'
Warning: ReBuffer 'SDRPostThreadVisualDataBuffers' count '118' exceeds threshold of '100'
Warning: ReBuffer 'SDRPostThreadVisualDataBuffers' count '119' exceeds threshold of '100'
Warning: ReBuffer 'SDRPostThreadVisualDataBuffers' count '120' exceeds threshold of '100'
Warning: ReBuffer 'SDRPostThreadVisualDataBuffers' count '121' exceeds threshold of '100'
Warning: ReBuffer 'SDRPostThreadVisualDataBuffers' count '122' exceeds threshold of '100'
Warning: ReBuffer 'SDRPostThreadVisualDataBuffers' count '123' exceeds threshold of '100'
Warning: ReBuffer 'SDRPostThreadVisualDataBuffers' count '124' exceeds threshold of '100'
Warning: ReBuffer 'SDRPostThreadVisualDataBuffers' count '125' exceeds threshold of '100'
Warning: ReBuffer 'SDRPostThreadVisualDataBuffers' count '126' exceeds threshold of '100'
Warning: ReBuffer 'SDRPostThreadVisualDataBuffers' count '127' exceeds threshold of '100'

sangar233, 2017-01-19:
Hi Cjcliffe,

After the update of  the Windows 7 at January 2017 the software closes normally but  the problem remains.
I send you the last resulting output_log.txt file

Ferdinando

I3BJP

 

https://www.avast.com/antivirus

Terminating SDR thread..
readLoop() ended.
device deinit()
SDR thread done.
Terminating SDR post-processing thread..
Terminating All Demodulators..
Terminating Visual Processor threads..

sangar233, 2017-01-20:
Hi Cjcliffe,

I would like to update on the situation.

After several tests on this laptop and other machines, I found that it was a problem connected to the video card Intel 965- Express chipset,  I updated it with the last driver.

The situation has changed completely. Everything works fine. I could think about it first

We are sorry for the time spent, compliments for the software

Thank you for your availability and your prompt reply.

 

Ferdinando

I3BJP

 

https://www.avast.com/antivirus

cjcliffe, 2017-01-24:
@sangar233 thanks for the update; I was thinking it might be intel graphics related; there's already a hack in CubicSDR to get around Intel graphics rendering the wrong texture orientation for the waterfall on some systems..  Cheers!

closed 2017-01-24 by cjcliffe

sangar233, 2017-01-25:
hI

It was sufficient to update the driver video with it described in annex

Thank you

Ferdy

I3BJP

 

https://www.avast.com/antivirus