cjcliffe/CubicSDR #505: Graphic Glitch 0.2.1

lantren, 2017-01-30:
I'll attach a screen capture since it's easier if you see it.  
The demodulation waterfall incorrectly displays the main waterfall or put another way, signals in the main waterfall are displayed in the wrong location in the demodulation waterfall. 

In this example, there is no signal within the active demodulator , but an adjacent signal appears in the demodulator waterfall, overlapping with the active demodulator. 
[image]

vsonnier, 2017-01-30:
@lantren Hello ! If you change your sample rate, or youy zoom level, and the problem desappears it means you likely hit the "phantom" demodulator issue #338.
@cjclife I wish I could help with this, but l'm not competent enough.

lantren, 2017-01-30:
I see.  I didn't notice this in 0.2.0, i did how ever see #338. Thought it was new to 0.2.1 since I just started using it a few hours ago.  What i'm seeing is definitely independent of sample rate or zoom level, the artifact seems remain unaltered by either of those two parameters.

cjcliffe, 2017-01-31:
@lantren @vsonnier for the demod waterfall issue (#295) the baseband-shifted image wrapped from the limited bandwidth polyphase filter bank channel the demod lies within is the source -- and is mostly just a visual artifact unless your channel edge rests on a signal and you want a signal on the other edge. The bias in @lantren 's demod waterfall image looks like that signal's been partially pulled from the opposing channel edge.  

I hope to provide some visual cue you can enable in the display menu to better visualize any distortion or artifacts from the channels as it's a rather complicated issue to describe otherwise.  Eventually it should be possible to automatically adjust the channel placement / size to keep active modems within a best-fit of the channels.

The main waterfall aliasing is a bigger issue that I should be able to solve with some more work to the visual processor code, and how it handles decimation and filtering for increasing zoom levels.   This would require additional CPU for DSP work and would be made optional via settings.

closed 2017-02-06 by cjcliffe