cjcliffe/CubicSDR #506: offset recall.

lantren, 2017-01-30:
I use a couple of different USB dongles with an up converter, and I switch back and for between the specific offset for each dongle and no offset. The offset is not _exactly_ -125000000, more like -124996800 for one dongle and -124998100 for the other. Being able to save offsets for later recall would be really handy.

cjcliffe, 2017-01-31:
@lantren the offset should be stored per-device right now; as long as you've provided a unique serial number for each device with the rtl_eeprom tool so that CubicSDR can recognize them -- this isn't often done in the factory for the ~$10 dongles so they all look like the same device initially.

Just plug in the dongle you want to re-serial and run something like:
```rtl_eeprom -s 0002```
To set it to serial 2 with some prefix space

Serial string shouldn't be limited to numbers though; so you can put more meaningful identifiers:
```rtl_eeprom -s HAMITUP:01```

Hope that helps!

lantren, 2017-01-31:
@cjcliffe That's very useful information, thank you. I can now at least set the offset for one dongle since I usually only use that one with the Hamitup. The other however I switch from -124996800 to 0 offset, not a big deal since I can just type it in at the device selection or in program.

cjcliffe, 2017-01-31:
@lantren good to hear; they're cheap enough I usually keep one hooked up to each up-converter and antenna and name them by their connection; it's worth the convenience and the configuration carries over to remoted devices too.

Going to close this one up for now and add a reminder to add this information to the docs; cheers!

closed 2017-01-31 by cjcliffe