cjcliffe/CubicSDR #508: SDRplay RSP2 Hi-Z Antenna, Not showing up in Select Antenna Drop Down in CubicSDR

patelkes, 2017-02-01:
I am using SDRPlay RSP2 and CubicSDR 0.2 version on Mac OSX 10 having SDRPlay Access using SoapySDRRemote which is running on Rasp PI 3.

I have installed SDRplay_RSP_API-RPi-2.09.1.run on Raspberry PI 3 where SoapySDRServer is running.

I have also installed latest SoapySDRPlay device library on Raspberry PI 3.

When I run CubicSDR on Mac Antenna Selection Drop Down shows up only 2 Antennas. Antenna A a Antenna B.
### **It does not show up Hi-Z Antenna.**

While using SDR-Radio Version 3 Preview 5 on Windows with SDRplay Latest API Driver ver 2.8. (shows up 3 Antenna.

Can you Please Fix up ASAP as I can not used Hi-Z Antenna in Cubic SDR as my Antenna is a PLAIN WIRE HF ANTENNA

Thank you for your time.

Regards

patelkes

SDRplay, 2017-02-01:
I've messaged you directly, this is for the github issue paper trail...

The way it's been implemented is that there are 2 controls...

Antenna selects between Ports A and B
AMport selects between the Antenna selection above (i.e. either A or B) OR the HiZ port

Hope that clarifies it for you.

Best regards,

SDRplay Support

patelkes, 2017-02-01:
Hi 

>>AMport selects between the Antenna selection above (i.e. either A or B) OR the HiZ port

Unfortunately  it is not working this way. By selecting Hi-z Port in AM Port Selection and having LSB mode on 40 Meters it does use my LONG WIRE ANTENNA. I tested this before.

Same thing I  tested using SDR-Console version 3.0 preview 5 on windows and it does pick up Hi-Z port and you can see chnage in signl strenth and you can listen to 40 Meters having LSB mode.

Please Check your self with LONG WIRE ANTENNA in HI-Z port and test your self..

Regards

Patelkes

patelkes, 2017-02-01:
Hi SDRplay Support,

I have attached 2 screen dumps. 

One from HDSDR version 2.75 downloaded from your website  (http://www.sdrplay.com/software/SDRplay_HDSDR_Installer_2.75_beta3.exe) .

Second one from SDR-CONSOLE.com version 3 preview 5 using your Exit_IO_RSP2

In Both cases Antenna selection drop down shows 3 Antennas to select from. When you select Hi-Z from this list,  IT DOES WORK...

This will help you to identify difference between Library used by Raspberry PI 3 /Mac OSX 10 and Windows and Identify issue.

Best Regards

Patelkes
Sydney
Australia
[image]
[image]

SDRplay, 2017-02-01:
Not sure I understand the issue. I've just started up CubicSDR, selected HiZ fom the AMport pulldown in the device selection window, connected my long wire to the P node of the HiZ connector and the display looks fine. In your post above you say it DOES work - so I'm a bit confused.

Can you explain what versions of CubicSDR, SoapySDR, SoapySDRPlay and the API you have?

patelkes, 2017-02-01:
Hi SDRPlay Support,

I wants to say that by selecting Hi-Z in SDR-Console and HDSDR  works fine with Long Wire Antenna but same Hi-Z DOES NOT work in CubicSDR when you select AM port as a Hi -Z 

My CubicSDR version is 0.2.1 bookmark1 and My  Mac OSX is 10.6 
SDRPlay API installed on Mac osx client is  SDRplay_RSP_API-MAC-OSX RPi-2.09.1.pkg and 
SDRPlay API installed on Raspbery Pi 3 is  SDRplay_RSP_API-RPi-2.09.1.run
I am using latest SopaySDR server downloaded from github on Raspbery PI 3.

If you need further info please let me know.

Thank you for your time.

Regards

Patelkes

SDRplay, 2017-02-01:
ok I think we need to separate out the issues. I've tested the HiZ port selection in Linux with the latest versions of everything on a device connected LOCALLY and it all works.

I haven't tried SoapyRemote for a while so it may be possible that the issue is there.

Can you confirm that you can get the HiZ working on Mac with the RSP connected locally.

I'll try to get SoapyRemote setup to replicate your setup.

patelkes, 2017-02-01:
Hi SDRPlay Support

>>Can you confirm that you can get the HiZ working on Mac with the RSP connected locally.

I did tested connecting SDRPlay RSP2  localy on my Mac and result is same which is BY selecting AM port as Hi-Z it DOES NOT pick up signals from LONG Wire Antena.

Whether you use RSP2 Locally or use Sopayremote result is same Hi-Z DOES NOT Work...

Regards

Patelkes

SDRplay, 2017-02-01:
ok, maybe we need some inputs from others, because it's working fine here. Not sure how to proceed. Suggestions? How do you have your wire connected to the HiZ port?

patelkes, 2017-02-01:
Hi SDRPlaySupport

One think I can not understand which is why it is implemented differently in CUBICSDR then two other application like SDR-Console and HDSDR. 

In above two windows application Hi-Z has been implemented  as 3rd Antenna while in CubicSDR it as been implemented as AM Port Selection?

In reality I understand that Hi-Z is a real 3rd antenna where you do not need to use stupid Baluns and for the same reason, I purchased RSP2..

Regards

patelkes

patelkes, 2017-02-01:
Hi I have connected wire correctly. when you hold rsp2 in your hand Hi-Z port is on left. In first left long wire Antenna goes and in middle I have a earth wire and last right is blank.

Same HI-Z setting works for SDR-CONSOLE and HDSDR so my wiring is correct Just I am moving my USB A wires to Windows from Mac and not touching Hi-Z port.

Regards

patelkes

SDRplay, 2017-02-01:
Looking at the RSP2 with the HiZ port on the left, the pins are (from left to right) - P N Gnd

Here's a guide: http://www.sdrplay.com/wp-content/uploads/2016/12/161201HighZPortGuidev2.pdf

In code, the SoapySDRPlay implementation is actually how it's done in the API. I simplified it in SDRuno as Simon did in SDR-Console, but the way it works in the API is one selector for the A/B ports and one for either A/B or HiZ - they use different hardware ports on the tuner so that's why it was presented that way in the API.

I'll ask some other CubicSDR users to test it and report back. It's hard to fix something that's working here.

I'll let you know what others report back to me with.

SDRplay, 2017-02-01:
[image]

Had this from our US support team. This is on a Mac. I'll check with some more people.

patelkes, 2017-02-01:
Hi 
Please get it checked for HI-Z Am port for frequency 7.000 Mhz to 7.300 (40 Meter) with LSB mode. You want find any signals there. MW signal can be seen even A/B port not having any Antenna.
Regards
patelkes

SDRplay, 2017-02-01:
[image]

HiZ port, LSB, 7MHz - I can remotely connect to your machine via TeamViewer if you would like to do some debug?

SDRplay, 2017-02-01:
what has just been reported to me is that switching to the HiZ port once CubicSDR is running is not reliable. However, setting HiZ port in the Start device selection window always works. Is this what you're seeing?

patelkes, 2017-02-02:
Hi SDRPlaySupport,

I confirm following behaviour of RSP2 on my Mac/OSX with Cubicsdr 0.2.1 Bookmark1 Edition

(A) In a Local Mode when RSP2 Is connected to MAC USB Directly.
==============================================
1. Selecting AM Port On Start-up Dialog as HI-Z
-----------------------------------------------
Hi-Z works most of the time for 7.00 Mhz LSB mode 

 2. Selecting AM Port From Drop Down Menu List as HI-Z
---------------------------------------------------------
Hi-Z works NEVER Worked for 7.00 Mhz LSB mode 

(B) Connecting RSP2 to SoapyRemoteServer
=================================
1. Selecting AM Port On Start-up Dialog as HI-Z
-----------------------------------------------
Hi-Z works NEVER Worked for 7.00 Mhz LSB mode  

2. Selecting AM Port From Drop Down Menu List as HI-Z
---------------------------------------------------------
Hi-Z works NEVER Worked for 7.00 Mhz LSB mode 

On summary Hi-Z works only from Startup screen selection of AM Port as Hi-Z in all other mode it DOES NOT work.

I am ready for TeamViewer session. My Local phone in Australia is +612 9896 2248. Please Email from SDRPLay Support to me and then give me a ring and I will provide you ID and Password for remote session on my Mac.

Thank you for Helping me for resolving the issue.

Best Regards

Patelkes

patelkes, 2017-02-03:
Hi All

On extensive testing I found that when we select AM port Select as Hi-Z on device startup screen Hi-Z is working in both LOCAL & SOAPYRemote mode but due to **MW Filter active on AM Port**, you can ONLY listen 600 Khz to 1.2 Mhz.                                                                      

Any RSP2 and CubicSDR user Can confirm the same? 

Regards

Patelkes

SDRplay, 2017-02-03:
The broadcast notch filter only works on ports A and B

rimb05, 2017-02-10:
I'm also seeing the same issues here.  HI-Z is not working at all.

SDRplay, 2017-02-10:
I think we've established that it works if you select it from the device selection window, but doesn't seem to be switchable during playback. The images I posted earlier in this thread show it working on a Mac.

dunaybd, 2017-08-06:
Changing the AMport setting for CubicSDR with SDRplay RSP2 does not actually take effect until the SDR has been stopped and started. Consistently observed with CubicSDR on Linux Mint 18.2.

vsonnier, 2017-08-14:
Hello @patelkes, @SDRplay, @dunaybd  and all. 
Since I acquired a RSP2 recently I quickly encountered the same problem described here, this time on Windows 7 x64.

I think I solved it by patching the SoapySDRPlay module code and so I submitted a [PR](https://github.com/pothosware/SoapySDRPlay/pull/30).

Could you please give a try of this change on your respective platforms ? 
It can be found on my branch https://github.com/vsonnier/SoapySDRPlay/tree/vso_RSP2_AMPort_dynamic_switch_fix
or you can simply cut-n-paste the code change described in the PR into a SoapySDRPlay master.

dunaybd, 2017-08-15:
Works for me in Linux Mint 18.2 (based on Ubuntu 16.04). Much appreciated.

vsonnier, 2017-08-18:
@dunaybd Sorry to bother you again. I've been in contact with SDRplay Support, and we agreed on the right piece of code, leading to a slightly different modification. 
Could you please try the new change described in the [PR](https://github.com/pothosware/SoapySDRPlay/pull/30)? 
I would love to see it confirmed on Linux (again), so I have arguments to make the PR accepted by the SoapySDR maintainer.

dunaybd, 2017-08-18:
With today's new code, AMport works as expected. Antenna Select (A/B) also works as expected. 

Thanks for the hard work!

vsonnier, 2017-08-18:
@dunaybd Thanks to you for your input ! I wish I get some feedback from Mac users too, but the issue has been clearly an logical / SDRplay API usage so the change should be platform-independent.

vsonnier, 2017-08-19:
The PR I submitted has just been merged into the SoapySDRPlay master. I can close this issue now.

closed 2017-08-19 by vsonnier