cjcliffe/CubicSDR #513: CubicSDR wtih LimeSDR and USB2

Analias, 2017-02-11:
Latest build from github as of 02/11/2017 0900 MST, on Ubuntu 16.10 with the latest SoapySDR from the same time.

Everything compiles and links. When I try to run CubicSDR I am shown the expected SDR Devices screen, with the LimeSDR being shown. When I select the LimeSDR and access the Sample Rate drop-down I am only giving 100Khz and 60Mhz. I need 10Mhz or 12Mhz for USB2 (both confirmed working with LimeSuite and GQRX). The two default values are very inappropriate for the LimeSDR on USB2 - 60Mhz would be fine with USB3. Unfortunately, My two LimeSDR boards are not recognized by Linux or Windows when plugged into the USB3 bus.

-Freeman, N5FPP

[image]

guruofquality, 2017-02-11:
A few notes so maybe we can come up with a better solution. The LimeSDR has a very wide open number of possible sample rates, this is achieved with a programmable clock PLL and a power of two decimation chain the lms7002m.

Current we return a high and low value for the sample rate rate, because otherwise we would return millions of possible values. However if a specific clock rate is set, the same function returns the specific enumerated sample rate options (power of two decimations):

* Reference: https://github.com/myriadrf/LimeSuite/blob/master/SoapyLMS7/Settings.cpp#L765

What would be better for limesdr under cubic sdr? Present it as a device that needs a master clock selected and an enumerated set of sample rates. Or how could we better present a range of rates that has very small granularity.

drelectro, 2017-02-18:
@Analias I had the same 'Issue' with the LimeSDR myself.

I then realized that it is possible to set the sample rate to whatever you need by clicking 'Sample Rate'->'Manual Entry' 

So just start with 100KHz and change it

martywittrock, 2017-03-22:
I'm having an issue with my Windows box where when I try to do a search for SoapySDR devices, there are no entries found for the LimeSDR at all with my Lime plugged into the USB 3.0 port and functioning fine, just prior, with SDRConsole V3.0. So then I tried one of my HF Upconverted RTL-SDRs and it found it fine on USB 3.0 and CubicSDR played great with it. But then I went back to my LimeSDR and CubicSDR doesn't see it. All my drivers are fine because I can use the LimeSDR with GQRX for Windows. too. 

What's happening here...?  de Marty, KN0CK

guruofquality, 2017-04-14:
I added an API call to query a list of ranges for SoapySDR Devices: https://github.com/pothosware/SoapySDR/issues/124

Much like the current bandwidth and frequency ranges, and the gr-osmosdr API, the getSampleRateRange() call returns a list of [min, max] values that represent the possible sample rates. When its a continuous range, there will be a single `[[min, max]]` entry in the list. When its an array of finite enumerated rates it will look like `[[rate0, rate0], [rate1, rate1], etc...]`

I'm updating the various drivers to support this where needed. By default, the list of ranges is populated from the listSampleRates() so switching over to it should not leave any gaps in existing drivers. Consider using in CubicSDR in place of listSampleRates() with a more apt GUI selection widget when its a range vs a list of discrete values. The call can be detected with `#ifdef SOAPY_SDR_API_HAS_GET_SAMPLE_RATE_RANGE` for backwards compat

martywittrock, 2017-04-14:
Josh,

Does this mean that there will be a new spin of the Pothosware install for Lime? Please let us know,

Thanks,

73 de Marty, KN0CK

cjcliffe, 2017-04-20:
@guruofquality thanks for the addition; I'll definitely use it to improve bandwidth selection.