cjcliffe/CubicSDR #514: Issues with building CubicSDR on debian: CubicSDR/src/process/VisualProcessor.h typename error

abousselmi, 2017-02-14:
Hi,
First, thank you for the good work.

I've tried to install CubicSDR on a debian machine using the install guide from: 
`https://github.com/cjcliffe/CubicSDR/wiki/Build-Linux`

Everything is fine until the step: `build$ make` under Build CubicSDR, when I got a scary error message. (I've put it in the end of this post).

Based on this stackoverflow answer: 
`http://stackoverflow.com/questions/6489351/nested-name-specifier`
I've removed `typename` keyword from `CubicSDR/src/process/VisualProcessor.h` and this build succeeded (See `git diff` in the end of the post). Still, gcc is complaining about other staff and displaying some warnings.

I dont know if I need to fork and do a pull request to have this issue resolved.
What do you think?

Thank you in advance.

Error message:
```
~/CubicSDR/CubicSDR/build$ make
Scanning dependencies of target CubicSDR
[  1%] Building CXX object CMakeFiles/CubicSDR.dir/external/rtaudio/RtAudio.cpp.o
[  2%] Building CXX object CMakeFiles/CubicSDR.dir/src/CubicSDR.cpp.o
In file included from /home/ab/CubicSDR/CubicSDR/src/process/ScopeVisualProcessor.h:6:0,
                 from /home/ab/CubicSDR/CubicSDR/src/visual/ScopeCanvas.h:13,
                 from /home/ab/CubicSDR/CubicSDR/src/AppFrame.h:15,
                 from /home/ab/CubicSDR/CubicSDR/src/CubicSDR.h:22,
                 from /home/ab/CubicSDR/CubicSDR/src/CubicSDR.cpp:17:
/home/ab/CubicSDR/CubicSDR/src/process/VisualProcessor.h:15:22: error: expected nested-name-specifier
     typedef typename ThreadQueue<InputDataType*> VisualInputQueueType;
                      ^
/home/ab/CubicSDR/CubicSDR/src/process/VisualProcessor.h:15:50: error: invalid declarator before ‘VisualInputQueueType’
     typedef typename ThreadQueue<InputDataType*> VisualInputQueueType;
                                                  ^
/home/ab/CubicSDR/CubicSDR/src/process/VisualProcessor.h:16:22: error: expected nested-name-specifier
     typedef typename ThreadQueue<OutputDataType*> VisualOutputQueueType;
                      ^
/home/ab/CubicSDR/CubicSDR/src/process/VisualProcessor.h:16:51: error: invalid declarator before ‘VisualOutputQueueType’
     typedef typename ThreadQueue<OutputDataType*> VisualOutputQueueType;
                                                   ^
/home/ab/CubicSDR/CubicSDR/src/process/VisualProcessor.h:17:35: error: ‘VisualOutputQueueType’ was not declared in this scope
     typedef typename std::vector< VisualOutputQueueType *>::iterator outputs_i;
                                   ^
/home/ab/CubicSDR/CubicSDR/src/process/VisualProcessor.h:17:58: error: template argument 1 is invalid
     typedef typename std::vector< VisualOutputQueueType *>::iterator outputs_i;
                                                          ^
/home/ab/CubicSDR/CubicSDR/src/process/VisualProcessor.h:17:58: error: template argument 2 is invalid
/home/ab/CubicSDR/CubicSDR/src/process/VisualProcessor.h:17:59: error: expected identifier before ‘::’ token
     typedef typename std::vector< VisualOutputQueueType *>::iterator outputs_i;
                                                           ^
/home/ab/CubicSDR/CubicSDR/src/process/VisualProcessor.h:17:61: error: typedef name may not be a nested-name-specifier
     typedef typename std::vector< VisualOutputQueueType *>::iterator outputs_i;
                                                             ^
/home/ab/CubicSDR/CubicSDR/src/process/VisualProcessor.h:17:61: error: expected ‘;’ at end of member declaration
/home/ab/CubicSDR/CubicSDR/src/process/VisualProcessor.h:17:70: error: ‘outputs_i’ does not name a type
     typedef typename std::vector< VisualOutputQueueType *>::iterator outputs_i;
                                                                      ^
/home/ab/CubicSDR/CubicSDR/src/process/VisualProcessor.h:52:19: error: ‘VisualInputQueueType’ has not been declared
     void setInput(VisualInputQueueType *vis_in) {
                   ^
/home/ab/CubicSDR/CubicSDR/src/process/VisualProcessor.h:60:23: error: ‘VisualOutputQueueType’ has not been declared
     void attachOutput(VisualOutputQueueType *vis_out) {
                       ^
/home/ab/CubicSDR/CubicSDR/src/process/VisualProcessor.h:67:23: error: ‘VisualOutputQueueType’ has not been declared
     void removeOutput(VisualOutputQueueType *vis_out) {
                       ^
/home/ab/CubicSDR/CubicSDR/src/process/VisualProcessor.h:116:5: error: ‘VisualInputQueueType’ does not name a type
     VisualInputQueueType *input = nullptr;
     ^
/home/ab/CubicSDR/CubicSDR/src/process/VisualProcessor.h:119:17: error: ‘VisualOutputQueueType’ was not declared in this scope
[... 110 lines cut ...]
CMakeFiles/Makefile2:60: recipe for target 'CMakeFiles/CubicSDR.dir/all' failed
make[1]: *** [CMakeFiles/CubicSDR.dir/all] Error 2
Makefile:117: recipe for target 'all' failed
make: *** [all] Error 2

```

Git diff output:
```
diff --git a/src/process/VisualProcessor.h b/src/process/VisualProcessor.h
index a385f4c..57a78d0 100644
--- a/src/process/VisualProcessor.h
+++ b/src/process/VisualProcessor.h
@@ -12,8 +12,8 @@
 template<typename InputDataType = ReferenceCounter, typename OutputDataType = ReferenceCounter>
 class VisualProcessor {
     //
-    typedef typename ThreadQueue<InputDataType*> VisualInputQueueType;
-    typedef typename ThreadQueue<OutputDataType*> VisualOutputQueueType;
+    typedef ThreadQueue<InputDataType*> VisualInputQueueType;
+    typedef ThreadQueue<OutputDataType*> VisualOutputQueueType;
     typedef typename std::vector< VisualOutputQueueType *>::iterator outputs_i;
 public:
        virtual ~VisualProcessor() {
```

vsonnier, 2017-02-14:
@abousselmi I'll take a look, and try to revert one  changes I've made on template syntax, trying your suggestion.  I work on Visual C++ 2017, but I didn't try other targets. The trick is to have a template syntax accepted by all compilers. Damn you C++ ! :-1:

mirrornoir, 2017-02-14:
I can confirm that I had the same issue as @abousselmi. His patch allows for the build to finish. I'm using GCC 6.3.1 on Arch Linux.

vsonnier, 2017-02-14:
@abousselmi , @mirrornoir Thank you for your input ! I tried @abousselmi  change, and MSVC compile and run it fine, so I've comited the change on the master branch.

closed 2017-02-14 by vsonnier

abousselmi, 2017-02-14:
@vsonnier @mirrornoir Thank you :)