cjcliffe/CubicSDR #517: Pulseaudio sinks

Wiki61, 2017-02-26:
I might be missing something, so not sure if this is a bug or not. 

Per the RTFM I found this

“Audio Out” can be configured per-modem; each modem can be routed to any available audio output.

What I would like to do is assign different modems to different null sinks that are setup with pulse audio. When I try to choose the out I only have one pulse audio device. Thanks

Wiki61, 2017-02-27:
Just verified that on Windows it works great using virtual audio cables.

cjcliffe, 2017-03-07:
@Wiki61 I believe it's been done in one of the other issues here and I think I have a text file with the instructions somewhere for doing the pulse and jack audio sinks -- will dig them up

clancy99, 2017-04-16:
t took 2 weeks, but I finally have CubicSDR v.2 running on my Raspberry pi v.2B.  It worked with the little postage stamp sized Adafruit rtl-SDR, except NO SOUND, and now I have CubicSDR talking to an SDRplay RSP2, thanks to the API/DLL kit. Now if I could JUST GET THE SOUND TO WORK! Getting tired of trying. Help!
Clancy99 ;D