cjcliffe/CubicSDR #519: Bookmark tree leaf editing crashes (Windows)

vsonnier, 2017-02-28:
@cjcliffe Hello again !  
I've tried toying a bit with the Bookmarks, as you've seen surely.

I have a problem I don't manage to workaround and solve, apparently also present in the 0.21-alpha builds.
When trying to edit a leaf of the bookmark tree (single long left-click, like a filename change in Explorer), 
it triggers
``BookmarkView::onTreeBeginLabelEdit( wxTreeEvent& event )`` then ``BookmarkView::onTreeEndLabelEdit( wxTreeEvent& event )`` as it should I suppose.

Then it crashes on executing ``onTreeEndLabelEdit( wxTreeEvent& event )`` 
because  ``m_treeView->GetEditControl()`` is NULL. 

I search here and there, browsed the wxWidgets code, to no avail.

Did you manage to make it work on you side on Windows ?
(I'm running on Win7x64, compiling on Visual Studio 2017 RC4)

vsonnier, 2017-03-04:
@cjcliffe  I've added guard code to check if ``m_treeView->GetEditControl()`` is NULL and bail out in this case, so the application no longer crashes.

cjcliffe, 2017-03-07:
@vsonnier thanks for your efforts recently; I've been away for a bit -- will take a look at this one; sounds very familiar to a bug I fixed during development on OSX side.

vsonnier, 2017-03-07:
@cjcliffe I'm glad you're back, I've started to worry you never came back... Then I realized sometimes Holidays(tm) happen :)
For me Bookmarks mostly works by now even if some other issues remain, such as:
-  More often than not, when activating a Bookmark, Active, Recent, the demod view align properly but the main Waterfall/Spectrum don't move.
-  When activating (in Active list ?) often a demod is created on top of a existing one. (thanks to OpenGL, it"s even visible on screen directly !) 

I've not searched / found simple use cases for these yet.
On the low-level side:
-   Maybe ``DataNode*`` need also some memory management fixes (I see them created, but never deleted ?)

Whatever, the Bookmarks function is a great addition for browsing the spectrum, thank you for having added it !

cjcliffe, 2017-03-15:
@vsonnier Lol, yup I'm here just slow getting started again -- got a little distracted replacing all the leaky capacitors in my Amiga 4000 but it's alive again now; A600 starting to go too so it's next :)  

After that I've gotten my Windows 10 desktop dusted off and fixed up with build environment up and running again; so I'll be using it as my primary windows dev machine instead of the VM / Bootcamp on my MacBook for the past year or so.

Just re-routing my USB hubs, cables and dongles to the various antennas around the apartment here and then I'll dig back into some issues.

cjcliffe, 2017-03-16:
@vsonnier -- I gave it a shakedown in Debug mode here in Windows 10 and it was flaky as hell -- it's redundant and I'm not really a fan of that switch block; not to mention the numerous times I've activated the thing by accident instead of when I wanted to.

If we want to revisit adding it to the context (right-click) menu that might be a better option for redundancy; there's one active context menu in there I kept as an example.

Since there's no longer leaf editing I'll consider this one closed ;)

closed 2017-03-16 by cjcliffe

vsonnier, 2017-03-16:
@cjcliffe Fine for me to remove it... like you say most of the time you enter in editing mode instead of just selecting the element. Good riddance !