cjcliffe/CubicSDR #520: Install issues for mac Serria 10.12.3

brrr, 2017-03-02:
for your instruction https://github.com/cjcliffe/CubicSDR/wiki/Build-OSX
there are 2 issues  for mac Serria 10.12.3:
1.report "qtkit.h" not found while installing wxWidgets

Resolution :get wxWidgets master git branch rather than 3.1 source code,and use '--disable-qtkit' for configuration

2.AppFrame.h extra qualification error while cpack:

diff --git a/src/AppFrame.h b/src/AppFrame.h
index 7ebb29e..eb9cc48 100644
--- a/src/AppFrame.h
+++ b/src/AppFrame.h
@@ -131,7 +131,7 @@ public:
 #endif
     //set tooltip to window
     void setStatusText(wxWindow* window, std::string statusText);
-    void AppFrame::setStatusText(std::string statusText, int value);
+    void setStatusText(std::string statusText, int value);

my gcc:

Configured with: --prefix=/Applications/Xcode.app/Contents/Developer/usr --with-gxx-include-dir=/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.12.sdk/usr/include/c++/4.2.1
Apple LLVM version 8.0.0 (clang-800.0.42.1)
Target: x86_64-apple-darwin16.4.0
Thread model: posix
InstalledDir: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin

vsonnier, 2017-03-02:
@brrr I'll leave it open for point 1.

vsonnier, 2017-03-03:
@brrr I've cut-n-paste your issue 1 and workaround in #455 which already gatehers OSX problems. I'm closing this duplicate then.

commit 5fe844b8 (2017-03-02): Should fix #520
closed 2017-03-03 by vsonnier