cjcliffe/CubicSDR #525: Bookmark Issues

cjcliffe, 2017-03-17:
From @vsonnier #519 --

-More often than not, when activating a Bookmark, Active, Recent, the demod view align properly but the main Waterfall/Spectrum don't move.
-When activating (in Active list ?) often a demod is created on top of a existing one. (thanks to OpenGL, it's even visible on screen directly !). You can even delete it, the old one then shows up.

cjcliffe, 2017-03-30:
Been tinkering; looks like some events are firing from programmatic calls causing properties to bounce back and forth as well as the selection event fires both a select and de-select event which isn't handled properly; should have this solved soon.

cjcliffe, 2017-03-31:
@vsonnier I've merged some fixes for this;  managed to boil it down to a few minor changes -- let me know if that's working any better for you as well.

vsonnier, 2017-03-31:
@cjcliffe Issue item 1 seem OK now, thanks. 
I'm now working on item 2 about the duplicated demodulators, so I'll get back to you when it is ready.

vsonnier, 2017-04-01:
@cjcliffe Hi there ! I've comitted on master c8cca67fc749242081f4c36b54f5ef40a6125c7e
which fixes item 2, plus some other cleanups around Bookmarks. Tell me what you think !

cjcliffe, 2017-04-20:
@vsonnier looks great thanks; will close this one up!

closed 2017-04-20 by cjcliffe