cjcliffe/CubicSDR #529: CubicSDR 0.2.2-alpha hangs when samplerate > 2.048MHz

acseuser, 2017-03-29:
I'm running Ubuntu 16.04, and the current version of CubicSDR, I think.

IT is connected to an SDRplay - seems to work OK: I can demodulate stuff, and I don't get warning msgs in the syslog about buffer errors. But if I increase the SampleRate to > 2.048MHz, the machine hangs; only way out is a power off reboot. I think the SDRplay is supposed to operate around 8MHz BW.

Any further info. required ?

Thanks,
SR

cjcliffe, 2017-03-31:
Haven't experienced this myself with Ubuntu 16 and RSPv1 -- I'll keep this open and tagged for 'help wanted' -- if others experience the same we might get more insight.

The only thing off the top of my head I can think to ask is whether you're using the RSP with a USB extension cable or Hub?  That's the only time I've experienced significant drop-outs and crashes with the RSP1 was when communication was poor -- if that's the case using a shorter cable or adding some ferrite beads to the USB cable might help.

acseuser, 2017-03-31:
Charles,
Nope, no hub - RSP1 directly to laptop. via a decent (I thought) USB cable. I will try another, just in case.
I can achieve 8MHz using a different, older laptop, so the RSP / cable combo can do it. Unfortunately, this laptop doesn't have a windows partition, so it's not a perfect comparison..

Thanks,
SR

ElectronicBattle, 2017-06-18:
I've had the same problem. Trying to get a locally compiled version of CubicSDR 022 or the (appimages 020 or 021) to recognise the SDRPlay RSP was only possible by adding three entries to modprobe.d/blacklist.conf, namely 
# blacklist sdr_msi3101
# blacklist msi001
# blacklist msi2500
(info from http://kb9rlw.blogspot.co.uk/2017/06/step-by-step-install-of-sdrplay-rsp2.html)

Changing the RSP's  i/p sample rate from 2048 causes my complete OS to lock up (linux Mint 18.1, Mate edition ==16.04 base) and even ctrl+alt+F1 etc. won't spawn a new session. Mains reboot needed!

The same happens for gqrx too. 

I don't know what to do to help you if you were to try to find a solution to this problem, which is clearly not peculiar to CubicSDR alone.

dunaybd, 2017-08-05:
No problem with Linux Mint 18.2 (based on Ubuntu 16.04) and RSP2.

vsonnier, 2017-10-11:
No problem either with Windows 7/10 x64 using the API v2.10 and RSP2.

stuzog, 2018-02-02:
CubicSDR-0.2.3-x86_64.AppImage launches in Linux Mint 18.2 but no signal showing, just this is Terminal:

```
SDRThread::readStream(): 2. SoapySDR read failed with code: -1
SDRThread::readStream(): 3.1 iqDataOutQueue output queue is full, discard processing of the batch...
```
repeated multiple times.

vsonnier, 2018-03-23:
Be bold and suppose the remaning issues are indeed AppImage ones, listed in #611, so I'm closing this one.

closed 2018-03-23 by vsonnier