cjcliffe/CubicSDR #530: Bookmarks - enhancement request

acseuser, 2017-03-30:
Bookmarks are great; I find that very useful!

Is it possible to add the following:

(1) SORT the entries in an alphabetical order ? I guess it can be done by massaging the XML directly, but...

(2) add further layers in the bookmarks, Eg:

AMATEUR
       --  HF
       --  VHF
       --  UHF

MARINE
       --  VHF

MISC
       -- SOMETHING
       -- SOMETHING-ELSE

Hope that's clear :-)

cjcliffe, 2017-03-31:
Thanks for the suggestions; I think some sort of automatic / right-click sorting option might be good so I'll add this to the enhancements list.

There would need to be some significant adjustments to support multiple levels of bookmark groups but it was intended to be able to do so.   Not a priority but will keep it in mind.

Toontje, 2017-08-30:
Aren't you already doing that?
[image]