cjcliffe/CubicSDR #531: Main spectrum - dB or dBm ?

acseuser, 2017-03-30:
The main spectrum displays a dB value - that would be C/N I guess?  Is it possible (is it modem specific?) to have the option to display dBm instead?

cjcliffe, 2017-03-31:
I'm not exactly sure how to show the dBm to instead of dB but it's something I would like to do; currently it's just using the visible FFT to calculate the dB.

Will add this as an enhancement.

vsonnier, 2018-02-25:
Also related to the #470 question by @vpnels I'm merging into this one:

> For the spectrum noise floor and peak dB - what is the measurement bandwidth? is it dBm/rootHz?
73,
Pascal - ac7n

righthalfplane, 2018-05-24:
I put a dB label into CubicSDR 0.2.4 and calibrated it for the NetSDR - it agrees within a db or two with  the dBm values output by CuteSDR. The relative db values work Ok with the other devices, but the absolute levels are off. I used the smeter.c routine from CubicSDR.  I do not know the program very well - so I had to take my best guess and crowbar it in. If someone would like to use what I have as a starting point to put it in correctly, I can make it available. I attached an image - the -42 dB near the top left of screen is the dBm for the NetSDR  and as best I can tell it works correctly.

Dale.
[image]

Zanoroy, 2019-03-08:
@righthalfplane 
I know this is closed but I'm trying to get  the dBm value from within CubicSDR 0.2.5 to display on the Demodumator Spectrum and was wondering how you went about it.