cjcliffe/CubicSDR #532: main spectrum window - add MIN hold ?

acseuser, 2017-03-30:
Would it be possible to have a MIN hold as well as a peak (we call it MAX) hold ?

Perhaps the following choices:
NONE, MAX HOLD, or MIN HOLD     (ONE only)
or combination of all three, perhaps using red for min hold.

Tks,

cjcliffe, 2017-03-31:
@acseuser sounds interesting; would that just be to hold the minimum noise floor value similar to the peak or is there a more specific use case you're aiming for?

acseuser, 2017-03-31:
Charles,
Thanks for the response. Well...

In the past, I have used many spectrum analysers - real 'portable'
(sort-of; SA's are rarely 'portable' ... ) and networked multi-user stuff,
that had this functionality.

Why?? looking to see if a signal goes away at all. The inverse of max hold.

Regards,
Simon R.

​

acseuser, 2017-03-31:
I might add another, slightly, related enhancement - save trace - for much the same reasons as my other suggestions. The ability to capture a trace (short of saving the raw data to disc for later replay!) would be useful.