cjcliffe/CubicSDR #534: Bounds checking issue in stream args

guruofquality, 2017-04-02:
Due to unspecified default when using a list of options, we got an index that was outside of the options size and a segfault. I'm fixing the issue in the LimeSuite, but just a note in case you want to do bounds checking here, get a warning, or safely pick element 0, rather than now it can bomb out:

* Line in question: https://github.com/cjcliffe/CubicSDR/blob/master/src/forms/SDRDevices/SDRDevices.cpp#L315

vsonnier, 2017-08-14:
Right, since the master is no longer the master, I reproduced the original code here:
````cpp
 if (dev) {
            args = dev->getSoapyDevice()->getStreamArgsInfo(SOAPY_SDR_RX, 0);
            
            if (args.size()) {
                for (args_i = args.begin(); args_i != args.end(); args_i++) {
                    SoapySDR::ArgInfo arg = (*args_i);
                    wxPGProperty *prop = streamProps[arg.key];
            
                    if (arg.type == SoapySDR::ArgInfo::STRING && arg.options.size()) {
                        streamArgs[arg.key] = arg.options[prop->GetChoiceSelection()];
                    } else if (arg.type == SoapySDR::ArgInfo::BOOL) {
                        streamArgs[arg.key] = (prop->GetValueAsString()=="True")?"true":"false";
                    } else {
                        streamArgs[arg.key] = prop->GetValueAsString();
                    }
                }
            }
        }
````

vsonnier, 2018-03-21:
This is the same as the #634, so close this old one as duplicate.

closed 2018-03-21 by vsonnier