cjcliffe/CubicSDR #535: ASCII or UNICODE?

gvanem, 2017-04-03:
I've been using the pre-built CubicSDR for a few days Then decided to build it myself. But the cmake
build-process is far from clear. I.e. the very basic requirement of *ASCII* or *UNICODE* for a Windows (32-bit) is not specified in `CMakeLists.txt` AFAICS. My generated `*.vcxproj` files does *not* specify
`-DUNICODE` and  `-D_UNICODE`. Hasn't a ASCII-build of wxWidgets been deprecated for some years?

 The same confusion goes for `-DWXUSINGDLL`;  Cmake forces that against my will.

And the fact that Cmake is obfuscating the whole build process doesn't make it easier. Is there a way to even do a *dry-run* of the build-process? (as with `make -n`).

I ended up with creating my own [GNU-makefile](https://gist.github.com/gvanem/004f7a6254ad897b5bc3ae694575eab7) which kinda works.

Can someone please advice on this?

vsonnier, 2017-04-03:
@gvanem Hello ! Have you followed the guide here to build CubicSDR: [Build Windows](https://github.com/cjcliffe/CubicSDR/wiki/Build-Windows) ? 

On my side, CMake generates the right Visual Studio solution which incidentally have _UNICODE defined as preprocessor, although I'm pretty sure it is irrelevant for Windows 7+.

As for wxWidgets, you are absolutely right, because v3.0+ are Unicode by default. So simply build it as the guide suggests, loading the pre-build Solution, and you should be fine.  

The minimum steps to compile your CubicSDR version is to follow the steps: 
- Build wxWidgets
- Install SoapySDR
- Build CubicSDR

The guide was done for Visual Studio 2015, but works of course for Visual Studio 2017. 
For the rest of the modules, you can simply use the pre-build binaries ones instead of rebuilding them as the guide suggests.

gvanem, 2017-04-03:
>  Have you followed the guide here to build CubicSDR: Build Windows ?

Yes. But for me, wxWidgets throws all kind of strange error when building with this method; `strftime() not declared` etc. etc. Besides, that Wiki does not go into much details. 

Anyway, I'm using my GNU-makefile for now. It creates CubicSDR fine, but I have some problems with the sound and the LodePNG stuff is just garbage IMHO. I.e from the audio-spectrum view:
[image]

I get these *wonderful* jagged fonts!

vsonnier, 2017-04-03:
@gvanem As for wxWidgets build, it can hardly be more straightforward than that...
So let's start with this. What is your OS, Visual Studio version ?

gvanem, 2017-04-03:
>  As for wxWidgets build, 

Well, true. I have no problems with wxWidgets. Except maybe for building it in ASCII-mode (which seems to be deprecated now).

>  What are your OS, CMake version, and Visual Studio version,

Win-10, 64-bits (Intel I7), cmake 3.7.1, Visual Studio 2015.

But as I've tried to explain; I don't understand what requirements CubicSDR have:
* ASCII or UNICODE or both?
* `_RELEASE`, `_DEBUG` or both? Seems mixing CRTs is creating troubles. E.g. allocations in `SoapySDR.lib`  (`_DEBUG`) will cause a crash in CubicSDR (`_RELEASE`) when releasing memory. Isn't that true?
* Is using wxWidgets dynamically possible (`WXUSINGDLL`)?

Questions like this which Cmake is very good at obfuscating. Hence the reason I wrote this [GNU-makefile](https://gist.github.com/gvanem/004f7a6254ad897b5bc3ae694575eab7) which helps me understand the details.

vsonnier, 2017-04-03:
@gvanem The only supported ways are as the guide implicitly states: wxWidgets satically built, and Release-like configurations, for all modules, and external dlls.
CMake generates also a RelWithDbInfo configuration which is precisely made for debugging.

vsonnier, 2017-04-23:
Closing by lack of activity.

vsonnier, 2018-05-18:
I've updated the Windows build instructions with the latest wxWidgets, trying not to drown in useless details. I'm closing this issue.

closed 2018-05-18 by vsonnier