cjcliffe/CubicSDR #537: Real need for specific liquid-dsp calls for Windows ?

vsonnier, 2017-04-05:
@cjcliffe Hello ! I have a question about existing Windows specific liquid-dsp code.
Indeed when looking up for ``#ifdef WIN32`` blocks of code, I fall upon a specific liquid-dsp code made for Windows in ``ModemLSB`` and ``ModemUSB``. 
The origin of this specific code is commit 305cd4bf, saying "Temporary FIR fix for windows"

Do you remember in what consists this fix, and if it is still valid today ?
When I experimented on Windows by enabling the same ``iirfilt_crcf_create_lowpass`` call as the others, neither LSB / USB crash, but I don't have LSB/USB stations around so couldn't test it really worked or not as expected.

This is merely a matter of removing platform-specific code whenever we can.

cjcliffe, 2017-04-20:
@vsonnier if liquid-dsp is up to date then there's no need for the windows-specific fix.  

The built-in SSB modems don't suppress the side bands well and have a tendency to drift like they're tracking a carrier regardless of the carrier setting; works brilliantly for AM /w carrier though. 

As a result I did my own LSB/USB implementation which works much better but the equivalent functions weren't available or behaved differently on windows -- So I found another combination  that did almost as well but is likely far less efficient and could easily be dropped now that liquid-dsp is ready.

vsonnier, 2017-04-20:
@cjcliffe Thanks for the explanation. Do you want to confirm by testing yourself if it is OK now for Windows, or are you confident enough I can provide the fix ? As I said, I only own a RTL-SDR, so I can't reach LSB/SSB ham bands and test by myself.

cjcliffe, 2017-04-21:
Sure I'll check out the HF on windows here; I'm sure it's fine but doesn't hurt to confirm 👍

cjcliffe, 2017-04-21:
@vsonnier worked fine, I've removed the temporary code and pushed to master; closing this one up.

closed 2017-04-21 by cjcliffe