cjcliffe/CubicSDR #538: Cubic Slowdown

starvald, 2017-04-14:
The fps of cubic begins slow down after about 15 minutes, mousing over cubic you can see the interface stutter.
i have noticed that if i leave a menu column open the fps goes back to normal.

`__GL_SYNC_TO_VBLANK=0 /usr/bin/CubicSDR` 
i have disable vsync or the fps is terrible to begin with.
using nvidia 378.13 with Mate & Compiz on Antergos (arch based).

any possible fixes?

cjcliffe, 2017-04-20:
@starvald I always have to open the Nvidia panel and disable then re-enable vertical sync to make it work properly for CubicSDR and others; then it caps to the refresh instead of tearing up the CPU with redundant frames -- you may have a similar issue.

Note that if CubicSDR is unfocused it will enable the "Reduce CPU usage" mode to free up graphics resources for other applications.

starvald, 2017-04-22:
i assume it's still a bug though?
from the roadmap "Resolve all driver/platform vertical sync issues"

i guess it will be fixed at some point.

cjcliffe, 2017-04-22:
@starvald yup; it's mostly linux related as that seems to be where problems crop up; I'm assuming if I can resolve that issue on my nvidia card it will fix it elsewhere -- I spent a significant amount of time trying to pinpoint early on and wasn't sure it was even a widespread issue -- but will revisit it at some point.

If you can record some CPU/GPU resource information up to the point it starts to fail this might help us figure out if it's really the refresh issue or possibly some sort of leak / cumulative problem.

ehidle, 2017-05-21:
You can wrap CubicSDR in commands that disable and then re-enable vsync. The sleep 5 allows Cubic to come up and go through OpenGL init before setting vsync back to 1: 

#!/bin/bash
nvidia-settings --load-config-only --assign="SyncToVBlank=0"
CubicSDR&
sleep 5
nvidia-settings --load-config-only --assign="SyncToVBlank=1"

cjcliffe, 2017-05-26:
@ehidle thanks; I'll have to try that myself and possibly add it to the manual as a tip for people experiencing the same issue

starvald, 2017-05-26:
Have been trying this during the week and it works, many thanks ehidle.

ehidle, 2017-05-31:
My pleasure. I have honestly had some mixed results as it seems at least possible that CubicSDR will go back to vsync at some point. But, I solved it permanently by ditching the nVidia card. Haha.

joshuarubin, 2020-08-25:
I'm running into this quite frequently. It makes the software unusable (and I'm running on a 32 core machine with 64GB of memory). I'm running Arch linux using Sway/Wayland for the desktop and an AMD Vega64 video card. Any help would be appreciated.

ehidle, 2020-08-26:
Yeah @joshuarubin I forgot about this but I am also still having this issue on the latest commit. I'm back to a GTX-1060 but disabling vsync in the end only seems to prolong the inevitable. I get maybe a couple of hours or so before I have to sigterm and restart.

joshuarubin, 2020-08-26:
Yeah, mine starts to completely bog down after only 10-15 min. I love the software, just wish I could use it.

pdvmeiring, 2021-09-26:
(Sept 2021) Slowdown is *still* a significant issue - usually starts about 5 minutes into use and then gets worse. Manifests as a jerky waterfall display on mouse movement inside rthe progam window and increasing cpu usage. "disable vsync" makes no difference. Great program otherwise - in particular the bidirectional integration with hamlib is *very* useful and not present in rival programs.

Wish list for "fixing":
1 Remember device for automatic startup
2. Sometimes devices are not recognised on startup and the device list "refresh" needs pressing to see them.
3. Remember state of all window settings startup - some are, some are not!

(using NUC i7 8 core; one core is soon fully used; top gives 132% CPU usage)

eNg1ne85, 2022-02-28:
The same problem occurs with integrated intel graphics and latest CubicSDR 0.2.7 version

cjcliffe, 2026-11-15:
I've managed to reproduce and fix most buffering and slowdown issues; since there were several different factors I'm going to close this one as assumed fixed but feel free to report any new slowdown issues against master or once 0.3.0 binaries land

closed 2026-11-15 by cjcliffe