cjcliffe/CubicSDR #539: Are you plan to support Airspy HF+?

bdn76, 2017-04-15:
Are you plan to support Airspy HF+?

cjcliffe, 2017-04-20:
@bdn76 I'm sure the AirSpy team will support it through the existing or a similar library in which case forking or adding it to the existing SoapyAirSpy module shouldn't be a problem.

vsonnier, 2018-02-25:
@bdn76 @cjcliffe FYI there is now a SoapySDR AirSpayHF module: https://github.com/pothosware/SoapyAirspyHF

andrewbradnan, 2018-05-03:
After I built airspyhf I could find my device with SoapySDRUtil, but not with CubicSDR.

I then changed CMakeLists.txt:718 and turned on bundling of local SoapySDR modules, and it started showing up.  The checked in code seems more correct, but hopefully this will get a few people going.

cjcliffe, 2018-05-08:
@vsonnier @andrewbradnan  ah, I hadn't noticed the new module -- I'll try to make sure that's included in 0.2.5 builds.

Toontje, 2018-07-29:
Hi @cjcliffe , @vsonnier!

Since my RSP1 died i am looking at buying a new device. I am torn between a new RSP2 or an Airspy HF+. Basic requirement is that the device is supported by CubicSDR. So two questions:

1. When are you planning to release 2.0.5?
2. Will this build include support for the Airspy HF+?

Thanks,

Ton.

vsonnier, 2018-07-29:
@Toontje I don't know of the Airspy HF+, but the RSP2 support is as complete as it can be, with contribution of your humble servant.
@stottsc is the author of the HF+ plugin [here](https://github.com/pothosware/SoapyAirspyHF) so you may ask him of its capabilities. 

About a v0.25 release, it's up to @cjcliffe to decide, of course. 
The best I could do on my side would be building Windows installers anyway.

cjcliffe, 2018-08-03:
@TonMachielsen @vsonnier I've already got Airspy HF+ module ready to go for the next build but it will be untested.

I've been busy with other projects but will be back from vacation on the 14th and see about getting 0.2.5 build ready.

vsonnier, 2022-01-15:
AirspyHF has landed in Soapy some time ago: https://github.com/pothosware/SoapyAirspyHF

closed 2022-01-15 by vsonnier