cjcliffe/CubicSDR #540: Firmware Issue with USRP B205-Mini

kfriede, 2017-04-18:
Using CubicSDR Release binary 0.2.0 on Mac OS Sierra with an Ettus B205-mini via USB 3.  Application was not seeing the device, so I ran CubicSDR via terminal to see more details.  Got the following:

`-- Loading firmware image: /usr/local/share/uhd/images/usrp_b200_fw.hex...
  default_input = True
  default_output = False
  device_id = 0
  driver = audio
  label = Apple Inc.: Built-in Microphone
Make device 0
  device_id=0
  origin=https://github.com/pothosware/SoapyAudio

  default_input = False
  default_output = False
  device_id = 2
  driver = audio
  label = squirrels: AirParrot
Make device 1
  device_id=2
  origin=https://github.com/pothosware/SoapyAudio

  driver = uhd
  name = B205i
  product = B205mini
  serial = 30EF556
  type = b200
Make device 2
[INFO] Detected Device: B205mini
[INFO] Loading FPGA image: /usr/local/share/uhd/images/usrp_b205mini_fpga.bin...
[INFO]   0%
[IN  1%
[IN  2%
[IN  3%
[IN  4%
[IN  5%
[IN  6%
[IN  7%
[IN  8%
[IN  9%
[IN 10%
[IN 11%
[IN 12%
[IN 13%
[IN 14%
[IN 15%
[IN 16%
[IN 17%
[IN 18%
[IN 19%
[IN 20%
[IN 21%
[IN 22%
[IN 23%
[IN 24%
[IN 25%
[IN 26%
[IN 27%
[IN 28%
[IN 29%
[IN 30%
[IN 31%
[IN 32%
[IN 33%
[IN 34%
[IN 35%
[IN 36%
[IN 37%
[IN 38%
[IN 39%
[IN 40%
[IN 41%
[IN 42%
[IN 43%
[IN 44%
[IN 45%
[IN 46%
[IN 47%
[IN 48%
[IN 49%
[IN 50%
[IN 51%
[IN 52%
[IN 53%
[IN 54%
[IN 55%
[IN 56%
[IN 57%
[IN 58%
[IN 59%
[IN 60%
[IN 61%
[IN 62%
[IN 63%
[IN 64%
[IN 65%
[IN 66%
[IN 67%
[IN 68%
[IN 69%
[IN 70%
[IN 71%
[IN 72%
[IN 73%
[IN 74%
[IN 75%
[IN 76%
[IN 77%
[IN 78%
[IN 79%
[IN 80%
[IN 81%
[IN 82%
[IN 83%
[IN 84%
[IN 85%
[IN 86%
[IN 87%
[IN 88%
[IN 89%
[IN 90%
[IN 91%
[IN 92%
[IN 93%
[IN 94%
[IN 95%
[IN 96%
[IN 97%
[IN 98%
[IN 99%
[IN100%
[IN done
[INFO] Operating over USB 3.
Error making device: RuntimeError: Expected FPGA compatibility number 4, but got 5:
The FPGA build is not compatible with the host code build.
Please run:

 "/usr/local/lib/uhd/utils/uhd_images_downloader.py"

  driver = uhd
  name = B205i
  product = B205mini
  serial = 30EF556
  type = b200
Make device 3
[INFO] Detected Device: B205mini
[INFO] Operating over USB 3.
Error making device: RuntimeError: Expected FPGA compatibility number 4, but got 5:
The FPGA build is not compatible with the host code build.
Please run:

 "/usr/local/lib/uhd/utils/uhd_images_downloader.py"

Reporting enumeration complete.
SDR enumerator done.`

I have run the uhd_images_downloader and it does not solve the problem.

Justinpiggy, 2017-06-25:
Same problem encountered! In fact, right after installing uhd and CubicSDR, I even received error that no 'UHD_IMAGES_DIR' environment variable is found. After adding that environment variable into terminal, I received the same FPGA build not compatible error. Re-downloading FPGA bitstreams does not help.

AgriVision, 2021-06-07:
same here:
```
Make device 2
[INFO] [b200_impl.cpp:383] [B200] Detected Device: B205mini
[INFO] [B200] Detected Device: B205mini
[INFO] [b200_impl.cpp:430] [B200] Operating over USB 3.
[INFO] [B200] Operating over USB 3.
Error making device: RuntimeError: Expected FPGA compatibility number 5, but got 7:
The FPGA build is not compatible with the host code build.
```