cjcliffe/CubicSDR #543: Compile Issues

cascmptrski, 2017-04-22:
Yesterday I cloned the repository and built according to the Linux instructions. When I ran make on the CubicSDR code I received many warnings about unused and uninitialized parameters. Then the compilation crashed unable to locate rs232.h. Searching I found the header file in the CubicSDR/external/rs232 directory, however make was looking for it in the CubicSDR/src directory. I tried copying it to the src directory and this time the build crashed unable to find several methods in the RigControlPortDialog and SetRigControlPort objects. 

It would appear that something is amiss in the configure step as the system seems to be looking in the wrong directories and perhaps is missing some objects (perhaps several modules?) that properly initialize and utilized the parameters identified in the warnings and build the actual objects that are to be used.

Any hints as to what might be missing and/or where to look to correct the files in the repository would be greatly appreciated.

Thanks,
-CAS

jbolger, 2017-04-25:
Hey,

The error message in this case about `rs232.h` is confusing because of the file with the same name within the cubicsdr codebase.  I tried the same solution you did and got the same errors as a result - I think it is a coincidence that the file names are the same, but the missing `rs232.h` is not the one we are looking for here.

The only way I was able to get it to compile through this was to enable hamlib support.  Once I enabled hamlib it built with no problem.

I do not know for sure if this has any other side effects.

Dantali0n, 2017-04-26:
This issue was resolved in commit 0cf09d1 around 4 days ago. If you update to the latest on master and try again it should be resolved.

cascmptrski, 2017-04-28:
Thank you for the update. The updated code now compiled without issue and works - sort of. It appears there is a buffer overflow or timing synchronization issue somewhere in the code.

Running CubicSDR from the command line works great for 1 to 5 minutes and then abruptly crashes. The display window is present, but the waterfall stops. The speaker issues a series of high-pitched squeaks  timed with the error messages scrolling across the terminal display.  Attached is a redirect of the terminal with the error messages in the terminal pasted at the end. 

The time to failure does not seem to be deterministic, but randomly occurs between 30 seconds and 10 minutes after starting the source. The only recovery is <cntl>C as the CubicSDR window is locked and the terminal window is scrolling through error messages at the maximum rate the display will support.

I am running this on 64-bit Ubuntu 16.04 LTS with 8 cores and 16 GB of memory and using an SDRPlay device as the source. While running the performance is excellent, but the run length is pretty limited.

[C-SDR.log.txt](https://github.com/cjcliffe/CubicSDR/files/965131/C-SDR.log.txt)

Thanks,
-CAS

jbolger, 2017-04-28:
That crash seems more likely to do with bugs in the SDRPlay or SoapySDRPlay software.  Linux support is not the best yet for SDRPlay.

I can run an SDRPlay on my built version of CubicSDR no problem for 12 hours + but if I attempt to increase the sample rate to anything above 2.048 MHz the OS crashes (unrecoverable, hard shutdown of machine required).

Using the same build of CubicSDR with an RTL device and a HackRF had no issues of any kind switching between bandwidths.

Anyone else have any insight into this?  I would love to have my SDRPlay device work at full bandwidth.

cascmptrski, 2017-04-30:
At some level this is true as I just ran it using the RTL-SDR for over an hour without it crashing. I did have one fault in CubicSDR in a issue with a gtk widget.  The setup log and error message follows.

Thanks for the assistance,
-CAS

==============Run Log==============
Audio Device #0 PulseAudio
	Default Output? Yes
	Default Input? Yes
	Input channels: 2
	Output channels: 2
	Duplex channels: 2
	Native formats:
		16-bit signed integer.
		32-bit signed integer.
		32-bit float normalized between plus/minus 1.0.
	Supported sample rates:
		8000hz
		16000hz
		22050hz
		32000hz
		44100hz
		48000hz
		96000hz

SDR enumerator starting.
SoapySDR init..
	API Version: v0.6.0
	ABI Version: v0.6-dev
	Install root: /usr/local
	Loading modules... 
	Available factories...null, remote, rtlsdr, sdrplay
Available vertical sync SwapInterval functions: 
	glxSwapIntervalEXT: No
	DRI2SwapInterval: No
	glxSwapIntervalMESA: No
	glxSwapIntervalSGI: No
No vertical sync swap interval functions available.
Loaded font 'Bitstream Vera Sans Mono' from '/home/corky/Downloads/Radio/CubicSDR/build/x64/fonts/vera_sans_mono12_0.png', parsed 255 characters.
Loaded font 'Bitstream Vera Sans Mono' from '/home/corky/Downloads/Radio/CubicSDR/build/x64/fonts/vera_sans_mono16_0.png', parsed 255 characters.
Loaded font 'Bitstream Vera Sans Mono' from '/home/corky/Downloads/Radio/CubicSDR/build/x64/fonts/vera_sans_mono18_0.png', parsed 255 characters.
Loaded font 'Bitstream Vera Sans Mono' from '/home/corky/Downloads/Radio/CubicSDR/build/x64/fonts/vera_sans_mono24_0.png', parsed 255 characters.
Loaded font 'Bitstream Vera Sans Mono' from '/home/corky/Downloads/Radio/CubicSDR/build/x64/fonts/vera_sans_mono27_0.png', parsed 255 characters.
Found Rafael Micro R820T tuner
  available = Yes
  driver = rtlsdr
  label = Generic RTL2832U OEM :: 00000001
  manufacturer = Realtek
  product = RTL2838UHIDIR
  rtl = 0
  serial = 00000001
  tuner = Rafael Micro R820T
Make device 0
Found Rafael Micro R820T tuner
  origin=https://github.com/pothosware/SoapyRTLSDR
  rtl=0

  driver = 
  label = Not Found ()
Make device 1

Reporting enumeration complete.
SDR enumerator done.
Found Rafael Micro R820T tuner
SDR thread starting.
device init()
[INFO] Using format CF32.
Stream MTU: 131072
[R82XX] PLL not locked!
Disabled direct sampling mode
starting readLoop()

Audio Device #0 PulseAudio
	Default Output? Yes
	Default Input? Yes
	Input channels: 2
	Output channels: 2
	Duplex channels: 2
	Native formats:
		16-bit signed integer.
		32-bit signed integer.
		32-bit float normalized between plus/minus 1.0.
	Supported sample rates:
		8000hz
		16000hz
		22050hz
		32000hz
		44100hz
		48000hz
		96000hz

DemodulatorThread::run() cannot push ati into audioOutputQueue, is full !
DemodulatorThread::run() cannot push ati into audioOutputQueue, is full !
DemodulatorThread::run() cannot push ati into audioOutputQueue, is full !
DemodulatorThread::run() cannot push ati into audioOutputQueue, is full !
DemodulatorThread::run() cannot push ati into audioOutputQueue, is full !
DemodulatorThread::run() cannot push ati into audioOutputQueue, is full !
DemodulatorThread::run() cannot push ati into audioOutputQueue, is full !
DemodulatorThread::run() cannot push ati into audioOutputQueue, is full !

(CubicSDR:2337): GLib-GObject-WARNING **: invalid unclassed pointer in cast to 'GtkWidget'

(CubicSDR:2337): Gtk-CRITICAL **: gtk_widget_set_size_request: assertion 'GTK_IS_WIDGET (widget)' failed
DemodulatorThread::run() cannot push ati into audioOutputQueue, is full !
DemodulatorThread::run() cannot push ati into audioOutputQueue, is full !
DemodulatorThread::run() cannot push ati into audioOutputQueue, is full !

cjcliffe, 2017-05-12:
@cascmptrski it looks like pulseaudio or an audio driver isn't keeping up; are you able to try compiling with something like OSS / ALSA? Add -DUSE_AUDIO_PULSE=0 and  -DUSE_AUDIO_OSS=1 or -DUSE_AUDIO_ALSA=1 and see if there's an improvement?  -DUSE_AUDIO_JACK=1 is also an option.

vsonnier, 2018-02-25:
@cascmptrski @cjcliffe @jbolger  Cleanup time ! The initial topic 'Compile issues' were solved for quite some time, but this issue tuned into a *pot-pourri* one (en français dans le texte)  with unrelated topics:
- SDRPlay vs. Sample rate > 2.048Mhz : other issues have shown it works well on Linux now with an up-to-date SDRplay driver, (do not mix-up with issues involving SoapySDRRemote)
- Crash on Ubuntu 16.04 due to old driver : idem,
- Audio output not keeping up: idem, in the worst case a new issue will pop-up eventually.

So I'm closing this one.

closed 2018-02-25 by vsonnier