cjcliffe/CubicSDR #544: Access violation in atio6axx.dll causes CubicSDR to hang after initial screens are shown

Dantali0n, 2017-04-23:
Hello I have switched from a Nvidia based graphics device to an AMD one in the past few months, An RX470 8gb edition running with version 17.4.2 of the Radeon software.

Unfortunately I have been unable start CubicSDR with this new graphics adapter which seems to undergo an access violation in atio6axx.dll (according to visual studio). I have since then used x64dbg as an disassembler to step by step analyze system calls until the error occurred.

> DLL Loaded: 000007FEE1190000 C:\Windows\System32\atig6txx.dll
> EXCEPTION_DEBUG_INFO:
>            dwFirstChance: 1
>            ExceptionCode: C0000005 (EXCEPTION_ACCESS_VIOLATION)
>           ExceptionFlags: 00000000
>         ExceptionAddress: 00000000000000CC
>         NumberParameters: 2
> ExceptionInformation[00]: 0000000000000008 DEP Violation
> ExceptionInformation[01]: 00000000000000CC Inaccessible Address
> First chance exception on 00000000000000CC (C0000005, EXCEPTION_ACCESS_VIOLATION)!

Using the disassembler I can use the program just fine and it is fully functional even while raising this access violation. I am uncertain on how we can avoid this behavior and what causes it. CubicSDR is the only program which has this behavior. I do want to test with some other programs using wxWidgets 3.1 if you know of any please let me know.

vsonnier, 2017-04-23:
@Dantali0n Hello ! It appears to be a DEP (Data Execution Prevention) execption. You can try to exclude DEP check for CubicSDR specifically (search Google how to do that) as a workaround attempt.

vsonnier, 2017-04-23:
- type Windows + Pause keys as shortcut, this open the System Information,
- Select Advanced...
- Performance tab, select Parameters
- Select Data Execution Prevention tab,

the rest is self-explanatory, add CubicSDR to the list of not-check programs.

Dantali0n, 2017-04-23:
I build wxWidgets with debug symbols to see if the problem is caused by wxWidgets and it definitely is. So we can close this issue and I'll see if I have time to gather enough info to make a bug report for wxWidgets.

closed 2017-04-23 by Dantali0n