cjcliffe/CubicSDR #546: Audio problem with SDRPlay on Ubuntu 16.04

frspin, 2017-05-22:
After upgrading from GIT Sunday I have some audio problem in AM reception, apparently distortion or a buffer problem. Latest commit, at first update yesterday, was about circular buffer.....

Rx is SDRPlay and OS is Ubuntu 16.04/64 bit. Before this update, no audio problems at all.

I have 2 fiile: am.wav is recorded from CubicSDR and am1.wav is recorded from Gqrx on same station and same song. A ZIP with both are attached to this mail. 
[am.zip](https://github.com/cjcliffe/CubicSDR/files/1019909/am.zip).
Same station, received as USB or as LSB, show same distortion.

I can do more test, if required.

Regards
Franco Spinelli
IW2DHW

vsonnier, 2017-05-22:
@frspin  Sorry to hear of your problems. I need to be sure of the revision to pinpoint your particular problem: For that, could you try first on current master (rev af20f680a402763f51edddfbb306c7587ab3d939) then compare it with a build differing by just the old ``ThreadBlockingQueue`` ? 
For that, start with the master and then do a `` git checkout 21a6d2b5f67c3103926a32d9833ef49b8cbc3ecb -- src/util/ThreadBlockingQueue.h`` (if in the CubicSDR dir).

vsonnier, 2017-05-22:
@frspin Hold on, I think i've found a big mistake in the code, another commit is to follow...

vsonnier, 2017-05-23:
@frspin Hello ! After some investigation, it turns out the problem has nothing to do with the circular buffer, but everything with the latest changes I made on the ``SoapySDRThread`` class. Indeed the 'overflow' buffer management were wrong, resulting lost batches on samples on the particular situation where the chosen buffer length is 'big' (bigger than 1/60 s of time) compared to the sample rate. 
I reproduced your case on NFM or FM broadcast by choosing a low sample rate (250Khz) on a RTL-SDR.

Please confirm the latest changes in master solved the problem for you too, so I'll close the issue.

Sorry again for the previous inconvenience.
Regards,
Vincent

vsonnier, 2017-05-28:
@frspin Well I suppose it works now :) Closing.

commit 87769083 (2017-05-23): Fixes #546, when MTU > nb samples per-frame, overflow management was wrong.
closed 2017-05-28 by vsonnier