cjcliffe/CubicSDR #548: Compile problem on Ubuntu 16.04/64 bit

frspin, 2017-05-26:
After last update from Github I am unable to compile CubicSDR on my Ubuntu 16.04/64 bit
This is error:

`
Scanning dependencies of target CubicSDR
[  1%] Building CXX object CMakeFiles/CubicSDR.dir/src/CubicSDR.cpp.o
In file included from /home/spin/Scaricati/ham/CubicSDR/CubicSDR/src/process/ScopeVisualProcessor.h:6:0,
                 from /home/spin/Scaricati/ham/CubicSDR/CubicSDR/src/visual/ScopeCanvas.h:13,
                 from /home/spin/Scaricati/ham/CubicSDR/CubicSDR/src/AppFrame.h:15,
                 from /home/spin/Scaricati/ham/CubicSDR/CubicSDR/src/CubicSDR.h:22,
                 from /home/spin/Scaricati/ham/CubicSDR/CubicSDR/src/CubicSDR.cpp:17:
/home/spin/Scaricati/ham/CubicSDR/CubicSDR/src/process/VisualProcessor.h: In member function ‘virtual void VisualDataReDistributor<OutputDataType>::process()’:
/home/spin/Scaricati/ham/CubicSDR/CubicSDR/src/process/VisualProcessor.h:171:17: error: ‘OutputDataTypePtr’ was not declared in this scope
                 OutputDataTypePtr outp = buffers.getBuffer();
                 ^
/home/spin/Scaricati/ham/CubicSDR/CubicSDR/src/process/VisualProcessor.h:174:19: error: ‘outp’ was not declared in this scope
                 (*outp) = (*inp);
                   ^`

I am missing some file or some compiler update?

Regards
Franco Spinelli
IW2DHW

vsonnier, 2017-05-26:
@frspin Ah Ah I knew I was going to break something:) 
Turns out ``VisualDataDistributor`` and ``VisualDataReDistributor`` classes are actually unused, and so MSVC didn't even try to compile it.
I guess GCC is not so lenient.
So I've made a test by really instantiating the 2 templates, and now it is valid on MSVC.
Please confirm this new ``VisualProcessor.h`` also compiles for you. 
Then, you'll be able to confirm that #546 is also OK.

starvald, 2017-05-27:
still fails to compile on arch x64.
https://paste.ubuntu.com/24675678/

vsonnier, 2017-05-27:
@starvald OK, let's do the compiler ask us. Is it right now ? 
If it still fails, I'll comment out this code, because it is not used anyway by now.

starvald, 2017-05-27:
still fails.
https://paste.ubuntu.com/24677375/

vsonnier, 2017-05-27:
@starvald One last time, with this modification ? At that point, feel free to propose something that compiles for you, else the next commit will see this part of the code commented. Damn I hate templates.

starvald, 2017-05-27:
it now compiles :)
https://paste.ubuntu.com/24677475/
thank you.

vsonnier, 2017-05-27:
@starvald Thanks to you as well for your patience. Have a nice day :+1:

closed 2017-05-27 by vsonnier

frspin, 2017-05-27:
Il 27/05/2017 12:47, Vincent Sonnier ha scritto:
> @starvald <https://github.com/starvald> Thanks to you as well for your
> patience. Have a nice day 👍
> 

Also on Ubuntu 16.04/64 bit CubicSDR now compile and run without problems.

Regards

Franco Spinelli
IW2DHW

commit b3dd0b0d (2017-05-27): Tentative fix #548 part 2