cjcliffe/CubicSDR #549: Does not 'make' with Ubuntu 14.04.5 LTS 32bits

braselectron, 2017-05-27:
I followed this installation framework:

1) First, build liquid-dsp
```
cd ~/Dev
git clone https://github.com/jgaeddert/liquid-dsp.git ./liquid-dsp
cd liquid-dsp
./bootstrap.sh
CFLAGS="-march=native -O3" ./configure --enable-fftoverride
make -j4
sudo make install
sudo ldconfig
```

2) Then build wxWidgets
```
cd ~/Dev
wget
https://github.com/wxWidgets/wxWidgets/releases/download/v3.1.0/wxWidgets-
3.1.0.tar.bz2
tar -xvjf wxWidgets-3.1.0.tar.bz2
cd wxWidgets-3.1.0
mkdir -p ~/Dev/wxWidgets-staticlib
./autogen.sh
./configure --with-opengl --disable-shared --enable-monolithic \
--with-libjpeg --with-libtiff --with-libpng --with-zlib --disable-sdltest \
--enable-unicode --enable-display --enable-propgrid --disable-webkit \
--disable-webview --disable-webviewwebkit \
--prefix=`echo ~/Dev/wxWidgets-staticlib` CXXFLAGS="-std=c++0x"
make -j4
make install
```

3) Build CubicSDR
```
cd ~/Dev
git clone https://github.com/cjcliffe/CubicSDR.git ./CubicSDR
cd CubicSDR
mkdir build
cd build
cmake ../ -DCMAKE_BUILD_TYPE=Release \
-DwxWidgets_CONFIG_EXECUTABLE=~/Dev/wxWidgets-staticlib/bin/wx-config
make
```

stdout:

```
$ make
[  1%] Building CXX object CMakeFiles/CubicSDR.dir/src/CubicSDR.cpp.o
In file included from ~/Dev/CubicSDR/src/IOThread.h:17:0,
                 from ~/Dev/CubicSDR/src/demod/DemodDefs.h:14,
                 from ~/Dev/CubicSDR/src/demod/DemodulatorThread.h:9,
                 from ~/Dev/CubicSDR/src/demod/DemodulatorInstance.h:10,
                 from ~/Dev/CubicSDR/src/demod/DemodulatorMgr.h:10,
                 from ~/Dev/CubicSDR/src/visual/PrimaryGLContext.h:14,
                 from ~/Dev/CubicSDR/src/CubicSDR.h:13,
                 from ~/Dev/CubicSDR/src/CubicSDR.cpp:17:
~/Dev/CubicSDR/src/util/Timer.h:35:7: warning: extra tokens at end of #endif directive [enabled by default]
 #endif;
       ^
<then a lot of other similar error messages>

make[2]: *** [CMakeFiles/CubicSDR.dir/src/CubicSDR.cpp.o] Error 1
make[1]: *** [CMakeFiles/CubicSDR.dir/all] Error 2
make: *** [all] Error 2
```

My thoughts: must be some makefile bug (in respect to compilers/tools versions used not compatible).

Please notice that I have GRC (ie. GNU Radio Companion) and GQRX (ie. gqrx.dk) working well with my NooElec NESDR Nano 2+ Tiny Black RTL-SDR USB

 
```
SoapySDRUtil --info
-######################################################
-## Soapy SDR -- the SDR abstraction library
-######################################################

Lib Version: v0.6.0-unknown
API Version: v0.6.0
ABI Version: v0.6
Install root: /usr
Search path: /usr/lib/i386-linux-gnu/SoapySDR/modules0.6
Search path: /usr/local/lib/i386-linux-gnu/SoapySDR/modules0.6
Search path: /usr/local/lib/SoapySDR/modules0.6
Module found: /usr/local/lib/SoapySDR/modules0.6/libremoteSupport.so
Module found: /usr/local/lib/SoapySDR/modules0.6/libsdrPlaySupport.so
Loading modules... done
Available factories...null, remote, sdrplay

Packages needed - dependencies that are installed:
||/ Name           Version      Architecture Description
+++-==============-============-============-==================================
ii  automake       1:1.14.1-2ub all          Tool for generating GNU Standards-
ii  build-essentia 11.6ubuntu6  i386         Informational list of build-essent
ii  cmake          2.8.12.2-0ub i386         cross-platform, open-source make s
ii  freeglut3:i386 2.8.1-1      i386         OpenGL Utility Toolkit
ii  freeglut3-dev: 2.8.1-1      i386         OpenGL Utility Toolkit development
ii  git            1:1.9.1-1ubu i386         fast, scalable, distributed revisi
ii  libgtk-3-dev   3.10.8-0ubun i386         development files for the GTK+ lib
ii  libpulse-dev:i 1:4.0-0ubunt i386         PulseAudio client development head

GNU Make 3.81  built for i686-pc-linux-gnu (i386 arch)

/usr/local/lib$ ls -la libmirsd*
lrwxrwxrwx 1 root root  36    May 26 00:43 libmirsdrapi-rsp.so -> /usr/local/lib/libmirsdrapi-rsp.so.2
lrwxrwxrwx 1 root root  39    May 26 00:43 libmirsdrapi-rsp.so.2 -> /usr/local/lib/libmirsdrapi-rsp.so.2.10
-rw-r--r-- 1 root root 309892 May 26 00:43 libmirsdrapi-rsp.so.2.10
```
Help!

vsonnier, 2017-05-27:
@braselectron You are right, make is panicking because of an extra semicolon after ``#endif``. 
The last commit in master remove it, that should work now.

closed 2017-05-27 by vsonnier

braselectron, 2017-05-27:
Dear @vsonnier 

We still have errors:

```
Dev/CubicSDR/build$ make
Scanning dependencies of target CubicSDR
[  1%] Building CXX object CMakeFiles/CubicSDR.dir/external/rtaudio/RtAudio.cpp.o
[  2%] Building CXX object CMakeFiles/CubicSDR.dir/src/CubicSDR.cpp.o
In file included from ~/Dev/CubicSDR/external/cubicvr2/math/aabb.h:13:0,
                 from ~/Dev/CubicSDR/external/cubicvr2/math/cubic_math.h:14,
                 from ~/Dev/CubicSDR/src/ui/GLPanel.h:10,
                 from ~/Dev/CubicSDR/src/panel/ScopePanel.h:6,
                 from ~/Dev/CubicSDR/src/process/ScopeVisualProcessor.h:8,
                 from ~/Dev/CubicSDR/src/visual/ScopeCanvas.h:13,
                 from ~/Dev/CubicSDR/src/AppFrame.h:15,
                 from ~/Dev/CubicSDR/src/CubicSDR.h:22,
                 from ~/Dev/CubicSDR/src/CubicSDR.cpp:17:
~/Dev/CubicSDR/external/cubicvr2/math/vec3.h: In static member function ‘static CubicVR::__float CubicVR::vec3::length(CubicVR::vec3, CubicVR::vec3)’:
~/Dev/CubicSDR/external/cubicvr2/math/vec3.h:47:22: warning: ISO C++ says that these are ambiguous, even though the worst conversion for the first is better than the worst conversion for the second: [enabled by default]
             a = ptb[0]-pta[0];
                      ^
~/Dev/CubicSDR/external/cubicvr2/math/vec3.h:33:18: note: candidate 1: CubicVR::__float& CubicVR::vec3::operator[](unsigned int)
         __float& operator [] (unsigned i)       { return ((__float *)this)[i]; }
                  ^
~/Dev/CubicSDR/external/cubicvr2/math/vec3.h:47:22: note: candidate 2: operator[](CubicVR::__float* {aka float*}, int) <built-in>
             a = ptb[0]-pta[0];
                      ^
~/Dev/CubicSDR/external/cubicvr2/math/vec3.h:47:29: warning: ISO C++ says that these are ambiguous, even though the worst conversion for the first is better than the worst conversion for the second: [enabled by default]
             a = ptb[0]-pta[0];
                             ^
~/Dev/CubicSDR/external/cubicvr2/math/vec3.h:33:18: note: candidate 1: CubicVR::__float& CubicVR::vec3::operator[](unsigned int)
         __float& operator [] (unsigned i)       { return ((__float *)this)[i]; }
                  ^
~/Dev/CubicSDR/external/cubicvr2/math/vec3.h:47:29: note: candidate 2: operator[](CubicVR::__float* {aka float*}, int) <built-in>
             a = ptb[0]-pta[0];
                             ^
~/Dev/CubicSDR/external/cubicvr2/math/vec3.h:48:22: warning: ISO C++ says that these are ambiguous, even though the worst conversion for the first is better than the worst conversion for the second: [enabled by default]
             b = ptb[1]-pta[1];
                      ^
(... many more errors...)

In file included from ~/Dev/CubicSDR/external/cubicvr2/math/cubic_math.h:16:0,
                 from ~/Dev/CubicSDR/src/ui/GLPanel.h:10,
                 from ~/Dev/CubicSDR/src/panel/ScopePanel.h:6,
                 from ~/Dev/CubicSDR/src/process/ScopeVisualProcessor.h:8,
                 from ~/Dev/CubicSDR/src/visual/ScopeCanvas.h:13,
                 from ~/Dev/CubicSDR/src/AppFrame.h:15,
                 from ~/Dev/CubicSDR/src/CubicSDR.h:22,
                 from ~/Dev/CubicSDR/src/CubicSDR.cpp:17:
~/Dev/CubicSDR/external/cubicvr2/math/mat4.h:123:53: note: candidate 2: operator[](CubicVR::__float* {aka float*}, int) <built-in>
             return mat4(m[0], m[4], m[8], m[12], m[1], m[5], m[9], m[13], m[2], m[6], m[10], m[14], m[3], m[7], m[11], m[15]);
                                                     ^
^Cmake[2]: *** [CMakeFiles/CubicSDR.dir/src/CubicSDR.cpp.o] Interrupt
make[1]: *** [CMakeFiles/CubicSDR.dir/all] Interrupt
make: *** [all] Interrupt
```

commit d170f731 (2017-05-27): Fix #549, plus more usual syntax for #548
commit 3de3d3ee (2017-05-27): Tentative fix #549 prt 3