cjcliffe/CubicSDR #550: Part 2: Does not 'make' with Ubuntu 14.04.5 LTS 32bits

braselectron, 2017-05-27:
Dear @vsonnier

We still have errors:

```
Dev/CubicSDR/build$ make
Scanning dependencies of target CubicSDR
[  1%] Building CXX object CMakeFiles/CubicSDR.dir/external/rtaudio/RtAudio.cpp.o
[  2%] Building CXX object CMakeFiles/CubicSDR.dir/src/CubicSDR.cpp.o
In file included from ~/Dev/CubicSDR/external/cubicvr2/math/aabb.h:13:0,
                 from ~/Dev/CubicSDR/external/cubicvr2/math/cubic_math.h:14,
                 from ~/Dev/CubicSDR/src/ui/GLPanel.h:10,
                 from ~/Dev/CubicSDR/src/panel/ScopePanel.h:6,
                 from ~/Dev/CubicSDR/src/process/ScopeVisualProcessor.h:8,
                 from ~/Dev/CubicSDR/src/visual/ScopeCanvas.h:13,
                 from ~/Dev/CubicSDR/src/AppFrame.h:15,
                 from ~/Dev/CubicSDR/src/CubicSDR.h:22,
                 from ~/Dev/CubicSDR/src/CubicSDR.cpp:17:
~/Dev/CubicSDR/external/cubicvr2/math/vec3.h: In static member function ‘static CubicVR::__float CubicVR::vec3::length(CubicVR::vec3, CubicVR::vec3)’:
~/Dev/CubicSDR/external/cubicvr2/math/vec3.h:47:22: warning: ISO C++ says that these are ambiguous, even though the worst conversion for the first is better than the worst conversion for the second: [enabled by default]
             a = ptb[0]-pta[0];
                      ^
~/Dev/CubicSDR/external/cubicvr2/math/vec3.h:33:18: note: candidate 1: CubicVR::__float& CubicVR::vec3::operator[](unsigned int)
         __float& operator [] (unsigned i)       { return ((__float *)this)[i]; }
                  ^
~/Dev/CubicSDR/external/cubicvr2/math/vec3.h:47:22: note: candidate 2: operator[](CubicVR::__float* {aka float*}, int) <built-in>
             a = ptb[0]-pta[0];
                      ^
~/Dev/CubicSDR/external/cubicvr2/math/vec3.h:47:29: warning: ISO C++ says that these are ambiguous, even though the worst conversion for the first is better than the worst conversion for the second: [enabled by default]
             a = ptb[0]-pta[0];
                             ^
~/Dev/CubicSDR/external/cubicvr2/math/vec3.h:33:18: note: candidate 1: CubicVR::__float& CubicVR::vec3::operator[](unsigned int)
         __float& operator [] (unsigned i)       { return ((__float *)this)[i]; }
                  ^
~/Dev/CubicSDR/external/cubicvr2/math/vec3.h:47:29: note: candidate 2: operator[](CubicVR::__float* {aka float*}, int) <built-in>
             a = ptb[0]-pta[0];
                             ^
~/Dev/CubicSDR/external/cubicvr2/math/vec3.h:48:22: warning: ISO C++ says that these are ambiguous, even though the worst conversion for the first is better than the worst conversion for the second: [enabled by default]
             b = ptb[1]-pta[1];
                      ^
(... many more errors...)

In file included from ~/Dev/CubicSDR/external/cubicvr2/math/cubic_math.h:16:0,
                 from ~/Dev/CubicSDR/src/ui/GLPanel.h:10,
                 from ~/Dev/CubicSDR/src/panel/ScopePanel.h:6,
                 from ~/Dev/CubicSDR/src/process/ScopeVisualProcessor.h:8,
                 from ~/Dev/CubicSDR/src/visual/ScopeCanvas.h:13,
                 from ~/Dev/CubicSDR/src/AppFrame.h:15,
                 from ~/Dev/CubicSDR/src/CubicSDR.h:22,
                 from ~/Dev/CubicSDR/src/CubicSDR.cpp:17:
~/Dev/CubicSDR/external/cubicvr2/math/mat4.h:123:53: note: candidate 2: operator[](CubicVR::__float* {aka float*}, int) <built-in>
             return mat4(m[0], m[4], m[8], m[12], m[1], m[5], m[9], m[13], m[2], m[6], m[10], m[14], m[3], m[7], m[11], m[15]);
                                                     ^
^Cmake[2]: *** [CMakeFiles/CubicSDR.dir/src/CubicSDR.cpp.o] Interrupt
make[1]: *** [CMakeFiles/CubicSDR.dir/all] Interrupt
make: *** [all] Interrupt
```

vsonnier, 2017-05-27:
@braselectron Well I didn't touch anything related to the ``cubicvr2`` dependency.  Are you sure those are errors ?  Maybe relevant ones  are buried in the rest of the log. 
For comparison, does the  ``0.2.2`` tag version compiles properly ?

braselectron, 2017-05-27:
Dear @vsonnier 

First of all, thank you for the coaching, it is really good to have help from my fellows from GitHub.

I will try to download `0.2.2` tag version and try it and post reply here.

Actually I got in to be interested in trying this port to Linux, because I saw the positive report from @starvald and @frspin that managed to compile the master for Ubuntu 16.04/64 bit (issue #548) successfully.

Since I did not upgrade from Trusty Tahr yet, because of issues I have identified on a VM test, I will not take the risk, for now, and will keep my 14.04.5 version until 18.04 comes with version 2 (just to be sure and stable).

I will continue with the tests here and post results.

braselectron, 2017-05-27:
make CubicSDR-0.2.2 (just a few parts of the log, because it has more than 100MB)

Notice that this time 'make' does not abort and finishes, but with many erros

```
~Dev/CubicSDR-0.2.2/build$ cmake ../ -DCMAKE_BUILD_TYPE=Release -DwxWidgets_CONFIG_EXECUTABLE=~/Dev/wxWidgets-staticlib/bin/wx-config
-- The C compiler identification is GNU 4.8.4
-- The CXX compiler identification is GNU 4.8.4
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
32 bit compiler detected
-- Looking for XOpenDisplay in /usr/lib/i386-linux-gnu/libX11.so;/usr/lib/i386-linux-gnu/libXext.so
-- Looking for XOpenDisplay in /usr/lib/i386-linux-gnu/libX11.so;/usr/lib/i386-linux-gnu/libXext.so - found
-- Looking for gethostbyname
-- Looking for gethostbyname - found
-- Looking for connect
-- Looking for connect - found
-- Looking for remove
-- Looking for remove - found
-- Looking for shmat
-- Looking for shmat - found
-- Looking for IceConnectionNumber in ICE
-- Looking for IceConnectionNumber in ICE - found
-- Found X11: /usr/lib/i386-linux-gnu/libX11.so
-- Found OpenGL: /usr/lib/i386-linux-gnu/libGL.so  
-- Found LIQUID: /usr/local/lib/libliquid.so  
-- Found wxWidgets: TRUE  
-- Copying ~/Dev/CubicSDR-0.2.2/font/*.fnt to directory ~/Dev/CubicSDR-0.2.2/build/x86/fonts
-- Configuring file vera_sans_mono36.fnt
-- Configuring file vera_sans_mono24.fnt
-- Configuring file vera_sans_mono18.fnt
-- Configuring file vera_sans_mono32.fnt
-- Configuring file vera_sans_mono12.fnt
-- Configuring file vera_sans_mono16.fnt
-- Configuring file vera_sans_mono48.fnt
-- Configuring file vera_sans_mono64.fnt
-- Configuring file vera_sans_mono72.fnt
-- Configuring file vera_sans_mono96.fnt
-- Configuring file vera_sans_mono27.fnt
-- Copying ~/Dev/CubicSDR-0.2.2/font/*.png to directory ~/Dev/CubicSDR-0.2.2/build/x86/fonts
-- Configuring file vera_sans_mono36_0.png
-- Configuring file vera_sans_mono64_0.png
-- Configuring file vera_sans_mono32_0.png
-- Configuring file vera_sans_mono72_0.png
-- Configuring file vera_sans_mono48_0.png
-- Configuring file vera_sans_mono24_0.png
-- Configuring file vera_sans_mono12_0.png
-- Configuring file vera_sans_mono18_0.png
-- Configuring file vera_sans_mono16_0.png
-- Configuring file vera_sans_mono96_0.png
-- Configuring file vera_sans_mono27_0.png
-- Copying ~/Dev/CubicSDR-0.2.2/icon/CubicSDR.ico to directory ~/Dev/CubicSDR-0.2.2/build/x86
-- Configuring file CubicSDR.ico
-- Configuring done
-- Generating done
-- Build files have been written to: ~/Dev/CubicSDR-0.2.2/build

cmake finished OK

/Dev/CubicSDR-0.2.2/build$ make
Scanning dependencies of target CubicSDR
[  1%] Building CXX object CMakeFiles/CubicSDR.dir/external/rtaudio/RtAudio.cpp.o
[  2%] Building CXX object CMakeFiles/CubicSDR.dir/src/CubicSDR.cpp.o
In file included from ~/Dev/CubicSDR-0.2.2/external/cubicvr2/math/aabb.h:13:0,
[... 148 lines cut ...]
         __float& operator [] (unsigned i)       { return ((__float *)this)[i]; }
                  ^
In file included from ~/Dev/CubicSDR-0.2.2/external/cubicvr2/math/cubic_math.h:25:0,
                 from ~/Dev/CubicSDR-0.2.2/src/ui/GLPanel.h:10,
                 from ~/Dev/CubicSDR-0.2.2/src/ui/UITestContext.h:7,
                 from ~/Dev/CubicSDR-0.2.2/src/ui/UITestContext.cpp:4:
~/Dev/CubicSDR-0.2.2/external/cubicvr2/math/frustum.h:135:66: note: candidate 2: operator[](CubicVR::__float* {aka float*}, int) <built-in>
             points[6] = vec3(bbox.max[0], bbox.max[1], bbox.min[2]);
                                                                  ^
[100%] Linking CXX executable x86/CubicSDR
[100%] Built target CubicSDR
```

the binary resulting from make does run and open GUI, but since my RSP2 did not arrive yet, I cannot test it to see if it can detected it.

I do have a NooElec NESDR Nano 2+ Tiny Black RTL-SDR USB, but since I followed the "steps" for sdrplay I understand that it does not detect it.

Conclusion there is a syntax problem on the sources in respect with the 32bit version I have installed here, which are:

-- The C compiler identification is GNU 4.8.4
-- The CXX compiler identification is GNU 4.8.4

I also understand that @starvald and @frspin are running on 64bits version.

So could it be a problem of 64bits versus 32bits ?

vsonnier, 2017-05-28:
@braselectron I see that ``cunbicvr2`` is very hacky, and authorize way too much convert-to-pointer operators that are really needed, causing clashes between ``operator *`` and ``operator []`` and the default convert-to-pointer from array...etc.
Well this was not very clear in the first place. 
Can you try the branch https://github.com/cjcliffe/CubicSDR/tree/vso_CubicVR2_cleaner_array_access where I managed to clear that up, I think.

braselectron, 2017-05-28:
Dear @vsonnier 

Thank you again for the help.  Sorry to the late reply, but we are in different time zones. 

I will download the branch suggested and post feedback here.

Please standby. (QRX as fast as possible)

vsonnier, 2017-05-28:
@braselectron  No worries, I've pushed another commit minutes ago. Take your own good time.

braselectron, 2017-05-28:
@vsonnier  just did the make.  Good job! Much better this time, but still some glitches.  Please see below the output for you analysis. 

Must say, you are excellent coder (and quick - based on my undersanding of C++ and the number of coded lines of this project).

```
~/Downloads/SDR/CubicSDR-VR2/build$ cmake ../ -DCMAKE_BUILD_TYPE=Release -DwxWidgets_CONFIG_EXECUTABLE=~/Downloads/SDR/Dev/wxWidgets-staticlib/bin/wx-config
-- The C compiler identification is GNU 4.8.4
-- The CXX compiler identification is GNU 4.8.4
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
32 bit compiler detected
-- Looking for XOpenDisplay in /usr/lib/i386-linux-gnu/libX11.so;/usr/lib/i386-linux-gnu/libXext.so
-- Looking for XOpenDisplay in /usr/lib/i386-linux-gnu/libX11.so;/usr/lib/i386-linux-gnu/libXext.so - found
-- Looking for gethostbyname
-- Looking for gethostbyname - found
-- Looking for connect
-- Looking for connect - found
-- Looking for remove
-- Looking for remove - found
-- Looking for shmat
-- Looking for shmat - found
-- Looking for IceConnectionNumber in ICE
-- Looking for IceConnectionNumber in ICE - found
-- Found X11: /usr/lib/i386-linux-gnu/libX11.so
-- Found OpenGL: /usr/lib/i386-linux-gnu/libGL.so  
-- Found LIQUID: /usr/local/lib/libliquid.so  
-- Found wxWidgets: TRUE  
-- Copying ~/Downloads/SDR/CubicSDR-VR2/font/*.fnt to directory ~/Downloads/SDR/CubicSDR-VR2/build/x86/fonts
-- Configuring file vera_sans_mono36.fnt
-- Configuring file vera_sans_mono24.fnt
-- Configuring file vera_sans_mono18.fnt
-- Configuring file vera_sans_mono32.fnt
-- Configuring file vera_sans_mono12.fnt
-- Configuring file vera_sans_mono16.fnt
-- Configuring file vera_sans_mono48.fnt
-- Configuring file vera_sans_mono64.fnt
-- Configuring file vera_sans_mono72.fnt
-- Configuring file vera_sans_mono96.fnt
-- Configuring file vera_sans_mono27.fnt
-- Copying ~/Downloads/SDR/CubicSDR-VR2/font/*.png to directory ~/Downloads/SDR/CubicSDR-VR2/build/x86/fonts
-- Configuring file vera_sans_mono36_0.png
-- Configuring file vera_sans_mono64_0.png
-- Configuring file vera_sans_mono32_0.png
-- Configuring file vera_sans_mono72_0.png
-- Configuring file vera_sans_mono48_0.png
-- Configuring file vera_sans_mono24_0.png
-- Configuring file vera_sans_mono12_0.png
-- Configuring file vera_sans_mono18_0.png
-- Configuring file vera_sans_mono16_0.png
-- Configuring file vera_sans_mono96_0.png
-- Configuring file vera_sans_mono27_0.png
-- Copying ~/Downloads/SDR/CubicSDR-VR2/icon/CubicSDR.ico to directory ~/Downloads/SDR/CubicSDR-VR2/build/x86
-- Configuring file CubicSDR.ico
-- Configuring done
-- Generating done
-- Build files have been written to: ~/Downloads/SDR/CubicSDR-VR2/build
john@john-965P-DS3:~/Downloads/SDR/CubicSDR-VR2/build$ make
Scanning dependencies of target CubicSDR
[  1%] Building CXX object CMakeFiles/CubicSDR.dir/external/rtaudio/RtAudio.cpp.o
[  2%] Building CXX object CMakeFiles/CubicSDR.dir/src/CubicSDR.cpp.o
[  3%] Building CXX object CMakeFiles/CubicSDR.dir/src/AppFrame.cpp.o
~/Downloads/SDR/CubicSDR-VR2/src/AppFrame.cpp:1330:6: warning: unused parameter ‘event’ [-Wunused-parameter]
 bool AppFrame::actionOnMenuRig(wxCommandEvent& event) {
      ^
[... 233 lines cut ...]
  setsockopt(IPV6_ADD_MEMBERSHIP) [19: No such device]

Reporting enumeration complete.
SDR enumerator done.
ERROR: thread '9SDRThread' has not terminated in time ! (> 2000 ms)
SDR enumerator starting.
SoapySDR init..
	API Version: v0.6.0
	ABI Version: v0.6
	Install root: /usr/local
	Loading modules... 
	Available factories...null, remote, sdrplay

Reporting enumeration complete.
SDR enumerator done.
Terminating SDR thread..
ERROR: thread '9SDRThread' has not terminated in time ! (> 3000 ms)
Terminating SDR post-processing thread..
Terminating All Demodulators..
Terminating Visual Processor threads..
```

And here is a screenshot.

[image]

braselectron, 2017-05-28:
I will try again, based on you last reply I believe I downloaded the first commit (not the last one).

vsonnier, 2017-05-28:
@braselectron Well thanks for the compliments :) 

I understand you own RTL-SDR, so maybe you can go further by building a ``SoapyRTLSDR`` module. and go further.

Anyway I'm going to commit that on master through a PR, and I'll let you close this issue when your Devices are available and you'll be able to confirm it is finally OK.

vsonnier, 2017-05-28:
@braselectron  Fine, waiting for your second confirmation :+1:

braselectron, 2017-05-28:
We are almost done.  Please see below:

```
~/Downloads/SDR/CubicSDR-VR2_2/build$ make
Scanning dependencies of target CubicSDR
[  1%] Building CXX object CMakeFiles/CubicSDR.dir/external/rtaudio/RtAudio.cpp.o
[  2%] Building CXX object CMakeFiles/CubicSDR.dir/src/CubicSDR.cpp.o
[  3%] Building CXX object CMakeFiles/CubicSDR.dir/src/AppFrame.cpp.o
~/Downloads/SDR/CubicSDR-VR2_2/src/AppFrame.cpp:1330:6: warning: unused parameter ‘event’ [-Wunused-parameter]
 bool AppFrame::actionOnMenuRig(wxCommandEvent& event) {
      ^
~/Downloads/SDR/CubicSDR-VR2_2/src/AppFrame.cpp:2010:6: warning: unused parameter ‘event’ [-Wunused-parameter]
 void AppFrame::OnAboutDialogClose(wxCommandEvent& event) {
      ^
~/Downloads/SDR/CubicSDR-VR2_2/src/AppFrame.cpp:2579:6: warning: unused parameter ‘center_freq’ [-Wunused-parameter]
 void AppFrame::setViewState(long long center_freq) {
      ^
~/Downloads/SDR/CubicSDR-VR2_2/src/AppFrame.cpp: In function ‘std::vector<std::basic_string<char> > str_explode(const string&, const string&)’:
~/Downloads/SDR/CubicSDR-VR2_2/src/AppFrame.cpp:2611:31: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
         if (j == std::string::npos && i < str_len)  j = str_len;
                               ^
~/Downloads/SDR/CubicSDR-VR2_2/src/AppFrame.cpp:2613:31: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
         if (j == std::string::npos) break;
                               ^
[  5%] Building CXX object CMakeFiles/CubicSDR.dir/src/AppConfig.cpp.o
[  6%] Building CXX object CMakeFiles/CubicSDR.dir/src/FrequencyDialog.cpp.o
[  7%] Building CXX object CMakeFiles/CubicSDR.dir/src/DemodLabelDialog.cpp.o
[  8%] Building CXX object CMakeFiles/CubicSDR.dir/src/IOThread.cpp.o
[ 10%] Building CXX object CMakeFiles/CubicSDR.dir/src/ModemProperties.cpp.o
~/Downloads/SDR/CubicSDR-VR2_2/src/ModemProperties.cpp: In member function ‘void ModemProperties::initDefaultProperties()’:
~/Downloads/SDR/CubicSDR-VR2_2/src/ModemProperties.cpp:76:13: warning: variable ‘defaultDevice’ set but not used [-Wunused-but-set-variable]
         int defaultDevice = 0;
             ^
~/Downloads/SDR/CubicSDR-VR2_2/src/ModemProperties.cpp: At global scope:
~/Downloads/SDR/CubicSDR-VR2_2/src/ModemProperties.cpp:264:6: warning: unused parameter ‘event’ [-Wunused-parameter]
 void ModemProperties::OnCollapse(wxPropertyGridEvent &event) {
      ^
~/Downloads/SDR/CubicSDR-VR2_2/src/ModemProperties.cpp:268:6: warning: unused parameter ‘event’ [-Wunused-parameter]
 void ModemProperties::OnExpand(wxPropertyGridEvent &event) {
      ^
[ 11%] Building CXX object CMakeFiles/CubicSDR.dir/src/BookmarkMgr.cpp.o
~/Downloads/SDR/CubicSDR-VR2_2/src/BookmarkMgr.cpp:502:18: warning: unused parameter ‘name_in’ [-Wunused-parameter]
 BookmarkEntryPtr BookmarkMgr::nodeToBookmark(const char *name_in, DataNode *node) {
                  ^
[ 12%] Building CXX object CMakeFiles/CubicSDR.dir/src/sdr/SDRDeviceInfo.cpp.o
[ 13%] Building CXX object CMakeFiles/CubicSDR.dir/src/sdr/SDRPostThread.cpp.o
~/Downloads/SDR/CubicSDR-VR2_2/src/sdr/SDRPostThread.cpp: In member function ‘void SDRPostThread::updateActiveDemodulators()’:
~/Downloads/SDR/CubicSDR-VR2_2/src/sdr/SDRPostThread.cpp:98:38: warning: unused variable ‘demodQueue’ [-Wunused-variable]
         DemodulatorThreadInputQueue *demodQueue = demod->getIQInputDataPipe();
                                      ^
[ 15%] Building CXX object CMakeFiles/CubicSDR.dir/src/sdr/SDREnumerator.cpp.o
[ 16%] Building CXX object CMakeFiles/CubicSDR.dir/src/demod/DemodulatorPreThread.cpp.o
[ 17%] Building CXX object CMakeFiles/CubicSDR.dir/src/demod/DemodulatorThread.cpp.o
[ 18%] Building CXX object CMakeFiles/CubicSDR.dir/src/demod/DemodulatorWorkerThread.cpp.o
[ 20%] Building CXX object CMakeFiles/CubicSDR.dir/src/demod/DemodulatorInstance.cpp.o
[ 21%] Building CXX object CMakeFiles/CubicSDR.dir/src/demod/DemodulatorMgr.cpp.o
~/Downloads/SDR/CubicSDR-VR2_2/src/demod/DemodulatorMgr.cpp: In member function ‘DemodulatorInstance* DemodulatorMgr::loadInstance(DataNode*)’:
~/Downloads/SDR/CubicSDR-VR2_2/src/demod/DemodulatorMgr.cpp:529:10: warning: variable ‘found_device’ set but not used [-Wunused-but-set-variable]
     bool found_device = false;
          ^
[ 22%] Building CXX object CMakeFiles/CubicSDR.dir/src/modules/modem/Modem.cpp.o
[ 23%] Building CXX object CMakeFiles/CubicSDR.dir/src/modules/modem/ModemAnalog.cpp.o
[ 25%] Building CXX object CMakeFiles/CubicSDR.dir/src/modules/modem/ModemDigital.cpp.o
[ 26%] Building CXX object CMakeFiles/CubicSDR.dir/src/modules/modem/analog/ModemAM.cpp.o
[ 27%] Building CXX object CMakeFiles/CubicSDR.dir/src/modules/modem/analog/ModemDSB.cpp.o
[ 28%] Building CXX object CMakeFiles/CubicSDR.dir/src/modules/modem/analog/ModemFM.cpp.o
[ 30%] Building CXX object CMakeFiles/CubicSDR.dir/src/modules/modem/analog/ModemNBFM.cpp.o
[ 31%] Building CXX object CMakeFiles/CubicSDR.dir/src/modules/modem/analog/ModemFMStereo.cpp.o
[ 32%] Building CXX object CMakeFiles/CubicSDR.dir/src/modules/modem/analog/ModemIQ.cpp.o
[ 33%] Building CXX object CMakeFiles/CubicSDR.dir/src/modules/modem/analog/ModemLSB.cpp.o
[... 98 lines cut ...]
~/Downloads/SDR/CubicSDR-VR2_2/src/forms/Dialog/AboutDialogBase.cpp:386:68: warning: ‘wxFont::wxFont(int, int, int, int, bool, const wxString&, wxFontEncoding)’ is deprecated (declared at ~/Downloads/SDR/Dev/wxWidgets-staticlib/include/wx-3.1/wx/gtk/font.h:89): use wxFONT{FAMILY,STYLE,WEIGHT [... cut ...]
[ 93%] Building CXX object CMakeFiles/CubicSDR.dir/external/tinyxml/tinyxmlparser.cpp.o
[ 95%] Building CXX object CMakeFiles/CubicSDR.dir/external/tinyxml/tinyxmlerror.cpp.o
[ 96%] Building CXX object CMakeFiles/CubicSDR.dir/external/cubicvr2/math/cubic_math.cpp.o
[ 97%] Building CXX object CMakeFiles/CubicSDR.dir/src/sdr/SoapySDRThread.cpp.o
[ 98%] Building CXX object CMakeFiles/CubicSDR.dir/src/ui/UITestCanvas.cpp.o
[100%] Building CXX object CMakeFiles/CubicSDR.dir/src/ui/UITestContext.cpp.o
Linking CXX executable x86/CubicSDR
[100%] Built target CubicSDR
```

braselectron, 2017-05-28:
I will then also try to  building a SoapyRTLSDR module. and go further, as you suggested and to clear this too if needed.

As I read once on a truck bumper sticker: 
"Do not despise our effort because we will bring happiness to many souls in despair."

braselectron, 2017-05-28:
SoapyRTLSDR tested sucessfully.

[image]

vsonnier, 2017-05-28:
@braselectron  I've merged the #551 on master.

braselectron, 2017-05-28:
I will download the new CubicSDR master (last one available) and make it to test and place it here.

braselectron, 2017-05-28:
@vsonnier I just downloaded master code and did the make. 

I still am getting some warnings but the make process ends at 100% and produces the binary. 

Can you also explain me how these warning messages impact the make process (other than syntax errors - in my knowledge), since \'make\' does not abort and actually finishes with a working binary ?  

I could be of better help to you on this if I could understand better why these warning messages are being generated.  Its seems to me (by the issue repost history) that MS Win users did not detect this problem (or did not try the code), or is this related to Linux c++, that is, GNU/C++ is more severe with syntax used within the coding ?

Please see below the log output:

```
~/Downloads/SDR/CubicSDR-0.2.2_2/build$ make
Scanning dependencies of target CubicSDR
[  1%] Building CXX object CMakeFiles/CubicSDR.dir/external/rtaudio/RtAudio.cpp.o
[  2%] Building CXX object CMakeFiles/CubicSDR.dir/src/CubicSDR.cpp.o
[  3%] Building CXX object CMakeFiles/CubicSDR.dir/src/AppFrame.cpp.o
~/Downloads/SDR/CubicSDR-0.2.2_2/src/AppFrame.cpp:1330:6: warning: unused parameter ‘event’ [-Wunused-parameter]
 bool AppFrame::actionOnMenuRig(wxCommandEvent& event) {
      ^
~/Downloads/SDR/CubicSDR-0.2.2_2/src/AppFrame.cpp:2010:6: warning: unused parameter ‘event’ [-Wunused-parameter]
 void AppFrame::OnAboutDialogClose(wxCommandEvent& event) {
      ^
~/Downloads/SDR/CubicSDR-0.2.2_2/src/AppFrame.cpp:2579:6: warning: unused parameter ‘center_freq’ [-Wunused-parameter]
 void AppFrame::setViewState(long long center_freq) {
      ^
~/Downloads/SDR/CubicSDR-0.2.2_2/src/AppFrame.cpp: In function ‘std::vector<std::basic_string<char> > str_explode(const string&, const string&)’:
~/Downloads/SDR/CubicSDR-0.2.2_2/src/AppFrame.cpp:2611:31: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
         if (j == std::string::npos && i < str_len)  j = str_len;
                               ^
~/Downloads/SDR/CubicSDR-0.2.2_2/src/AppFrame.cpp:2613:31: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
         if (j == std::string::npos) break;
                               ^
[  5%] Building CXX object CMakeFiles/CubicSDR.dir/src/AppConfig.cpp.o
[  6%] Building CXX object CMakeFiles/CubicSDR.dir/src/FrequencyDialog.cpp.o
[  7%] Building CXX object CMakeFiles/CubicSDR.dir/src/DemodLabelDialog.cpp.o
[  8%] Building CXX object CMakeFiles/CubicSDR.dir/src/IOThread.cpp.o
[ 10%] Building CXX object CMakeFiles/CubicSDR.dir/src/ModemProperties.cpp.o
~/Downloads/SDR/CubicSDR-0.2.2_2/src/ModemProperties.cpp: In member function ‘void ModemProperties::initDefaultProperties()’:
~/Downloads/SDR/CubicSDR-0.2.2_2/src/ModemProperties.cpp:76:13: warning: variable ‘defaultDevice’ set but not used [-Wunused-but-set-variable]
         int defaultDevice = 0;
             ^
~/Downloads/SDR/CubicSDR-0.2.2_2/src/ModemProperties.cpp: At global scope:
~/Downloads/SDR/CubicSDR-0.2.2_2/src/ModemProperties.cpp:264:6: warning: unused parameter ‘event’ [-Wunused-parameter]
 void ModemProperties::OnCollapse(wxPropertyGridEvent &event) {
      ^
~/Downloads/SDR/CubicSDR-0.2.2_2/src/ModemProperties.cpp:268:6: warning: unused parameter ‘event’ [-Wunused-parameter]
 void ModemProperties::OnExpand(wxPropertyGridEvent &event) {
      ^
[ 11%] Building CXX object CMakeFiles/CubicSDR.dir/src/BookmarkMgr.cpp.o
~/Downloads/SDR/CubicSDR-0.2.2_2/src/BookmarkMgr.cpp:502:18: warning: unused parameter ‘name_in’ [-Wunused-parameter]
 BookmarkEntryPtr BookmarkMgr::nodeToBookmark(const char *name_in, DataNode *node) {
                  ^
[ 12%] Building CXX object CMakeFiles/CubicSDR.dir/src/sdr/SDRDeviceInfo.cpp.o
[ 13%] Building CXX object CMakeFiles/CubicSDR.dir/src/sdr/SDRPostThread.cpp.o
~/Downloads/SDR/CubicSDR-0.2.2_2/src/sdr/SDRPostThread.cpp: In member function ‘void SDRPostThread::updateActiveDemodulators()’:
~/Downloads/SDR/CubicSDR-0.2.2_2/src/sdr/SDRPostThread.cpp:98:38: warning: unused variable ‘demodQueue’ [-Wunused-variable]
         DemodulatorThreadInputQueue *demodQueue = demod->getIQInputDataPipe();
                                      ^
[ 15%] Building CXX object CMakeFiles/CubicSDR.dir/src/sdr/SDREnumerator.cpp.o
[ 16%] Building CXX object CMakeFiles/CubicSDR.dir/src/demod/DemodulatorPreThread.cpp.o
[ 17%] Building CXX object CMakeFiles/CubicSDR.dir/src/demod/DemodulatorThread.cpp.o
[ 18%] Building CXX object CMakeFiles/CubicSDR.dir/src/demod/DemodulatorWorkerThread.cpp.o
[ 20%] Building CXX object CMakeFiles/CubicSDR.dir/src/demod/DemodulatorInstance.cpp.o
[ 21%] Building CXX object CMakeFiles/CubicSDR.dir/src/demod/DemodulatorMgr.cpp.o
~/Downloads/SDR/CubicSDR-0.2.2_2/src/demod/DemodulatorMgr.cpp: In member function ‘DemodulatorInstance* DemodulatorMgr::loadInstance(DataNode*)’:
~/Downloads/SDR/CubicSDR-0.2.2_2/src/demod/DemodulatorMgr.cpp:529:10: warning: variable ‘found_device’ set but not used [-Wunused-but-set-variable]
     bool found_device = false;
          ^
[ 22%] Building CXX object CMakeFiles/CubicSDR.dir/src/modules/modem/Modem.cpp.o
[ 23%] Building CXX object CMakeFiles/CubicSDR.dir/src/modules/modem/ModemAnalog.cpp.o
[... 106 lines cut ...]
~/Downloads/SDR/CubicSDR-0.2.2_2/src/forms/Dialog/AboutDialogBase.cpp:386:68: warning: ‘wxFont::wxFont(int, int, int, int, bool, const wxString&, wxFontEncoding)’ is deprecated (declared at ~/Downloads/SDR/Dev/wxWidgets-staticlib/include/wx-3.1/wx/gtk/font.h:89): use wxFONT{FAMILY,STYLE,WEIGHT}_ [... cut ...]
[ 93%] Building CXX object CMakeFiles/CubicSDR.dir/external/tinyxml/tinyxmlparser.cpp.o
[ 95%] Building CXX object CMakeFiles/CubicSDR.dir/external/tinyxml/tinyxmlerror.cpp.o
[ 96%] Building CXX object CMakeFiles/CubicSDR.dir/external/cubicvr2/math/cubic_math.cpp.o
[ 97%] Building CXX object CMakeFiles/CubicSDR.dir/src/sdr/SoapySDRThread.cpp.o
[ 98%] Building CXX object CMakeFiles/CubicSDR.dir/src/ui/UITestCanvas.cpp.o
[100%] Building CXX object CMakeFiles/CubicSDR.dir/src/ui/UITestContext.cpp.o
Linking CXX executable x86/CubicSDR
[100%] Built target CubicSDR
```

vsonnier, 2017-05-29:
@braselectron  What your last log shows are "warnings" not errors, and those warnings is an indication that something troubles the compiler, but it can continue to build anyway. 
Warnings are diagnostic messages, and so some are more important than others, but it is up to the author to determine if they are harmless or not for the point of view of its program.

In  order to help the developer, compilers classify warnings by levels of importance, and some specific warnings can be enabled or not individually.
Besides, MSVC and Gcc are not equivalent in reporting errors or warnings, so they cannot be compared directly.
What this particular Gcc log shows is:
-  ``warning: unused parameter ‘evt’ [-Wunused-parameter]`` : mostly related to wxWidgets, normal and unavoidable,
-  ``warning [-Wdeprecated-declarations]`` : related to wxWidgets, idem above,
... and so on about unused variables.

The previous warnings related to compiling failure are gone, which is the point. 
Now why ``make`` failed ? Maybe there were some errors reported among the other warnings, maybe ``make`` stops when too many warnings are produced, I don't know.

All in all this log shows very few warnings w.r.t amount of code in CubicSDR, i.e compilation is very clean, so I'm closing this issue now.

PR #551 (2017-05-28): CubicVR2  removal of convert to pointer operators (fixes #550)
@cjcliffe This fixes #550 where @braselectron couldn't compile on Ubuntu 14.04.5 LTS 32bit.
The reason is that the ``operator __float*() const`` confuses the Gcc 4.8 compiler.

So this PR removes all convert-to-pointer of ``CubicVR2`` structures, which turns were not really needed, and authorized incredibly dangerous conversions and tests against NULL, and so on.
Modifications:
-  Removed all ``operator __float*() const``, 
-  Changed  ``CubicVR2`` to remove the few NULL-checks based on this operator,
-  Added a ``__float mat4::to_ [... cut]
commit 3095f820 (2017-05-28): CubicVR2: more restricted operators on structures by removing convert-to-pointer. (Fix #550 ?)
commit 83b3bb1c (2017-05-28): Merge pull request #551 from cjcliffe/vso_CubicVR2_cleaner_array_access

CubicVR2  removal of convert to pointer operators (fixes #550)
closed 2017-05-29 by vsonnier