cjcliffe/CubicSDR #552: hackrf on Win7

rgbx559, 2017-05-28:
I'm using the version (2.2) that comes with POTHOS SDR on Win7. I works with the LimeSDR but even though it detects the hackRF SDR I can't get it to start. The POTHOS utility indicates that the hackRF SOAPY interface is detected an usable. Any ideas?

vsonnier, 2017-05-29:
@rgbx559 Hello ! 
The commit 5f478bf referenced your issue by mistake, sorry for the spam. 
Although I also use the PothosSDR Windows distrib, (which is amazing) I don't have a HackRF so I cannot help you there.

f4grx, 2017-08-03:
my hackrf on windows 10 (rad1o badge) is not detected at all by version 0.2, even when starting as administrator. It is detected by gnuradio/gqrx/etc.

The device list is just empty.

vsonnier, 2017-09-01:
@rgbx559 @f4grx Could you please try again with the just-released PothosSDR 2017.08.31 distrib here : http://downloads.myriadrf.org/builds/PothosSDR/?C=M;O=D where some recent HackRF fixes has been integrated ?

vsonnier, 2017-09-19:
@f4grx Did you try the latest PothosSDR distribution ?

f4grx, 2017-09-19:
not recently. I will test and report.

vsonnier, 2018-02-21:
I'm closing this issue following the closing of #602, which shows HackRF working on Windows 10.

closed 2018-02-21 by vsonnier