cjcliffe/CubicSDR #556: Noise blanker, crackles in pulse, center frequency confusion, and easy build in linux

loughkb, 2017-06-04:
Hello,

   First,  this is the best SDR on linux so far, approaching greatness.  THANK YOU to the author.

   Now,   Suggestion,  a noise blanker.  (please?)  

   A query.   I don't understand the utility of the center frequency tuning.  If I'm scrolling through the spectrum and find something worth listening to.  Simply clicking on the signal sometimes leads to a weird confusion in the software where a signal in the main spectrum won't show in the modem spectrum.  I have to manually shift the center frequency, make sure I haven't created extra modems and juggle five kittens while standing on one toe in order to get things to behave again.
  It seems like there should be a single modem browse mode where I can just zoom and scroll around the spectrum, freely clicking on an interesting signal without having to manually shift the center frequency.  Also, a syncup button that brings the center freq to the currently clicked freq might be nice.
  Admittedly, I'm still learning the software, so I could be doing something insanely incorrect here.

  Occasionally, while listening to AM,  I get a 'crackling' sound in the audio that comes and goes.  I suspect it's buffering getting weird, but it kind of sounds like clipping of a too strong audio signal.  I have yet to discover a condition that will reliably reproduce it.  Only a minor annoyance.  Probably not even worth mentioning.   (computer is beefy, more than ample cpu, so it's not resources.)

  Finally,   I've built a couple of scripts that will semi-automate building all needed bits and pieces for SDRPlay owners under linux.   Everything for SoapySDR gets built, and finally Cubic gets pulled down and built by the scripts.  Greatly simplifies getting Cubic running on a linux box with Debian 8, Ubuntu 16.04 or later and all it's derivatives.   Video with Blog link below.

  Thanks again to the author for a great piece of software.
Kevin.
Video link:
https://www.youtube.com/watch?v=V8lb3FGpSCA

vsonnier, 2017-06-07:
@loughkb  Hello ! I assume you already have read a bit of the the manual : http://cubicsdr.readthedocs.io/en/latest/

I'm going to try to answer your questions the best I can, although I don't have all the features in mind.
-  Center frequency vs. the bandwidth. Well, the center frequency is the frequency your device (SDRPlay here) is tuned to. 
Now since the SDRPlay can have an bandwith of say 8Mhz (configurable in SDR play ?)
it means you can create  as many modems (you create one when you click on the spectrum like you said), also called VCO in other software, in the range [center - 4MHz; center + 4Mhz].
-  As expected, the 'best' frequency is the one the device is tuned to. So if you have a modem and want the frequency centered here, select the modem it and 'C'. This is entirely a matter of context and taste to use it. After all the SDR principle is to be able to use a whole band [center -B/2 ; center + B/2] simultaneously, and not only the center frequency. 
- The default mode when you simply click on the signal is to create/recreate one modem, so normally when you click elsewhere the previous modem is deleted and a new one is created at the new place. So you have your 'single modem' mode, by default.
-  'Ctrl+click' on the other hand is intended to create multiple modems.
-  If indeed you created modems you don't want, select them and 'D', or even easier use the Boookmark view => 'Active' and delete them from here. 
- If you have indeed multiple modems set and only want to listen to a single one at a time, the 'Solo Mode' is what you want. Select 'S' for a given modem. Now for all modem in view (or in bandwidth ?) pressing TAB will jump to one modem to the next, and you only listen to the current 'S' at a time.
- Select a modem, use 'Space' to edit its frequency, press 'E' to edit its label, i.e you can add a short description it.
- The list a modems, together with their frequencies, bandwidth, and labels...etc., can be saved in a XML file: File ==> Save session as... and later reloaded. 
- Squelch level is managed by clicking on the Audio meter. The Squelch level try to be dynamic with the mean level. Well you'll see.
 
 Concerning noise filters, there is none for now.

About your specific problem about AM i can't answer for the SDRPlay. I only have a RTL-SDR and also use it to listen to AM Air Band (110-145MHz) and it works OK but of course its extremely noisy. I don't now if that counts for 'cracking sound' though. Maybe this issue is worth a short video so we can try to guess what it is.

loughkb, 2017-06-07:
Hi,
   Thanks for the detailed response.    I've mostly figured out the modems now, but you did clarify a few other things for me.   This program is on it's way to being the go to SDR software for linux.  A much better interface than gqrx.

   The crackling is weird, sounds like audio buffer under runs to me.  It's not consistent either.    I was thinking about doing a video on using the software.   My video on building CubicSDR for SDRPlay got linked on the actual SDRPlay company website.   
   I think I've uncovered a bug you may not be seeing due to the smaller bandwidth of the RTL.   It's hard to explain without showing, but I'll try.

   Yes, the bandwidth of SDRPlay can go pretty high.  Up to 10Mhz max.   But, at any bandwidth, I've noticed this problem.   When zooming or panning around the spectrum,  signals seen in the main waterfall may be ghosts.  Clicking to put a modem there will show no signal in the modems waterfall.   If I grab the main waterfall and drag across a bit of the spectrum,  that signal may disappear.  Zooming will sometimes cause signals and groups of signals to appear or disappear.
   I've seen a signal 1Khz away from the current modem vanish and reappear 1Mhz away.   You can even have the modem on a signal, listening to it, and have it disappear when you zoom in a bit more.   It's like a math error is occuring at certain zoom depths.   I work around it be panning up or down until it reappears.
   I've built 0.2.0 to see if the same thing happens and it does.
   I could do a video demonstrating and leave it unlisted so you could view it without it going public.

Kevin

vsonnier, 2017-06-07:
@loughkb Ah Ah you noticed the zoom, ghost signals, and so on, congratulations :)
These are indeed 'optimization' artifacts and  you already noticed that centering freq or changing the zoom level can make them go away.
If you want to read the whole story : #338, #295, #441, #505.

Long story short, those artifacts are 'normal' as for now,and still are the same in the older versions so no need to go back to 0.20 or anything. 
As a rule, the current master is the best version ever, (unless I've broken some *Nix template syntax..hem) so you can stick to it. 

As for the Audio buffer underrun, I doubt it since you said your computer is powerful enough. 
If you run CubicSDR from a terminal what  kind of traces do you get ? 

I think CubicSDR author @cjcliffe also owns a SDRPlay so he may share some experience with you in case of problems.

loughkb, 2017-06-07:
Well, I do hope the ghosts are not settled on as 'normal' in the long run.   That behavior is unique to Cubic and darned frustrating!   Scanning across the spectrum looking for interesting signals is where an SDR shines.   At least half the time, something interesting turns out to be a ghost and extra work is required to continue the search. (sigh,  zoom, center, fiddle, finally I can continue searching.)
If continuity and reliability is brought to the waterfall,  the program will be an easy recommendation to anyone on linux wanting to do SDR.  I'd figure CJ would make that a priority fix. ;-)
Thanks again.     I'll be doing a video covering it's use in a few days.
Kevin

vsonnier, 2017-11-25:
I'm closing this one because #338 will take of this.

closed 2017-11-25 by vsonnier