cjcliffe/CubicSDR #557: Speed up squelch after TX ends?

starvald, 2017-06-09:
Listening to a channels where you have a mix of strong and weak TX's the squelch stays open too long.
can the speed of it be increased?

imme5150, 2018-04-11:
+1 I'd even rather have some kind of delay so the system could auto-squeltch after the fact and produce really clean audio

ve1jot, 2018-04-11:
I could be wrong, but "squelch tail" is a reflection of AGC time 
constant? IE: AGC fast setting also speeds up squelch closure, AGC slow 
make squelch closure slower...so "AGC fast/slow" and or AGC RF/AGC audio 
switches perhaps?

erichahn, 2019-04-25:
I would love to see faster auto squelch, or ctcss tone abilities.

starvald, 2019-04-28:
it stopped me using cubic for what i really wanted it for (unattended recording of marine band 24/7), hearing that white noise every few seconds was too much. that and cubic's forever problem of the fps slowdown after a few hours.

i switched to rtl_airband where i never hear the squelch tail, though sometimes clips the end of a TX and it doesn't have as good audio as cubic but i can live with that.

i still use cubic on my desktop day to day though.