cjcliffe/CubicSDR #558: Another buffer problem?

frspin, 2017-06-11:
Listening to VOA Radiogram and receiving a long image in MFSK32 with Fldigi. CubicSDR is tuned to 15670 kHz AM mode. Image have an "out of sync" near the end. It seem a buffer problem in CubicSDR or in SoapySDR.
Some image, received with a Ft897D and Fldigi on an Asus T200TA don't have problem
My setup was a I5 processor, 4 cores, 4 Gb of RAM and Ubuntu 16.04/64 bit
Same image, received on same Linux machine Saturday using an old CiaoRadio H101 SDR don't have problem.
[image]

k9wkj, 2017-06-11:
ill add to this
ive been trying to get consistent JT65 decodes on 6M
and its really seeming like there is jitter trouble
i may have to fire up my desktop and give it whirl
more investigation needed  will report back

vsonnier, 2017-08-14:
Summer cleanup time !  Any news on the issue @frspin @k9wkj ? 
If not I'll close the issue.

vsonnier, 2017-08-18:
Closing by lack of activity.

closed 2017-08-18 by vsonnier