cjcliffe/CubicSDR #559: Add antenna selection support for LimeSDR.

romeojulietthotel, 2017-06-21:
I thought maybe Soapy needed to provide antenna info and that's why no antenna selection appears here in CubicSDR. But I looked at Gqrx and it's got antenna selection. 

Is it a matter of adding the menu to CubicSDR or is it a different problem?

romeojulietthotel, 2017-07-19:
I looked into this and not proficient enough to create a PR but the SoapySDR API has the getAntenna function so that part I understand. I am unsure where the antenna selection would make the most sense in the CubicSDR gui.  Logically I would say that under Settings->Antennas->R{0,1}->Drop_Down->{L,H,W} and something similar for TX.

There are 6 possible RX antennas that could be active on the LimeSDR.  R0,R1 with each of L,H,W.  And 4 for TX with T0,T1 with each of Band1,Band2.

vsonnier, 2017-08-23:
Hello @romeojulietthotel ! 
I'm about to finish Antenna support for CubicSDR but I don't think worth the effort to support TX antenna within CubicSDR, which is receive only anyway.... 
We may change TX antenna with CubicSDR, fine, but you must control the TX you want to emit in another application to be useful anyway, so why not the TX in this application ? 
Besides for UI ergonomy sake, in this case it should need 2 different Antenna menus, one for RX, another for TX (Cascading are impractical..) 

Please explain a bit more your use case for TX within CubicSDR !

vsonnier, 2017-08-25:
@romeojulietthotel The master now contains RX antenna choice management. As you suggested, I've also added a read-only menu entry to indicate the active TX antenna, it was not so difficult to do so.
I'm closing this issue then.

closed 2017-08-25 by vsonnier

romeojulietthotel, 2017-08-25:
I tested this yesterday and it works great! Also good to know it wasn't too hard to implement.