cjcliffe/CubicSDR #561: Ubuntu / Debian packages install SoapySDR in different directory then build guide

Dantali0n, 2017-07-16:
Hello,
I have a short note / improvement for the Linux build guide, I noticed the apt packages for Ubuntu & Debian use different directory structure for installing SoapySDR then the build guide. So by following the build guide the system can't find the modules installed through apt anymore.

I suggest we add a short note to the linux build guide providing the user with how to create a symlink for LimeSDR for example (Since I can't find the source for this module so I had to install it through apt).
`sudo ln -s /usr/lib/x86_64-linux-gnu/SoapySDR/modules0.6/libLMS7Support.so /usr/local/lib/SoapySDR/modules0.6/libLMS7Support.so`

Or we could add parameters while installing SoapySDR so it uses the same directories as the packages.

Kind regards,
Dantalion

guruofquality, 2017-07-16:
Can you clarify which packages you installed from APT and which ones from source? SoapySDR, LimeSuite, CubicSDR?

Dantali0n, 2017-07-16:
SoapySDR is from source & SoapySDR-module-LMS7 is from apt. This is because the linux build-guide describes (https://github.com/cjcliffe/CubicSDR/wiki/Build-Linux) to install SoapySDR from source, I wanted to install SoapySDR lms7 from source as well but couldn't find the source and because of that hit into this problem.

romeojulietthotel, 2017-07-19:
Source for lms7 is in https://github.com/myriadrf/LimeSuite

Dantali0n, 2017-08-02:
I downloaded the source for the lms7 and managed to install the SoapySDR module from source. I suggest adding the following to the build guide:

### SoapyLMS7 (similar to other Soapy modules):
```bash
$ sudo apt-get install g++ libsqlite3-dev libi2c-dev libusb-1.0-0-dev libwxgtk3.0-dev freeglut3-dev
$ git clone https://github.com/myriadrf/LimeSuite.git
$ cd LimeSuite
LimeSuite$ mkdir builddir
LimeSuite$ cd builddir
builddir$ cmake .. -DENABLE_DESKTOP=0 -DENABLE_LIME_UTIL=0 -DENABLE_GUI=0 -DENABLE_EXAMPLES=0 -DCMAKE_BUILD_TYPE=Release
builddir$ make
builddir$ sudo cp SoapyLMS7/libLMS7Support.so /usr/local/lib/SoapySDR/modules0.6/libLMS7Support.so
# should now show LMS7 (LimeSDR) device if connected
builddir$ SoapySDRUtil --probe     
```

This will only generate the libLMS7Support.so file and we will manually copy it to the desired directory, leaving the end user free to decide if they want to install the entire LimeSuite and whether they do that from source or from ppa / other binary distribution.

Dantali0n, 2017-09-05:
If no one has any objections or remarks regarding this issue I'll be adding the SoapyLMS7 guide to the overall Linux build guide and then I can close this issue.

Dantali0n, 2018-05-10:
I have greatly improved the above instructions and it is now trivial to add the SoapyLMS7 module using the Linux build instructions~

closed 2018-05-10 by Dantali0n