cjcliffe/CubicSDR #563: Simplified UI

nrcharles, 2017-07-30:
When listening to a single modem, is there a way to hide all but the main spectrum and main waterfall?

dunaybd, 2017-08-06:
+1. A desirable enhancement.

cjcliffe, 2017-09-19:
@nrcharles it can be built that way with compile flags but it might be better to add it as a toggle with default off instead.

nrcharles, 2017-09-19:
Nice! What compile flags are the correct incantation?

cjcliffe, 2017-09-23:
@nrcharles you can use the custom build cmake options:

Enable 'CUSTOM_BUILD' and then you can set the feature flags:

```IF(CUSTOM_BUILD)
  ...
    # feature flags
    SET (CUBICSDR_ENABLE_VIEW_DEMOD ON CACHE BOOL "Enable Second Demodulator Spectrum/Waterfall view.")
    SET (CUBICSDR_ENABLE_VIEW_SCOPE ON CACHE BOOL "Enable Demodulator Scope/Spectrum view.")
   ...
```

CUBICSDR_ENABLE_VIEW_DEMOD enables/disables the smaller modem view
CUBICSDR_ENABLE_VIEW_SCOPE enables/disables the scope area view