cjcliffe/CubicSDR #565: Solo setting is not saved in cubicsession.xml

dunaybd, 2017-08-03:
Each time you start CubicSDR, you must reestablish "Solo" mode. Ideally, I'd like to see it saved in both the session and bookmarks.

I know this is very minor. Should I move this to enhancement requests?

vsonnier, 2017-08-07:
@dunaybd  Hello again :) It is fine to report this issue as you did. 
For that matter, Delta-Lock don't seem to be saved as well which make me think there is maybe a good reason not to for both, so I'll have a look on these.

vsonnier, 2017-08-12:
My bad, delta-lock is just fine.

vsonnier, 2017-08-12:
@dunaybd Hi! I've updated CubicSDR and now Solo Mode is saved into the session files.

closed 2017-08-12 by vsonnier

dunaybd, 2017-08-12:
Using the new source, I can do File/"Save Session As..", specify a session name, exit, restart, File/"Load Session" and the Solo mode is restored. Cool!

However, if I just use the default session, set Solo Mode on, exit and restart, Solo is cleared. 
Maybe this is related to the fact that I do not appear to have a default session. After I start CubicSDR, I cannot just do a "Save Session", I always have to point the file dialog to a previously saved session.

vsonnier, 2017-08-12:
@dunaybd Well I suppose we can imagine the same separation of concerns as Bookmarks vs. Sessions vs. application config. In this case, Solo mode is a modem functionality, while the general config is dedicated to things that are both modem and bookmark independent, something that is common to the whole application and something that is not changed very often. (UI element positions, for instance).
Solo Mode here is likely to change from context to context..

I myself never really use a 'blank' session : I always load one session file of interest as starting point so I don't really care of previous settings. From here, it is simple enough to do a 'Save as..' to get another ready-made configuration that can be modified, saved, and reused later.

There is a 'Reset session' command that is precisely intended to create a clean state from where a session with reference settings can be created.

dunaybd, 2017-08-12:
Makes perfect sense to me. I should load a session on startup, and that can easily be automated in a script. However, I can't seem to load a session at startup. For example, if I change directory to the .CubicSDR folder where "solo.xml" is located and run 
   "~/CubicSDR/build/x64/CubicSDR -c solo"
I see "Loading:: configuration file '/home/dunay/.CubicSDR/solo.xml" on the command line, but the solo session does not show up in the title bar until I manually do a File/"Load Session" -> solo.xml. 
The Solo mode always starts cleared, regardless of the contents of solo.xml. I also realized that saving a session with Solo mode set on and reloading it only works when Solo mode is already on. If I turn it off, re-loading the session will not turn Solo mode on.

Also, solo.xml always contains " <solo_mode>0</solo_mode>, even if I set Solo Mode to on and save the session. 

Another puzzler is that I can type 
   CubicSDR -c config
but the first message back on the command line is:
   Loading:: configuration file '/home/dunay/.CubicSDR/config-config.xml'

vsonnier, 2017-08-13:
You are actually trying  to  use the `-c` option to load a session file, while it is dedicated to load application config. This operation has therefore no effect, then all you observe are consequences of this misinterpretation of `-c` option. 
Long story short, you can't auto-load  a session file.

Now about the solo setting saved on the session proper, it works well so far as I tested it. Beware though, that on the XML content side it behaves like the `<active>` tag:
-  Solo mode On: `<solo_mode>1</solo_mode>` present in the session file,
-  Solo mode Off: `<solo_mode>` tag absent from the file. Although if you add  `<solo_mode>0</solo_mode>` manually to the file it should set Solo Mode Off.

dunaybd, 2017-08-13:
Thanks. Makes sense. :+1:

commit 33107bfa (2017-08-12): Fix #565, Solo mode is now saved into session files