cjcliffe/CubicSDR #566: Cubicsdr Windows 10 Nesdr Smart

WalkDN, 2017-08-04:
I am attempting an install on a Win 10 64 bit PC without success. The dongle is a Nesdr Smart from NooElec.com. NooElec provides links to the installation software. 

1.) I ran the Zadig driver install as suggested and think I have installed it properly (it completed the install without any error messages). How can I confirm that the driver for the RTL 2838 is installed?

2.) I ran the CubicSDR install without apparent problems until I got a prompt to install the Microsoft C++ 2015 X64 14.0.23026 package. This is where the install failed since I have a later version of the C++ package installed (14.0.24123.0) as shown in add/remove programs. The error message implied that I should remove 14.0.24123 but I am reluctant since other installs likely depends on it.

Suggestions?

Thanks
Dave W.

vsonnier, 2017-08-06:
@WalkDN Hello ! 
1.  When running Zadig again, you should have something similar as described [§3.1.4, with WinUSB present on both sides.](https://1plus1blog.com/2017/04/19/fabriquer-un-recepteur-ais-avec-antenne-rtl-sdr-et-opencpn/?utm_content=bufferc5346&utm_medium=social&utm_source=facebook.com&utm_campaign=buffer)
[See image.](https://unplusunblog.files.wordpress.com/2017/04/zadig4.jpg)

2. You have a more recent Visual C++ runtime ? It is perfectly fine.

So everything should be OK. 
Did you try to run CubicSDR ?

WalkDN, 2017-08-07:
First of all thanks for helping with this!

Below is a print/screen of the devices seen by my PC when I insert the SDR
into a USB port. The PC appears to identify the SRD as a keyboard!

[image: Inline images 1]

When I run Zadig, the RTL2838 does not show up as a device. The list below
shows the recognized devices. The USB port used by the RTL is Bulk-in,
Interface (Interface 1).

[image: Inline images 2]

Here is what I get when selecting that USB port:

[image: Inline images 3]
 Zadig will run at this point and install the driver shown (or so it says)
but Cubic won't find the device. BTW, I ignored the prompt when installing
Cubic to install the older version of C++, as you suggested. Cubic seems to
run fine, but as noted above, does not find the RTL2838.

Any additional suggestions are appreciated!

Dave W.

vsonnier, 2017-08-07:
@WalkDN Sorry I can't see your attached images. Please respond through GitHub next time, not by mail so you'll be able to link images and more without problem.
Anyway, a quick summary of the [RTL-SDR guide](http://www.rtl-sdr.com/rtl-sdr-quick-start-guide/) is:

-  Be sure to Use the lastest Zadig version (2.3) found [here.](http://zadig.akeo.ie/)
-  Plug your dongle in a particular port, one that is DIRECTLY attached to your PC (I don't think it works when plugged on a hub)
- Only then run your Zadig installer as Administrator (Right-Click --> Run as Administrator)
- Select Options --> List all devices in menu
- Select 'Bulk-In, Interface (Interface 0)' on the drop-down list.
- Click 'Replace' or 'Reinstall' driver. 

That was steps 7-8-9-10-11 in the [guide.](http://www.rtl-sdr.com/rtl-sdr-quick-start-guide/)
You should see screen 11 in particular, where Windows is to ask to confirm the installation of the driver.

WalkDN, 2017-08-07:
I havn't been running Zadig as admin...I'll give that a try.

Thanks, again.

WalkDN, 2017-08-07:
[image]

I ran Zadig as administrator, but never got to the screen where Win 10 asks if I want to install the driver. It all looked the same as my previous attempts. I decided to try running Cubic anyway and it seemed to work for a few moments with the waterfall and all, them quit with the message that I pasted above (if it shows up!). Win 10 said that there was a problem and shut Cubic down.

Whew...where to go from here?

vsonnier, 2017-08-08:
@WalkDN You can try to install the PothosSDR distribution, which includes a build of CubicSDR and GQRX for Windows x64 ==> [here.](http://downloads.myriadrf.org/builds/PothosSDR/?C=M;O=D)

vsonnier, 2017-08-18:
@WalkDN Any news ?

WalkDN, 2017-08-18:
The project is on hold for now, overwhelmed by other events. However, its
still on the desk waiting for attention. Thanks for you interest.

Dave W.

WalkDN, 2017-09-17:
The PothosSDR works perfectly on my Win 10 PC. Thanks

vsonnier, 2017-09-19:
@WalkDN thanks for your input. Have fun with Cubic SDR !

closed 2017-09-19 by vsonnier