cjcliffe/CubicSDR #568: No Audio/Sound with CubicSDR

AngryJoe13, 2017-08-10:
Sorry if this has been answered, I could not find it. I need some help.

I'm running Ubuntu 17.04 and I'm using CubicSDR + RTL-SDR (USB dongle)
I installed CubicSDR and was quickly up and running, no issues. Sound was working.
I shut down for the night, come back the next day, launch CubicSDR and everything appears to be working, however I have no sound/audio from CubicSDR.

More information:
I have audio from the system, GQRX, YouTube and other applications. 
Audio tests work fine via Ubuntu sound settings.
When running GQRX or another app, the app shows up in sound settings under Applications.
However CubicSDR does not.

I have removed and reinstalled CubicSDR.
I have removed and reinstalled PulseAudio.
I have reinstalled Unity control center and Ubuntu desktop to ensure there are no files missing.
I have created and added the TV dongle to the blacklist file.
I have run update, upgrade and dist-upgrade to make everything is up-to-date.

I have tried every setting I can find in CubicSDR but none have changed anything.
Yes, I have the volume in both Ubuntu and CubicSDR pushed to 100%.
Yes, I have made sure nothing is muted in Ubuntu or CubicSDR.

What else am I missing? Why would CubicSDR work and then not work, while everything else works?
On the same note, I know CubicSDR is compatible, because as I said it worked when I first installed it.

Any help is appreciated. Thanks!

vsonnier, 2017-08-18:
In CubicSDR you choose your Audio Device out per-modem. Create a modem, then select its source (collapsible window showing a +/- on the left of the screen). Maybe you have selected a muted output. Screenshots please ?

dunaybd, 2017-08-27:
Occasionally, the audio is not there for me with Linux Mint 18.2 (based on Ubuntu 16.04). I can get it back by changing the settings in pulse audio mixer, or logging out and back in, or even rebooting. Just a bit of instability.

egamboacr, 2018-04-02:
I have same problem with CubicSDR. Have reinstalled it and audio, and audio working in other programs, no go.

"In CubicSDR you choose your Audio Device out per-modem. Create a modem, then select its source (collapsible window showing a +/- on the left of the screen). Maybe you have selected a muted output. Screenshots please ?"  That does not work. I have Modem settings on upper left; coincidentally the first time I ran CubicSDR it had audio option there and I select default and it worked; now its blank. Right clicking there, waterfall, or any area does nothing, same with left click.

Benitoite, 2018-06-07:
Raspbian 9 stretch also exhibits this behavior. Worked ok the first time, then on next run audio drops out after 100 or so milliseconds. If I toggle the system’s audio mute button, I get a few ms of audio, then drops out again.  Error emitted is:
`DemodulatorThread::run() cannot push ati into audioOutputQueue, is full !`

vsonnier, 2019-03-31:
CubicSDR has works countless times already with Ubuntus from 16 to 18. Everything there is related to sound packages not properly installed vs. RtAudio configuration in Cubic, or a particular Linux installation. Since this issue is old, let's close it.

closed 2019-03-31 by vsonnier

safesecs, 2020-07-06:
You closed this with blame on the user and his machine. What a joke based on open source assumptions.  The same happened to me and I KNOW I am good... now, this POS application has shut down my right channel... nice job.

nlucis, 2022-07-02:
This is still an issue. Closing it because it was "old" was not the right move.

NikosEfthias, 2023-01-28:
i have the same issue sound just disappeared all of a sudden and no way to bring it back

tro-co, 2024-03-11:
probably just need to adjust the squelch

TwinSteels, 2024-06-13:
> In CubicSDR you choose your Audio Device out per-modem. Create a modem, then select its source (collapsible window showing a +/- on the left of the screen). Maybe you have selected a muted output. Screenshots please ?

I see no selection on the UI or ANY of the drop-down menus to "Create a modem".
I find No 'collapsible window' on the left of the screen showing a "+/-"...
My outputs are functioning via Speaker Tests, System Sounds are working, VLC working, Youtube working, but no audio from CubicSDR and  you can see the Dongle has a Commercial Broadcast signal at 99.5 mHz...

As you can also see the Dongle IS being AutoMagically detected by CSDR.

Tried several methods to get a freq in the 'dial' thinking the demod might be dependent on that field, but nothing I tried entered a freq there...

System:
  Kernel: 5.15.0-112-generic x86_64 bits: 64 compiler: gcc v: 11.4.0
    Desktop: Xfce 4.16.0 tk: Gtk 3.24.20 info: xfce4-panel wm: xfwm 4.16.1
    vt: 7 dm: LightDM 1.30.0 Distro: Linux Mint 21 Vanessa
    base: Ubuntu 20.04 focal
Audio:
  Sound Server-1: ALSA v: k5.15.0-112-generic running: yes
  Sound Server-2: JACK v: 0.125.0rc1 running: no
  Sound Server-3: PulseAudio v: 15.99.1 running: yes
  Sound Server-4: PipeWire v: 0.3.48 running: yes

Am I doing something wrong ?

73 - Bruce

[image]
[image]
[image]
[image]
[image]

TwinSteels, 2024-06-13:
SOLVED...

But didn't see a way to delete my post...
Stupid human trick...   Clicked on the blank segment upper left WF and bingo-bango audio erupted...   Along with the new modem designation in the top-left selection menu...

DOH !!!

jcoles56, 2024-10-20:
I installed cubicsdr from the repo on Linux Mint 22. I see spectrum but hear nothing. Starting from command line, I see the error message "No audio devices found!". How stupid! No other application has this problem. Is there something I can tweak in the config file to give this dumb app a clue?

Manual says, "'Audio Out' can be configured per-modem; each modem can be routed to any available audio output." HOW?

wzab, 2024-10-26:
I have the same issue in Debian/testing with pipewire. I have installed pipewire-pulse. Other applications using pulseaudio are working. In CubicSDR the "ModemSettings" window remains empty. The same with "AudioSampleRate" menu.
I have filled the bug report https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=1084177 but no solution up to now.

wzab, 2024-10-26:
I have tried to use "helvum" as described in https://github.com/cjcliffe/CubicSDR/issues/766 . But no CubicSDR output appears in helvum windows (again, other applications are visible).

Frayedknot64, 2025-03-20:
Still an issue for me - if you have configuration in pavcontrol output set to pro audio, which is what cubic sdr apparently wants, when you go to sound setting in the os it forces that to change to some other output from HDMI or something. If you set the output to play stereo quality music, cubic sdr will give no sound. It's just a mess.