cjcliffe/CubicSDR #569: Gdk-ERROR again

alfille, 2017-08-13:
This is similar to https://github.com/cjcliffe/CubicSDR/issues/377 which was closed but not addressed (that I can tell)

Basically, cloned and built CubicSDR on a Fedora system 

It crashes at startup:
[paul@new_snowmass build]$ CubicSDR 

(CubicSDR:28241): Gdk-ERROR **: The program 'CubicSDR' received an X Window System error.
This probably reflects a bug in the program.
The error was 'BadValue (integer parameter out of range for operation)'.
  (Details: serial 207 error_code 2 request_code 153 minor_code 3)
  (Note to programmers: normally, X errors are reported asynchronously;
   that is, you will receive the error a while after causing it.
   To debug your program, run it with the --sync command line
   option to change this behavior. You can then get a meaningful
   backtrace from your debugger if you break on the gdk_x_error() function.)
Trace/breakpoint trap (core dumped)

Ok, using --sync:

[paul@new_snowmass build]$ CubicSDR --sync

(CubicSDR:28349): Gdk-ERROR **: The program 'CubicSDR' received an X Window System error.
This probably reflects a bug in the program.
The error was 'BadValue (integer parameter out of range for operation)'.
  (Details: serial 323 error_code 2 request_code 153 minor_code 3)
  (Note to programmers: normally, X errors are reported asynchronously;
   that is, you will receive the error a while after causing it.
   To debug your program, run it with the --sync command line
   option to change this behavior. You can then get a meaningful
   backtrace from your debugger if you break on the gdk_x_error() function.)
Trace/breakpoint trap (core dumped)

This is a system with a LimeSDR (and SoapySDR installed) and an rtl-sdr dongle.

Any help would be appreciated.

DougCollinge, 2018-03-31:
Me too, on Ubuntu 17.10

cjcliffe, 2018-05-08:
@alfille @DougCollinge If you can please try https://github.com/cjcliffe/CubicSDR/releases/tag/0.2.4 AppImage build and let me know if you experience the same issue.  It is also the first build to include support for LimeSDR (though not fully tested yet).

DougCollinge, 2018-05-09:
Yes, it works! Much faster display, too.

DougCollinge, 2018-05-10:
Perhaps not, actually. The display is about the same as it was before.

ghulands, 2018-05-23:
I'm on Ubuntu 18.04 and just tried to build the 0.2.4 tag and it still fails for me.

`(CubicSDR:4053): Gdk-ERROR **: 19:42:15.505: The program 'CubicSDR' received an X Window System error.
This probably reflects a bug in the program.
The error was 'BadValue (integer parameter out of range for operation)'.
  (Details: serial 266 error_code 2 request_code 154 (GLX) minor_code 3)
  (Note to programmers: normally, X errors are reported asynchronously;
   that is, you will receive the error a while after causing it.
   To debug your program, run it with the GDK_SYNCHRONIZE environment
   variable to change this behavior. You can then get a meaningful
   backtrace from your debugger if you break on the gdk_x_error() function.)
Trace/breakpoint trap (core dumped)
`

vsonnier, 2018-05-23:
@JanuszWaw  also reported in #653:

> After compiling (from src) under Mint 18.3 Cinnamon 64bit (Linux PC-IRX 4.13.0-41-generic #46~16.04.1-Ubuntu SMP Thu May 3 10:06:43 UTC 2018 x86_64 x86_64 x86_64 GNU/Linux) and trying to start CubicSDR 0.2.4 I get a message: (the previous version 0.2.3, compiled in the same way works without a problem)

> $ ./CubicSDR
Loading:: configuration file '/<...>/.CubicSDR/config.xml'
Loaded 217 rig models via hamlib.
(CubicSDR:8922): Gdk-ERROR **: The program 'CubicSDR' received an X Window System error.
This probably reflects a bug in the program.
The error was 'BadValue (integer parameter out of range for operation)'.
(Details: serial 228 error_code 2 request_code 154 (GLX) minor_code 3)
(Note to programmers: normally, X errors are reported asynchronously;
that is, you will receive the error a while after causing it.
To debug your program, run it with the GDK_SYNCHRONIZE environment
variable to change this behavior. You can then get a meaningful
backtrace from your debugger if you break on the gdk_x_error() function.)
Pułapka debuggera/breakpoint
S

vsonnier, 2018-05-23:
A quick search on the Internet suggests a problem related to GLX libraries vs. drivers (Intel, Nvidia) and re-installing them. Please also note that wxWidgets **3.11** must be compiled with OpenGL support for CubicSDR to work.

Take a look at the [Build Linux wiki](https://github.com/cjcliffe/CubicSDR/wiki/Build-Linux) for instructions, or at https://github.com/loughkb/SDRplaylinux scripts, notably:
````
apt-get install mesa-utils libeglw1-mesa libglw1-mesa-dev
apt-get install freeglut3-dev freeglut3
````

JanuszWaw, 2018-05-23:
OK. Thank you for the solution. I will try this soon

vsonnier, 2018-06-07:
@JanuszWaw  Ping ! anay news on this issue ?

vsonnier, 2018-06-25:
Closing by lack of feedback.

closed 2018-06-25 by vsonnier

playaspec, 2020-03-26:
> Please also note that wxWidgets 3.11 must be compiled with OpenGL support for CubicSDR to work.

@vsonnier

How is this possible when the [the most recent **_development_** version is 3.1.3](https://www.wxwidgets.org/downloads/)? Is there a version of CubicSDR for mere mortals that only have their distro's stable installed?