cjcliffe/CubicSDR #575: CubicSDR not recognizing RTL2338UHIDR

J-Z-C, 2017-09-06:
I am running W7 Pro, 64 bit, SP1

Drivers installed via Zadig (a couple of times).
In Zadig, there is a window with choices:
USB Serial (CDC)
libusbK (v3.0.7.0)
libusb-win32 (v1.2.6.0)
WinUSB (v6.1.7600.16385)   - this one seems most appropriate

I have tried all of these except the win32 version, CubicSDR still does not find the SDR.
Which one is correct, do you know?

Windows sees the device under a variety of types, depending on which choice is used for the driver installation, windows says: 
device operating properly.
Location: Port_#0002.Hub_#0008

Driver info:
RTL2838UHIDIR(COM10)
provider: libwdi   date: 9/5/2017  version:1.0.0.0

CubicSDR did find the microphone, with fun results, but regardless of selctions, cannot find the USB radio.

Note that when trying to add the device manually, the whole dialog box does not appear, and cannot be re-sized.  I made some assumptions about what was not visible, and selected by using the tab key.
The field for typing in the location (port, etc.) is just barely visible.

vsonnier, 2017-09-06:
Hello, this is likely a duplicate of #566. Please follow the steps described there and report your experiments here.

vsonnier, 2017-09-19:
@J-Z-C Any news ?

J-Z-C, 2017-09-19:
Vincent;

I have not had time to work on the problem beyond confirming the proper installation of the various components.

HDSDR runs OK on this PC, so I know the device driver is there, but so far have not been able to get CubicSDR to recognize it, or connect manually.

Thanks for checking in.  When I get some time I will try again.
- JC

 

On Tue, 19 Sep 2017 10:58:18 +0000 (UTC), Vincent Sonnier  wrote:
 
@J-Z-C Any news ?

J-Z-C, 2018-02-25:
Using CubicSDR now for Airspy.  Working OK. 

I never did get it to work with the Noo SDR. 

I am ok with closing out the issue.

Thanks;
- JC

 

On Sun, 25 Feb 2018 08:21:48 +0000 (UTC), Vincent Sonnier  wrote:
 
Cleanup time: I'm closing this one by lack of feedback, and because I explained the steps already in its duplicate #566.