cjcliffe/CubicSDR #576: rig-enabled drops after invalid hamlib commands

tkzic, 2017-09-16:
I'm controlling the frequency of CubicSDR using "rig control" with "Hamlib net rigctl" device selected. Commands are being sent through a rigctld server on port 4532. It works great. But if you send an unrecognized command to rigctld, rig-control in Cubic SDR will drop and rig-control is unchecked. You then need to go to the menu and set the checkmark again. 

It will also happen sometimes if commands are sent too quickly for rigcltd/CubicSDR to handle. 

This is not a good situation when operating remotely because you need to go back to the computer running CubicSDR and reconnect rig-control. Is it possible to change the behavior so that CubicSDR ignores any unrecognized rig control commands?

Also, for remote operation, is there any way to set CubicSDR to initialize with the correct set of parameters for remote operation? This would allow you to cycle the power, for example, and have CubicSDR restart with the correct configuration and start running.

Thanks,
Tom

tkzic, 2017-09-16:
Forgot to mention I am using version 0.2.2 in Mac OS 10.12.6

cjcliffe, 2017-09-19:
@tkzic Sounds like the behavior when the rig thread errors out internally (since it resets the menu); most likely a bug, thanks!

There's a way to choose a specific -c configuration from the command line to load/save a specific init state but nothing that currently allows starting it automatically.  I'll add this as a separate enhancement request as a simple command line switch could be added for basic auto-start of session and device.