cjcliffe/CubicSDR #577: Auto-start session file and device from command line

cjcliffe, 2017-09-19:
From #576 it would be nice to be able to start automatically with a specific device and session file which could be done simply via command line.

edyirdaw, 2018-11-25:
Hi again @cjcliffe,

1. So this means that cubicSDR would have a terminal only running option, that is, without a GUI ever displayed while it receives a radio station? This can be very helpful indeed since one can start cubicSDR as a service when the computer it is currently running on restarts for some reason! This would make the management of this awesome tool completely autonomous. Making it more awesome :D

2. Indeed if this terminal option is available one needs to be able to have a rich set of command line inputs. Option 2.1 is quite crucial in my case :)

2.1 In my case there are multiple rtl-sdr devices plugged into the computer. So when the terminal based command of cubicSDR is run, one should be able to specify which device should be selected. Say, if three devices are plugged into the system, then an option from the values {0,1,2} can help to choose which of the devices to work with.

2.2 If #228 is made available in cubicSDR, it would be very great to simply specify which sink input cubicSDR should work with from the command line.

3. Do you have a time estimate of  when this issue would be made available in cubicSDR? Well, I need this feature so badly so soon that I tried to do the required coding myself. But looking at the code you wrote, it seems quite impractical for me to do it in terms of the time it takes me and the effort, it would require me perhaps upgrading my c++ skills and detailed knowledge of SDR.

I would like to take this opportunity to extend my gratitude and appreciation to you for building this app and letting us to use it!  A very useful app it turns out. It is motivating me to build my own OSS and help others in need of some functionality just as I needed this one.

cjcliffe, 2019-01-04:
@edyirdaw unfortunately this issue doesn't provide any sort of headless operation it simply automates the startup so that you can start from a specific state/setup.

There's https://github.com/rxseger/rx_tools which provides a decent set of command-line only headless tools which you might be able to script together with some basic bash/python/etc to automate some of the features.

Another one is https://github.com/vsergeev/luaradio which provides a high-level LUA scripting language interface which should let you use relatively simple LUA script to build the type of application you're describing.

Otherwise I don't have any plan to decouple CubicSDR from the GUI as a console application as I think tools like rx_tools, luaradio or gnuradio already have it covered.