cjcliffe/CubicSDR #580: something more next to Scope area?

pidloop, 2017-09-23:
I am using Version 0.2.2 on MacOS 10.12.6 with RFSpace CloudIQ. Everything works fine but the app hangs and must be force-quit if I click the Scope area. One time for a brief moment I something else try to rotate into view but never again. So now I'm curious what it could be. Hope this is fixed so I can find it :-)

cjcliffe, 2017-09-23:
@pidloop dragging the scope area left should reveal the spectrum analyzer; haven't tried on macOS 10.12 here myself so it's possible there's an issue.

pidloop, 2017-09-24:
Thanks for the reply. Yes, my report was not precise, just clicking on the Scope does nothing. But if I click and drag in the Scope all the graphics in the entire app, not just the Scope, immediately freeze. However the audio continues normally. After a second I get the infamous spinning beachball. When I open Force Quit, it says CubicSDR is "not responding". After forcing the quit, I can start again without problems as long as I avoid dragging the scope. 

To gain further insight, I ran the executable from the command line as 
`pushd /Applications/CubicSDR.app/Contents/MacOS; ./CubicSDR
`

After enumerating the OS audio devices there is no further output as I use the app until I perform the drag within Scope. After about a second of being hung I get a flurry of reports as follows:

`
Warning: ReBuffer 'SDRPostThreadVisualDataBuffers' count '101' exceeds threshold of '100'
Warning: ReBuffer 'SDRPostThreadVisualDataBuffers' count '102' exceeds threshold of '100'
Warning: ReBuffer 'SDRPostThreadVisualDataBuffers' count '103' exceeds threshold of '100'
Warning: ReBuffer 'SDRPostThreadVisualDataBuffers' count '104' exceeds threshold of '100'
Warning: ReBuffer 'SDRPostThreadVisualDataBuffers' count '105' exceeds threshold of '100'
Warning: ReBuffer 'SDRPostThreadVisualDataBuffers' count '106' exceeds threshold of '100'
Warning: ReBuffer 'SDRPostThreadVisualDataBuffers' count '107' exceeds threshold of '100'
Warning: ReBuffer 'SDRPostThreadVisualDataBuffers' count '108' exceeds threshold of '100'
Warning: ReBuffer 'SDRPostThreadVisualDataBuffers' count '109' exceeds threshold of '100'
Warning: ReBuffer 'SDRPostThreadVisualDataBuffers' count '110' exceeds threshold of '100'
Warning: ReBuffer 'SDRPostThreadVisualDataBuffers' count '111' exceeds threshold of '100'
Warning: ReBuffer 'SDRPostThreadVisualDataBuffers' count '112' exceeds threshold of '100'
Warning: ReBuffer 'SDRPostThreadVisualDataBuffers' count '113' exceeds threshold of '100'
Warning: ReBuffer 'SDRPostThreadVisualDataBuffers' count '114' exceeds threshold of '100'
Warning: ReBuffer 'SDRPostThreadVisualDataBuffers' count '115' exceeds threshold of '100'
Warning: ReBuffer 'SDRPostThreadVisualDataBuffers' count '116' exceeds threshold of '100'
Warning: ReBuffer 'SDRPostThreadVisualDataBuffers' count '117' exceeds threshold of '100'
Warning: ReBuffer 'SDRPostThreadVisualDataBuffers' count '118' exceeds threshold of '100'
Warning: ReBuffer 'SDRPostThreadVisualDataBuffers' count '119' exceeds threshold of '100'
Warning: ReBuffer 'SDRPostThreadVisualDataBuffers' count '120' exceeds threshold of '100'
Warning: ReBuffer 'SDRPostThreadVisualDataBuffers' count '121' exceeds threshold of '100'
Warning: ReBuffer 'SDRPostThreadVisualDataBuffers' count '122' exceeds threshold of '100'
Warning: ReBuffer 'SDRPostThreadVisualDataBuffers' count '123' exceeds threshold of '100'
Warning: ReBuffer 'SDRPostThreadVisualDataBuffers' count '124' exceeds threshold of '100'
Warning: ReBuffer 'SDRPostThreadVisualDataBuffers' count '125' exceeds threshold of '100'
Warning: ReBuffer 'SDRPostThreadVisualDataBuffers' count '126' exceeds threshold of '100'
Warning: ReBuffer 'SDRPostThreadVisualDataBuffers' count '127' exceeds threshold of '100'
Warning: ReBuffer 'SDRPostThreadVisualDataBuffers' count '128' exceeds threshold of '100'
`

Nothing further appears so I then created a core dump by typing:
`^\Quit (core dumped)`

The new file in /Cores is too large to attach here but I have attached a screen shot from the crash reporter if that helps.

Let me know if I can be of any assistance in troubleshooting this.

Thank you,

73

[image]

vsonnier, 2018-01-22:
@pidloop Hello ! Binaries for v0.23 and MacOS have been relased. Please tell us if the problem persists. Thanks !

pidloop, 2018-01-24:
Good now on macOS 10.13.2. Thanks!

vsonnier, 2018-01-24:
Great ! Have fun with Cubic !

closed 2018-01-24 by vsonnier