cjcliffe/CubicSDR #581: Make Errors - Using Ubuntu 14.04 32 bit

Davem734, 2017-10-16:
Following this procedure to install SDRplay and CubicSDR on a 32bit Ubuntu 14.04 system:

[http://www.sdrplay.com/docs/SDRplay_non_Windows_Flow.pdf](url)

I get the following errors when running CubicSDR make:

<[ 67%] Building CXX object CMakeFiles/CubicSDR.dir/src/visual/ImagePanel.cpp.o
/home/gmcmh/Apps/CubicSDR/src/visual/ImagePanel.cpp:13:6: warning: unused parameter ‘evt’ [-Wunused-parameter]
 void ImagePanel::paintEvent(wxPaintEvent & evt) {
      ^
[ 68%] Building CXX object CMakeFiles/CubicSDR.dir/src/process/VisualProcessor.cpp.o
[ 70%] Building CXX object CMakeFiles/CubicSDR.dir/src/process/ScopeVisualProcessor.cpp.o
In file included from /home/gmcmh/Apps/CubicSDR/src/process/ScopeVisualProcessor.h:6:0,
                 from /home/gmcmh/Apps/CubicSDR/src/process/ScopeVisualProcessor.cpp:4:
/home/gmcmh/Apps/CubicSDR/src/process/VisualProcessor.h: In constructor ‘VisualProcessor<AudioThreadInput, ScopeRenderData>::VisualProcessor()’:
/home/gmcmh/Apps/CubicSDR/src/process/VisualProcessor.h:14:7: error: conversion from ‘std::nullptr_t’ to non-scalar type ‘VisualProcessor<AudioThreadInput, ScopeRenderData>::VisualInputQueueTypePtr {aka std::shared_ptr<ThreadBlockingQueue<std::shared_ptr<AudioThreadInput> > >}’ requested
 class VisualProcessor {
       ^
/home/gmcmh/Apps/CubicSDR/src/process/ScopeVisualProcessor.cpp: In constructor ‘ScopeVisualProcessor::ScopeVisualProcessor()’:
/home/gmcmh/Apps/CubicSDR/src/process/ScopeVisualProcessor.cpp:8:90: note: synthesized method ‘VisualProcessor<AudioThreadInput, ScopeRenderData>::VisualProcessor()’ first required here 
 ScopeVisualProcessor::ScopeVisualProcessor(): outputBuffers("ScopeVisualProcessorBuffers") {
                                                                                          ^
make[2]: *** [CMakeFiles/CubicSDR.dir/src/process/ScopeVisualProcessor.cpp.o] Error 1
make[1]: *** [CMakeFiles/CubicSDR.dir/all] Error 2
make: *** [all] Error 2
>

vsonnier, 2017-10-16:
@Davem734 Hello ! Please use the lastest master to see if it fixes the problem for you.

Davem734, 2017-10-17:
Quite a few warnings but it did finish successfully. Thank you...

vsonnier, 2017-10-17:
@Davem734 Great ! Closing this one then.

commit 3cc49bdb (2017-10-16): Fixes #581 ?: Do not set std::shared_ptr to nullptr in constructors (useless anyway) since they could be wrongly interpreted
closed 2017-10-17 by vsonnier