cjcliffe/CubicSDR #582: USRP B210: app does not work

dmitryelj, 2017-10-21:
Hi.

Thanks for the great app.
A pity, but I cannot launch my USRP B210 with CubicSDR. 

CubicSDR show the receiver in the list:
https://s1.postimg.org/5of1404q5b/B210.jpg

But nothing works, only black screen, RX led on receiver is not on.
When I press Start-stop in the app again, it crashes.

System: Windows 10

Probably this will help for debugging:

**SoapySDRUtil.exe --probe="driver=uhd"** info:
  driver=b200
  hardware=B210
  fpga_version=14.0
  fw_version=8.0
  mboard_id=B210

  Channels: 2 Rx, 2 Tx
  Timestamps: YES
  Clock sources: internal, external, gpsdo
  Time sources: none, internal, external, gpsdo
  Sensors: ref_locked
  GPIOs: FP0, RXA, TXA

----------------------------------------------------
-- RX Channel 0
----------------------------------------------------
  Full-duplex: YES
  Supports AGC: NO
  Stream formats: CS8, CS12, CS16, CF32, CF64
  Native format: CS16 [full-scale=32768]
 
**Sample rates**:
Gains: PGA:(0.0, 76.0); 
Sample rates:
[31250.0, 31496.062992125986, 31746.031746031746, 32000.0, 32258.064516129034, 32520.325203252032, 32786.88524590164, 33057.85123966942, 33333.333333333336, 33613.445378151264, 33898.30508474576, 34188.03418803419, 34482.75862068965, 34782.608695652176, 35087.71929824561, 35398.23008849558, 35714.28571428572, 36036.036036036036, 36363.63636363636, 36697.247706422015, 37037.03703703704, 37383.17757009346, 37735.84905660377, 38095.23809523809, 38461.53846153846, 38834.95145631068, 39215.686274509804, 39603.960396039605, 40000.0, 40404.0404040404, 40816.32653061225, 41237.113402061856, 41666.666666666664, 42105.26315789474, 42553.1914893617, 43010.75268817204, 43478.260869565216, 43956.043956043955, 44444.444444444445, 44943.8202247191, 45454.545454545456, 45977.011494252874, 46511.62790697674, 47058.82352941176, 47619.04761904762, 48192.77108433735, 48780.48780487805, 49382.71604938272, 50000.0, 50632.91139240506, 51282.05128205128, 51948.05194805195, 52631.57894736842, 53333.333333333336, 54054.05405405405, 54794.520547945205, 55555.555555555555, 56338.028169014084, 57142.857142857145, 57971.014492753624, 58823.529411764706, 59701.49253731343, 60606.06060606061, 61538.46153846154, 62500.0, 62992.12598425197, 63492.06349206349, 64000.0, 64516.12903225807, 65040.650406504064, 65573.77049180328, 66115.70247933884, 66666.66666666667, 67226.89075630253, 67796.61016949153, 68376.06837606838, 68965.5172413793, 69565.21739130435, 70175.43859649122, 70796.46017699115, 71428.57142857143, 72072.07207207207, 72727.27272727272, 73394.49541284403, 74074.07407407407, 74766.35514018692, 75471.69811320755, 76190.47619047618, 76923.07692307692, 77669.90291262136, 78431.37254901961, 79207.92079207921, 80000.0, 80808.0808080808, 81632.6530612245, 82474.22680412371, 83333.33333333333, 84210.52631578948, 85106.3829787234, 86021.50537634408, 86956.52173913043, 87912.08791208791, 88888.88888888889, 89887.6404494382, 90909.09090909091, 91954.02298850575, 93023.25581395348, 94117.64705882352, 95238.09523809524, 96385.5421686747, 97560.9756097561, 98765.43209876544, 100000.0, 101265.82278481012, 102564.10256410256, 103896.1038961039, 105263.15789473684, 106666.66666666667, 108108.1081081081, 109589.04109589041, 111111.11111111111, 112676.05633802817, 114285.71428571429, 115942.02898550725, 117647.05882352941, 119402.98507462686, 121212.12121212122, 123076.92307692308, 125000.0, 125984.25196850394, 126984.12698412698, 128000.0, 129032.25806451614, 130081.30081300813, 131147.54098360657, 132231.40495867768, 133333.33333333334, 134453.78151260506, 135593.22033898305, 136752.13675213675, 137931.0344827586, 139130.4347826087, 140350.87719298244, 141592.9203539823, 142857.14285714287, 144144.14414414414, 145454.54545454544, 146788.99082568806, 148148.14814814815, 149532.71028037384, 150943.3962264151, 152380.95238095237, 153846.15384615384, 155339.8058252427, 156862.74509803922, 158415.84158415842, 160000.0, 161616.1616161616, 163265.306122449, 164948.45360824742, 166666.66666666666, 168421.05263157896, 170212.7659574468, 172043.01075268816, 173913.04347826086, 175824.17582417582, 177777.77777777778, 179775.2808988764, 181818.18181818182, 183908.0459770115, 186046.51162790696, 188235.29411764705, 190476.19047619047, 192771.0843373494, 195121.9512195122, 197530.86419753087, 200000.0, 202531.64556962025, 205128.20512820513, 207792.2077922078, 210526.31578947368, 213333.33333333334, 216216.2162162162, 219178.08219178082, 222222.22222222222, 225352.11267605633, 228571.42857142858, 231884.0579710145, 235294.11764705883, 238805.97014925373, 242424.24242424243, 246153.84615384616, 250000.0, 253968.25396825396, 258064.51612903227, 262295.08196721313, 266666.6666666667, 271186.4406779661, 275862.0689655172, 280701.7543859649, 285714.28571428574, 290909.0909090909, 296296.2962962963, 301886.7924528302, 307692.3076923077, 313725.49019607843, 320000.0, 326530.612244898, 333333.3333333333, 340425.5319148936, 347826.0869565217, 355555.55555555556, 363636.36363636365, 372093.0232558139, 380952.38095238095, 390243.9024390244, 400000.0, 410256.41025641025, 421052.63157894736, 432432.4324324324, 444444.44444444444, 457142.85714285716, 470588.23529411765, 484848.48484848486, 500000.0, 516129.03225806454, 533333.3333333334, 551724.1379310344, 571428.5714285715, 592592.5925925926, 615384.6153846154, 640000.0, 666666.6666666666, 695652.1739130435, 727272.7272727273, 761904.7619047619, 800000.0, 842105.2631578947, 888888.8888888889, 941176.4705882353, 1000000.0, 1066666.6666666667, 1142857.142857143, 1230769.2307692308, 1333333.3333333333, 1454545.4545454546, 1600000.0, 1777777.7777777778, 2000000.0, 2285714.285714286, 2666666.6666666665, 3200000.0, 4000000.0, 5333333.333333333, 8000000.0, 16000000.0]

vsonnier, 2017-10-22:
I made a test with a modified SoapySDRPlay returning a insane amount of sample rates: It works, but the menus and drop-down-list are practically unusable because too long. 
I was half expecting a crash because at some distant time putting > 255 items in a menu would create a crash on Windows.
Long story short, this do not look like a GUI problem.

Still, we need to find a way to restraint the amount of displayed items, and/or allowing a better way to enter samples rates so the GUI part improvement is directly related to #513.

@dmitryelj Unfortunatly I do not own the device so I can't help you here. Did you try the PothosSDR distribution [here](http://downloads.myriadrf.org/builds/PothosSDR/?C=M;O=D) to run your device ?

dmitryelj, 2017-10-22:
Hi. I using CubicSDR from Pothos pack, yes. 

I'm not sure that its a menu issue - the menu itself works:
https://s1.postimg.org/7ihojqz8e7/usrp2.jpg
Bit its just a black screen, no signal and no waterfall. 

Interesting to mention, that GQRX from the same Pothos pack works, there is a waterfall and decoding, but the top area is empty:
https://s1.postimg.org/5vnd78y4fz/usrp3.jpg

I understand that its hard to test without a device.

vsonnier, 2017-10-22:
@dmitryelj Out of curiosity, what are the names of the Antennas listed in both CubicSDR and GQRX ?

dmitryelj, 2017-10-22:
The device has 4 connectors, but both CubicSDR and GQRX allows to display only 2:
https://s1.postimg.org/6jfmikg3vz/usrp_antennas.jpg

vsonnier, 2017-10-22:
CubicSDR (and apparently GQRX) only displays the RX antennas as reported by the SoapySDR API, so the others may be TX only antennas.

dmitryelj, 2017-10-22:
Yes, its a bit complicated in B210 - it has 2 "motherboards" inside, so first the "mboard" should be selected, and each board has 2 antenna ports. If Soapy API has no method to select the board, 2 other inputs cannot be in use. In GNU radio it works, btw.

vsonnier, 2017-10-24:
@dmitryelj Hello again, can you try to run CubicSDR on the command line as I showed you ? The log could be useful.

dmitryelj, 2017-10-24:
Hi. I tried now.

[cubic.log](https://github.com/cjcliffe/CubicSDR/files/1409662/cubic.log)

vsonnier, 2017-10-24:
Summing up the log, the important part are:
Loading:: configuration file 'C:\Users\DMITRII\AppData\Roaming\CubicSDR\config.xml'

...

SDR enumerator starting.
SoapySDR init..
	API Version: v0.6.0
	ABI Version: v0.6
	Install root: C:\Program Files\PothosSDR
	Loading modules... 
...
Available factories...airspy, audio, bladerf, hackrf, lime, miri, null, osmosdr, redpitaya, remote, rtlsdr, sdrplay, **uhd**
Loaded font 'Bitstream Vera Sans Mono' from '..\share\cubicsdr\fonts\vera_sans_mono27_0.png', parsed 255 characters.
Loaded font 'Bitstream Vera Sans Mono' from '..\share\cubicsdr\fonts\vera_sans_mono32_0.png', parsed 255 characters.
Loaded font 'Bitstream Vera Sans Mono' from '..\share\cubicsdr\fonts\vera_sans_mono36_0.png', parsed 255 characters.
**-- Loading firmware image: C:\Program Files\UHD\share\uhd\images\usrp_b200_fw.hex...**
  default_input = True
  default_output = True
  device_id = 0
 **...**
  **driver = uhd**
  label = B210 3113F1D
  name = MyB210
  product = B210
  serial = 3113F1D
  type = b200
Make device 3
  fpga_version=14.0
  fw_version=8.0
  mboard_id=B210
  mboard_name=MyB210
  mboard_serial=3113F1D
  rx0_antenna=RX2
  rx0_id=Unknown (0xffff)
  rx0_serial=
  rx0_subdev_name=FE-RX2
  rx0_subdev_spec=A:A A:B
  rx1_antenna=RX2
  rx1_id=Unknown (0xffff)
  rx1_serial=
  rx1_subdev_name=FE-RX1
  rx1_subdev_spec=A:A A:B
  tx0_antenna=TX/RX
  tx0_id=Unknown (0xffff)
  tx0_serial=
  tx0_subdev_name=FE-TX2
  tx0_subdev_spec=A:A A:B
  tx1_antenna=TX/RX
  tx1_id=Unknown (0xffff)
  tx1_serial=
  tx1_subdev_name=FE-TX1
  tx1_subdev_spec=A:A A:B

Reporting enumeration complete.
SDR enumerator done.
SDR thread starting.
device init()
**Device Stream MTU: 0** _<== The horror !_
starting readLoop()
SDRThread::readStream(): 2. SoapySDR read failed with code: -4
SDRThread::readStream(): 3.1 iqDataOutQueue output queue is full, discard processing of the batch...
SDRThread::readStream(): 2. SoapySDR read failed with code: -3
SDRThread::readStream(): 3.1 iqDataOutQueue output queue is full, discard processing of the batch...
SDRThread::readStream(): 2. SoapySDR read failed with code: -4
SDRThread::readStream(): 3.1 iqDataOutQueue output queue is full, discard processing of the batch...
...

SDRThread::readStream(): 2. SoapySDR read failed with code: -3
SDRThread::readStream(): 3.1 iqDataOutQueue output queue is full, discard processing of the batch...
readLoop() ended.
device deinit()
SDR thread done.
SDR thread starting.
device init()
**Device Stream MTU: 0** _<== The horror, once again !_
starting readLoop()
SDRThread::readStream(): 2. SoapySDR read failed with code: -4
SDRThread::readStream(): 3.1 iqDataOutQueue output queue is full, discard processing of the batch...

This means that instead of reading the samples from the device, `SDRThread::readStream()` returns error code -3 and -4, so this explains the empry CubicSDR interface.

Codes -3 and -4 are described as:

```cpp 
/*!
 * Returned when read has data corruption.
 * For example, the driver saw a malformed packet.
 */
#define SOAPY_SDR_CORRUPTION (-3)

/*!
 * Returned when read has an overflow condition.
 * For example, and internal buffer has filled.
 */
#define SOAPY_SDR_OVERFLOW (-4) 
```
At that point SOAPY_SDR_CORRUPTION and SOAPY_SDR_OVERFLOW are minor annoyances compared to a MTU size of ZERO, with which CubicSDR cannot certainly work. This is because the SoapySDR API call `getStreamMTU` returned 0, and I don't see why.

I suggest opening an issue  on https://github.com/pothosware/SoapyUHD copying this post.

dmitryelj, 2017-10-24:
Thanks a lot, I've created a ticket.

dmitryelj, 2017-10-24:
Thanks again for help, I launched it with this settings.
[image]

closed 2017-10-24 by dmitryelj