cjcliffe/CubicSDR #583: v2.0 no "Audio Output" Menue on MacOS Sierra (10.12.6)

aneitzel, 2017-10-22:
Hi,

on latest CubicSDR version i dont have menue "audio output" to select the audio device. I tested it with older v1.20 - with this it works and audio menue is shown as expected.

I would pipe sound thruth "Loopback" device to a digital decoder (DSD+).
Seems to be a bug :-(

vsonnier, 2017-10-23:
The audio menu no longer exists, audio output has been moved : see in  http://cubicsdr.readthedocs.io/en/latest/application-window.html "Modem properties" where you'll find it again.

vsonnier, 2017-10-23:
@cjcliffe Hello ! A side note: maybe you should modify your site www.cubicsdr.com a bit to point on GirHub releases because plenty of people do not seem to go beyond the original site and so use an outdated version.

aneitzel, 2017-10-23:
Thanks for this hint.... i was not aware of this change. Its ok and working!

closed 2017-10-23 by aneitzel

cjcliffe, 2017-11-02:
@vsonnier I'll see what I can do about making it publish the releases automatically; should probably get a new one up soon anyways with all the latest fixes applied